crankyoldgit/IRremoteESP8266 #2007: Gree protocol sending issue using IRac class

HomeKidd, 2023-06-25:
### Version/revision of the library used
2.8.5

### Describe the bug
Mainly I'm using IRac class and stdAc::state_t for controlling ACs (like an universal remote). Since (at least) the last update my Gree AC acting weird, like turning the  stdAc::state_t .light toggle on is toggling the Econo setting. And something else going on too, because the XFan settings also toggled by not even touching this setting. Real remote toggles it, decoding results and copying it to the corresponding stdAc::state_t also works as expected. After reading the source code it my be something since Gree class received some updates for available features. Guessing the problem is In IRac.cpp, [line 1323](https://github.com/crankyoldgit/IRremoteESP8266/blob/80821751269b07bcf0385da75e8404792b8fd317/src/IRac.cpp#L1313) and IRac.cpp, [line 3257](https://github.com/crankyoldgit/IRremoteESP8266/blob/80821751269b07bcf0385da75e8404792b8fd317/src/IRac.cpp#L3257) doesn't match up.

```
#if SEND_GREE
void IRac::gree(IRGreeAC *ac, const gree_ac_remote_model_t model,
                const bool on, const stdAc::opmode_t mode, const bool celsius,
                const float degrees, const stdAc::fanspeed_t fan,
                const stdAc::swingv_t swingv, const stdAc::swingh_t swingh,
                const bool iFeel, const bool turbo, const bool econo,
                const bool light, const bool clean, const int16_t sleep) {
  ac->begin();
  ac->setModel(model);
  ac->setPower(on);
  ac->setMode(ac->convertMode(mode));
  ac->setTemp(degrees, !celsius);
  ac->setFan(ac->convertFan(fan));
  ac->setSwingVertical(swingv == stdAc::swingv_t::kAuto,  // Set auto flag.
                       ac->convertSwingV(swingv));
  ac->setSwingHorizontal(ac->convertSwingH(swingh));
  ac->setIFeel(iFeel);
  ac->setLight(light);
  ac->setTurbo(turbo);
  ac->setEcono(econo);
  ac->setXFan(clean);
  ac->setSleep(sleep >= 0);  // Sleep on this A/C is either on or off.
  // No Econo setting available.
  // No Filter setting available.
  // No Beep setting available.
  // No Quiet setting available.
  // No Clock setting available.
  ac->send();
}
#endif  // SEND_GREE
```

```
#if SEND_GREE
    case GREE:
    {
      IRGreeAC ac(_pin, (gree_ac_remote_model_t)send.model, _inverted,
                  _modulation);
      gree(&ac, (gree_ac_remote_model_t)send.model, send.power, send.mode,
           send.celsius, send.degrees, send.fanspeed, send.swingv, send.swingh,
           send.turbo, send.econo, send.light, send.clean, send.sleep);
      break;
    }
#endif  // SEND_GREE
```

#### To Reproduce
Sending IR Messages using IRAC class for Gree AC, mainly my issues are with Light on/off. 

#### Output of raw data from [IRrecvDumpV2.ino]

**Behaviour using the IRRemote (toggle the LED/Light):**
```
13:00:31.387 -> Protocol  : GREE
13:00:31.387 -> Code      : 0x4907C050112000D4 (64 Bits)
13:00:31.387 -> Mesg Desc.: Model: 1 (YAW1F), Power: On, Mode: 1 (Cool), Temp: 23C, Fan: 0 (Auto), Turbo: Off, Econo: On, IFeel: Off, WiFi: Off, XFan: On, Light: Off, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 1 (Auto), Swing(H): 1 (Auto), Timer: Off, Display Temp: 0 (Off)
```

**Expected behaviour using the remote controller (toggle the LED/Light):**

```
13:03:58.032 -> Protocol  : GREE
13:03:58.032 -> Code      : 0x49074050014000E0 (64 Bits)
13:03:58.032 -> Mesg Desc.: Model: 1 (YAW1F), Power: On, Mode: 1 (Cool), Temp: 23C, Fan: 0 (Auto), Turbo: Off, Econo: Off, IFeel: Off, WiFi: On, XFan: Off, Light: Off, Sleep: Off, Swing(V) Mode: Auto, Swing(V): 1 (Auto), Swing(H): 0 (Off), Timer: Off, Display Temp: 0 (Off)
```

### What brand/model IR demodulator are you using?
Vishay TSOP4838

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yep

### Has this library/code previously worked as expected for you?
Yes, but i don't know which version was it, as i remember 2.8.3

NiKiZe, 2023-06-25:
If you know this has worked previously, would you consider doing a git bisect to find what broke?

I like to refer to https://ipxe.org/howto/bisect as a quick start guide.

HomeKidd, 2023-06-25:
> I like to refer to https://ipxe.org/howto/bisect as a quick start guide.

Instead of doing that, i've went the simple way.

Changed [line 3257](https://github.com/crankyoldgit/IRremoteESP8266/blob/80821751269b07bcf0385da75e8404792b8fd317/src/IRac.cpp#L3257) in IRac.cpp to match with the declaration in [line 1323](https://github.com/crankyoldgit/IRremoteESP8266/blob/80821751269b07bcf0385da75e8404792b8fd317/src/IRac.cpp#L1313), and  only by adding  _send.iFeel_  parameter to the right place (to match the parameter order in the declaration) it **works as expected!** 😄 

**So changed this:**
```
      gree(&ac, (gree_ac_remote_model_t)send.model, send.power, send.mode,
           send.celsius, send.degrees, send.fanspeed, send.swingv, send.swingh,
           send.turbo, send.econo, send.light, send.clean, send.sleep);
```
**To this:**
```
          gree(&ac, (gree_ac_remote_model_t)send.model, send.power, send.mode,
           send.celsius, send.degrees, send.fanspeed, send.swingv, send.swingh,
           send.iFeel, send.turbo, send.econo, 
           send.light, send.clean, send.sleep);
```

NiKiZe, 2023-06-25:
Still want to know why something that did work, stopped working, and be sure it doesn't happen again.

HomeKidd, 2023-06-27:
> Still want to know why something that did work, stopped working, and be sure it doesn't happen again.

If i have some spare time i'll make it using bisect. 
Until that checking the file history in commit #1928 the first part of the Gree function was modified in [Line 1334](https://github.com/crankyoldgit/IRremoteESP8266/blob/80821751269b07bcf0385da75e8404792b8fd317/src/IRac.cpp#L1334) but i couldn't find any modification near [line 3257](https://github.com/crankyoldgit/IRremoteESP8266/blob/80821751269b07bcf0385da75e8404792b8fd317/src/IRac.cpp#L3257). The code was compiling, running, but the wrong settings were sent and this messed up the whole Gree (sending) protocol

PR #2008 (2023-06-28): [BUG] Fix missing parameter in call to `IRac::gree()`
#1928 missed extending the call to `gree()` resulting in subsequent parameters being shifted by one place.

Fixes #2007
commit f5f8710a (2023-06-28): [BUG] Fix missing parameter in call to `IRac::gree()` (#2008)

PR #1928 missed extending the call to `gree()` resulting in subsequent parameters being shifted by one place.

Fixes #2007
closed 2023-06-28 by crankyoldgit