crankyoldgit/IRremoteESP8266 #2009: Request for RecPro / Houghton RV Air Conditioner Protocol

mrmees, 2023-06-29:
This is becoming a very popular model of RV air conditioner and I would love to have it integrated into IRremote.  

Unit: RP-AC3501-W-KT
Remote: HD01-C1

A raw dump is included along with one that has comments regarding the actions being taken.  

auto_analyze_raw_data.py output:
```
Found 267 timing entries.
Potential Mark Candidates:
[9296, 4624, 392]
Potential Space Candidates:
[20966, 20464, 6662, 4912, 2640, 966, 436]
DANGER: Unusual number of mark timings!
Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 4615
kHdrSpace  = 2640
kBitMark   = 363
kOneSpace  = 903
kZeroSpace = 379
kLdrMark   = 9296
[IR Dump.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/11911129/IR.Dump.txt)
[IR Dump With Comments.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/11911130/IR.Dump.With.Comments.txt)

kSpaceGap1 = 20966
kSpaceGap2 = 20464
kSpaceGap3 = 6662
kSpaceGap4 = 4912

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+0110100001001000001010101110100000000000000000000110010000001100GAP(20464)
  Bits: 64
  Hex:  0x68482AE80000640C (MSB first)
        0x3026000017541216 (LSB first)
  Dec:  7514303154188477452 (MSB first)
        3469460563326931478 (LSB first)
  Bin:  0b0110100001001000001010101110100000000000000000000110010000001100 (MSB first)
        0b0011000000100110000000000000000000010111010101000001001000010110 (LSB first)
kHdrMark+UNEXPECTED->GAP(6662)kLdrMark+UNEXPECTED->GAP(4912)kBitMark(UNEXPECTED)0000000000000000000101101001111000000000000000000100101000001010GAP(20966)
  Bits: 64
  Hex:  0x0000169E00004A0A (MSB first)
        0x5052000079680000 (LSB first)
  Dec:  24867860662794 (MSB first)
        5787688473161367552 (LSB first)
  Bin:  0b0000000000000000000101101001111000000000000000000100101000001010 (MSB first)
        0b0101000001010010000000000000000001111001011010000000000000000000 (LSB first)
kHdrMark+
Total Nr. of suspected bits: 128

Generating a VERY rough code outline:

// Copyright 2020 David Conran (crankyoldgit)
/// @file
/// @brief Support for TBD protocol

// Supports:
//   Brand: TBD,  Model: TODO add device and remote

#include "IRrecv.h"
#include "IRsend.h"
#include "IRutils.h"

// WARNING: This probably isn't directly usable. It's a guide only.

// See https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol
[... 219 lines cut ...]
  if (used == 0) return false;  // We failed to find any data.
  offset += used;  // Adjust for how much of the message we read.
  pos += 8;  // Adjust by how many bytes of data we read

  // Data Section #2
  // e.g.
  //   bits = 64; bytes = 8;
  //   *(results->state + pos) = {0x00, 0x00, 0x16, 0x9E, 0x00, 0x00, 0x4A, 0x0A};
  used = matchGeneric(results->rawbuf + offset, results->state + pos,
                      results->rawlen - offset, 64,
                      kLdrMark, 0,
                      kBitMark, kOneSpace,
                      kBitMark, kZeroSpace,
                      kBitMark, kSpaceGap, true);
  if (used == 0) return false;  // We failed to find any data.
  offset += used;  // Adjust for how much of the message we read.
  pos += 8;  // Adjust by how many bytes of data we read

  // Success
  results->decode_type = decode_type_t::TBD;
  results->bits = nbits;
  return true;
}
#endif  // DECODE_TBD
```

NiKiZe, 2023-06-30:
Please start with raw values, from the lowest temperature, to the highest, one step at a time.

mrmees, 2023-07-03:
So I think I've got it pretty well worked out.  Spent a really inappropriate amount of time going through the signals and have a good grasp on everything that's happening with the exception of one unmarked button in the remote.   The signal is recognized as Carrier-128 protocol, but it does not seem to follow the same breakdown as far as I can tell from reading through the Carrier stuff.  

I created a 3 sheet spreadsheet that can be accessed here:  https://1drv.ms/x/s!Alz2mRtd8II7itATCDTB9I0j-8cZIg?e=tB9whg

DIRECT GITHUB LINK HERE:  [Houghton AC IR Protocol.xlsx](https://github.com/crankyoldgit/IRremoteESP8266/files/11941077/Houghton.AC.IR.Protocol.xlsx)

- Sheet "Data Import" - Raw bit values imported from serial dump / power query. 
- Data Analysis - Visual representation / decoding of data from dump. 
- Byte Breakdown - Byte by byte explanation of analysis results.  

I think it's good to go based on my testing.  Byte 11 is a little weird, but only when I press an unmarked button on the remote control.  It doesn't look like any other manufacturer models use it, so I'm thinking it must be leftover junk from whatever design they stole it from originally.  

Let me know if there's anything else that would help to get this into the project.

NiKiZe, 2023-07-04:
Yes decoding protocols is a pain :D
For that reason there is a FAQ how we go about it.

It is great effort to do what you have done, but we need to go back a bit.

Please post the raw collection of timing data for your full temperature range, the lowest to the highest, one step at a time.
It is intended to be posted as text.
The code can then be generated from that, including test cases.

This will allow to create basic support for the protocol, and with the most reasonable endiness.
After that comes decoding the actual data. - and that depends on how the structure comes out after the previous step.

mrmees, 2023-07-05:
Here is the result of resetting the remote to defaults and going from minimum to maximum temperature (in Celsius).  

```
Timestamp : 000021.528
Library   : v2.8.5

Protocol  : CARRIER_AC128
Code      : 0x16220412101016B000000800000016F0 (128 Bits)
uint16_t rawData[267] = {4574, 2690,  362, 378,  366, 892,  358, 896,  360, 384,  360, 894,  360, 388,  358, 382,  366, 384,  366, 384,  360, 894,  362, 380,  364, 386,  362, 380,  366, 892,  362, 382,  362, 382,  366, 380,  368, 384,  362, 894,  360, 384,  366, 340,  406, 388,  360, 382,  366, 384,  362, 332,  412, 892,  362, 382,  340, 412,  358, 898,  358, 382,  364, 384,  366, 382,  364, 388,  362, 388,  358, 380,  368, 388,  362, 892,  360, 384,  362, 388,  362, 384,  366, 380,  368, 380,  366, 384,  366, 380,  364, 894,  360, 380,  366, 386,  360, 390,  358, 384,  362, 892,  358, 892,  360, 382,  364, 894,  360, 384,  364, 382,  366, 384,  362, 342,  408, 388,  360, 388,  362, 382,  366, 894,  358, 892,  362, 380,  366, 892,  360, 20488,  4582, 6696,  9296, 4912,  358, 384,  364, 386,  358, 386,  368, 384,  362, 388,  358, 388,  362, 382,  366, 388,  360, 388,  360, 388,  360, 388,  386, 344,  376, 390,  358, 388,  360, 388,  358, 388,  360, 388,  360, 386,  364, 382,  362, 896,  360, 388,  358, 390,  360, 384,  338, 412,  360, 390,  358, 388,  360, 390,  358, 388,  358, 388,  362, 388,  332, 414,  360, 388,  358, 388,  362, 388,  360, 388,  360, 388,  336, 410,  362, 388,  360, 384,  340, 410,  360, 388,  360, 386,  364, 388,  364, 382,  362, 386,  360, 388,  360, 388,  358, 388,  360, 384,  362, 892,  362, 892,  360, 386,  362, 890,  366, 384,  358, 388,  360, 390,  358, 388,  360, 388,  360, 388,  360, 388,  360, 892,  360, 892,  358, 896,  358, 894,  358, 20982,  4578};  // CARRIER_AC128
uint8_t state[16] = {0x16, 0x22, 0x04, 0x12, 0x10, 0x10, 0x16, 0xB0, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x16, 0xF0};

Timestamp : 000028.527
Library   : v2.8.5

Protocol  : CARRIER_AC128
Code      : 0x16220412101017C000000800000023D0 (128 Bits)
uint16_t rawData[267] = {4582, 2692,  362, 378,  370, 890,  362, 890,  362, 388,  362, 888,  360, 384,  366, 382,  368, 382,  366, 384,  358, 896,  362, 380,  368, 384,  362, 340,  408, 894,  358, 384,  362, 382,  368, 384,  362, 384,  362, 894,  360, 386,  362, 386,  360, 388,  360, 388,  360, 338,  410, 388,  358, 894,  360, 388,  358, 384,  362, 894,  360, 384,  362, 384,  362, 384,  366, 382,  366, 384,  362, 384,  364, 334,  386, 918,  362, 384,  336, 414,  358, 388,  360, 386,  362, 384,  362, 384,  362, 390,  360, 890,  360, 390,  358, 380,  366, 384,  362, 892,  334, 920,  358, 894,  358, 344,  402, 894,  334, 414,  358, 384,  340, 414,  334, 412,  334, 414,  358, 388,  360, 380,  340, 414,  358, 386,  338, 918,  358, 894,  358, 20482,  4580, 6696,  9296, 4912,  356, 390,  334, 412,  358, 390,  338, 410,  360, 388,  360, 382,  366, 380,  366, 384,  338, 412,  360, 388,  360, 388,  360, 386,  364, 388,  360, 388,  358, 388,  360, 390,  356, 392,  360, 386,  334, 412,  360, 894,  358, 392,  358, 388,  358, 390,  358, 390,  358, 386,  362, 388,  358, 388,  360, 386,  360, 390,  358, 390,  360, 388,  358, 388,  360, 388,  360, 388,  360, 390,  360, 388,  358, 388,  362, 388,  358, 388,  358, 390,  358, 388,  360, 390,  358, 390,  358, 390,  358, 388,  362, 390,  356, 390,  356, 390,  358, 894,  358, 896,  356, 390,  358, 390,  356, 388,  358, 898,  358, 390,  356, 392,  358, 388,  358, 390,  356, 392,  356, 390,  356, 900,  354, 390,  356, 896,  356, 896,  358, 20998,  4598};  // CARRIER_AC128
uint8_t state[16] = {0x16, 0x22, 0x04, 0x12, 0x10, 0x10, 0x17, 0xC0, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x23, 0xD0};

Timestamp : 000030.527
Library   : v2.8.5

Protocol  : CARRIER_AC128
Code      : 0x16220412101018D000000800000025F0 (128 Bits)
uint16_t rawData[267] = {4576, 2688,  360, 384,  364, 892,  362, 890,  362, 380,  366, 892,  360, 384,  364, 386,  364, 380,  370, 380,  366, 894,  358, 380,  366, 384,  366, 380,  366, 894,  362, 382,  364, 382,  366, 384,  366, 388,  358, 894,  360, 384,  366, 384,  362, 384,  366, 380,  366, 338,  410, 384,  362, 892,  362, 382,  364, 384,  362, 892,  360, 386,  364, 382,  364, 388,  360, 384,  362, 384,  366, 388,  360, 386,  336, 916,  360, 388,  360, 388,  362, 380,  366, 384,  366, 384,  362, 382,  366, 384,  362, 892,  362, 386,  336, 412,  336, 414,  358, 382,  342, 414,  358, 380,  366, 894,  360, 896,  360, 384,  366, 384,  362, 384,  366, 386,  362, 384,  362, 382,  364, 382,  366, 892,  360, 386,  362, 892,  360, 892,  362, 20486,  4556, 6718,  9298, 4912,  360, 386,  360, 390,  362, 384,  362, 384,  362, 388,  360, 390,  360, 382,  366, 384,  366, 388,  358, 388,  362, 388,  358, 388,  362, 388,  358, 388,  360, 390,  360, 388,  360, 388,  360, 388,  360, 388,  358, 894,  358, 390,  360, 388,  360, 388,  358, 388,  358, 390,  360, 388,  358, 390,  336, 412,  360, 388,  358, 390,  358, 388,  362, 386,  338, 412,  362, 386,  362, 386,  356, 390,  362, 388,  360, 388,  362, 388,  358, 388,  360, 388,  360, 388,  362, 388,  360, 388,  358, 390,  360, 388,  360, 386,  360, 388,  360, 894,  358, 388,  360, 894,  358, 388,  360, 386,  360, 896,  360, 388,  358, 390,  358, 390,  358, 388,  362, 388,  358, 388,  360, 892,  362, 890,  362, 894,  358, 894,  360, 21004,  4600};  // CARRIER_AC128
uint8_t state[16] = {0x16, 0x22, 0x04, 0x12, 0x10, 0x10, 0x18, 0xD0, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x25, 0xF0};

Timestamp : 000033.027
Library   : v2.8.5

Protocol  : CARRIER_AC128
[... cut ...] 332, 414,  334, 416,  332, 414,  332, 416,  334, 412,  334, 414,  332, 416,  332, 416,  332, 416,  330, 416,  332, 418,  330, 416,  332, 418,  330, 416,  334, 414,  332, 922,  330, 416,  330, 922,  332, 416,  332, 920,  330, 418,  332, 922,  330, 418,  328, 418,  330, 418,  330, 418,  328, 420,  330, 416,  332, 922,  328, 418,  328, 420,  328, 21028,  4572};  // CARRIER_AC128
uint8_t state[16] = {0x16, 0x22, 0x04, 0x12, 0x10, 0x10, 0x30, 0x70, 0x00, 0x00, 0x08, 0x00, 0x00, 0x00, 0x55, 0x20};
```

mrmees, 2023-07-05:
Created a bit map of the information that is hopefully a little easier to read than the other sheet.  

https://docs.google.com/spreadsheets/d/1LJzJOy8rIhit1_dYZVBFdrprVubOtzBqAdxtzQidcsw/edit?usp=sharing

mrmees, 2023-07-12:
> Yes decoding protocols is a pain :D For that reason there is a FAQ how we go about it.
>
> It is great effort to do what you have done, but we need to go back a bit.
>
> Please post the raw collection of [...]

Anything else I can do to help out here?  Think I've provided everything that was requested.  Thanks!

mrmees, 2024-03-18:
Checking in again.  Would really like to get this put in this year before the hot weather starts.