crankyoldgit/IRremoteESP8266 #2010: When SEND_LG is true and SEND_SAMSUNG is false compilation fails

dilyanpalauzov, 2023-07-08:
… because ir_LG.cpp/IRSend::sendLG contains:
```c++
    // LG 32bit protocol is near identical to Samsung except for repeats.
    sendSAMSUNG(data, nbits, 0);  // Send it as a single Samsung message.
    repeatHeaderMark = kLg32RptHdrMark;
    duty = 33;
    repeat++;
```

crankyoldgit, 2023-07-08:
Thanks for the report. Can you please test our PR #2011 to see if that fixes the issue?

dilyanpalauzov, 2023-07-08:
This is also necessary
```diff
diff --git a/IRremoteESP8266/src/IRsend.h b/IRremoteESP8266/src/IRsend.h
index f8a447197..a778df309 100644
--- a/IRremoteESP8266/src/IRsend.h
+++ b/IRremoteESP8266/src/IRsend.h
@@ -310,7 +310,7 @@ class IRsend {
   void sendSherwood(uint64_t data, uint16_t nbits = kSherwoodBits,
                     uint16_t repeat = kSherwoodMinRepeat);
 #endif
-#if SEND_SAMSUNG
+#if SEND_SAMSUNG || SEND_LG
   void sendSAMSUNG(const uint64_t data, const uint16_t nbits = kSamsungBits,
                    const uint16_t repeat = kNoRepeat);
   uint32_t encodeSAMSUNG(const uint8_t customer, const uint8_t command);
```

crankyoldgit, 2023-07-08:
@dilyanpalauzov  Thanks for that. I totally forgot that bit. I've updated PR #2011. Can you please confirm if the update now covers everything?

dilyanpalauzov, 2023-07-09:
This looks good.  I cannot test the patch 1:1, since I use IRremoteESP8266 within Tasmota, where IRremoteESP8266 is part of the version-controlled code of Tasmota.   In any case with the change I provided above, and you have integrated, the code compiles.

Besides, the parentheses after `#if` are not necessary.

PR #2011 (2023-07-10): Fix compilation dependency of LG on Samsung send protocol
`sendSAMSUNG()` is needed by `sendLG()`, so fix the `#ifdef`s appropriately.

i.e. Compilation shouldn't break if `SEND_LG == TRUE` and `SEND_SAMSUNG == FALSE`

Fixes #2010
commit 5778cbff (2023-07-10): Fix compilation dependency of LG on Samsung send protocol (#2011)

`sendSAMSUNG()` is needed by `sendLG()`, so fix the `#ifdef`s appropriately.

i.e. Compilation shouldn't break if `SEND_LG == TRUE` and `SEND_SAMSUNG == FALSE`

Fixes #2010
closed 2023-07-10 by crankyoldgit