crankyoldgit/IRremoteESP8266 #2012: About Coolix

hassbian-ABC, 2023-07-24:
When coolix receives the infrared code from the remote control, it becomes uncontrollable

hassbian-ABC, 2023-07-24:
Using bin compiled from IRMQTTServer

NiKiZe, 2023-07-24:
Why did you ignore the issue template?
What are you sending to the AC when this happens?
And what does the original remote send when you are trying to do the same action?

You will have to collect this data. With the information provided so far we can't even guess what you are doing and what is happening.

hassbian-ABC, 2023-07-24:
The remote control only controls the air conditioning on, there is nothing extra

NiKiZe, 2023-07-24:
With this issue, what is your expected actions from this library?
Taking in to consideration that we don't know how to reproduce the issue you are seeing.

hassbian-ABC, 2023-07-25:
After receiving the infrared code, Coolix will reset the sensorTemperature to 0.00, which will become uncontrollable

hassbian-ABC, 2023-07-25:
If the sensorTemperature is reset to kNoTempValue, everything becomes normal

crankyoldgit, 2023-07-25:
Can you please supply the sequence of messages that actual remote is sending? e.g.
Using [`IRrecvDumpV3` example code](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRrecvDumpV3) to capture the actual IR messages.

hassbian-ABC, 2023-07-25:
Okay, I will finish it in my free time

crankyoldgit, 2023-07-25:
I've reviewed the relevant code in the library, and nothing jumps out at me. Thus, I'm going to need the actual message (The Coolix 24-bit message code/number) to look any further into this.

hassbian-ABC, 2023-07-25:
Is IRrecvDumpV2 okay?

hassbian-ABC, 2023-07-25:
Is IRrecvDumpV2 okay? @crankyoldgit

hassbian-ABC, 2023-07-25:
https://github.com/crankyoldgit/IRremoteESP8266/issues/821

crankyoldgit, 2023-07-25:
> Is IRrecvDumpV2 okay? @crankyoldgit

Yes.

crankyoldgit, 2023-07-25:
> #821

Not relevant as we need to know the message that YOUR remote is generating, and with the current version of the library.

hassbian-ABC, 2023-07-25:
[新建 文本文档.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/12163183/default.txt)
use IRremoteESP8266-2.8.5

crankyoldgit, 2023-07-26:
> After receiving the infrared code, Coolix will reset the sensorTemperature to 0.00, which will become uncontrollable

Interesting, because my testing it seems to report the sensorTemperature as 31.
Regardless, I think the problem is it's reporting a sensor temp in the `.toCommon()` method when it shouldn't be.

crankyoldgit, 2023-07-26:
@hassbian-ABC Can you please download and test PR #2015 / branch [Issue2012](https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue2012) and confirm if that fixes your problem.

Thanks in advance.

@NiKiZe The continuous build is failing due to what seems to be an update in the Blynk library behaviour. The failing test should be fixed by https://github.com/crankyoldgit/IRremoteESP8266/pull/2016. When approved & merged, I'll rebase and merge etc.

hassbian-ABC, 2023-07-26:
Okay, I'll test it later @crankyoldgit

hassbian-ABC, 2023-07-26:
Everything has changed very well @crankyoldgit

hassbian-ABC, 2023-07-26:
Thank you @crankyoldgit

commit 2e666807 (2023-07-26): [BUG] Ensure `IRCoolixAC::toCommon()` returns `kNoTempValue` when no sensor temp is detected.
It seems we were reporting an incorrect sensor temp (31) which normally means "There is no sensor temp.", instead of `kNoTempValue`.

Fixes #2012
PR #2015 (2023-07-26): [BUG] Ensure `IRCoolixAC::toCommon()` returns `kNoTempValue` correctly.
It seems we were reporting an incorrect sensor temp (31) which normally means "There is no sensor temp.", instead of `kNoTempValue`.

Fixes #2012
commit 1bab0712 (2023-07-26): [BUG] Ensure `IRCoolixAC::toCommon()` returns `kNoTempValue` when no sensor temp is detected. (#2015)

It seems we were reporting an incorrect sensor temp (31) which normally means "There is no sensor temp.", instead of `kNoTempValue`.

Fixes #2012
closed 2023-07-26 by hassbian-ABC