crankyoldgit/IRremoteESP8266 #2013: Irremoteesp8266 and rfrh22 from radiohead

Vorms, 2023-07-25:
I use an esp32.
When I receive an infrared message I have to transmit an rf message, the corresponding node send a message to the node just receiving the infrared message and this not has to send an other message and so on.

I have a lot of crash (timer interrupt) I dont understand what'happend and how to solve it.

crankyoldgit, 2023-07-25:
Can I suggest disabling collection of IR messages when you transmit via RF? Maybe the RF library you are using uses the same hardware timer.
See [documentation](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRrecv.html#a8fe4d26ef1f863db1db9994fed5fc209): https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRrecv.html#a8fe4d26ef1f863db1db9994fed5fc209

e.g.
```c++
IRrecv irrecv(RecvPin,  // GPIO Pin to use
              CaptureBufferSize, //  Size of the capture buffer
              Timeout,  // How long to wait before we give up on a signal
              false,  // Wait for a .resume() call before we start capturing again
              TimerNumber);  // Which hardware timer should we use?

void setup() {
  irrecv.enableIRIn();  // Start up the IR receiver.
}

void loop() {
  // Check if an IR message has been received.
  if (irrecv.decode(&results)) {  // We have captured something.
    // The capture has stopped at this point.
    // Do your RF thing.
    // then
    // Resume capturing IR messages. It was not restarted until after we did the RF
    // message.
    irrecv.resume();
  }
  yield();
}
```
Or see the [DumbIRRepeater.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/DumbIRRepeater/DumbIRRepeater.ino) example code.

Vorms, 2023-07-25:
Hello,
Many thanks for your reply.
It's exactly what I do . I resume after sending the rf message.
I try to add delay before and and after rf send but it doesn't solve the
issue.
Regards
Thierry

crankyoldgit, 2023-07-25:
Do you have a minimum viaable example of your code that you can share that produces the issue/errors/etc?