crankyoldgit/IRremoteESP8266 #2014: About the ESP32-C3 infrared correlation

hzhh110, 2023-07-26:
`#include <Arduino.h>
#include <ETH.h> //引用以使用ETH
#include <WiFiUdp.h>
#include <WiFi.h>
#include <WiFiClient.h>
#include <HTTPClient.h>
#include <WiFi.h>
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRrecv.h>
#include <IRutils.h>
#include <IRtext.h>
#include <IRutils.h>
#include <IRac.h>
#define DEBUG 1
#if DEBUG
// 调试定义
#define DebugBegin(baud_rate) Serial.begin(baud_rate)
#define DebugPrintln(message) Serial.println(message)
#define DebugPrint(message) Serial.print(message)
#define DebugPrintF(...) Serial.printf(__VA_ARGS__)
#define BTSerial Serial1
#else
#define DebugBegin(baud_rate)
#define DebugPrintln(message)
#define DebugPrint(message)
#define DebugPrintF(...)
#define BTSerial Serial1
#endif

const char *ssid = "HUAWEI-10G9TB";

const char *password = "88888888";
const uint16_t kRecvPin = 7;
const uint16_t kCaptureBufferSize = 1024;
const uint8_t kTimeout = 50;
const uint16_t kFrequency = 38000;
IRrecv irrecv(kRecvPin, kCaptureBufferSize, kTimeout, true);
decode_results results;

const uint16_t kIrLedPin = 5;

IRsend irsend(kIrLedPin);
void doWiFiTick();
void linkAll(void *parameter)
{

  while (1)
  {

    doWiFiTick();
    vTaskDelay(50);
  }
  vTaskDelete(NULL);
}
void doWiFiTick()
{
  if (WiFi.status() == WL_CONNECTED)
  {
    return;
  }
  wifi_mode_t mode = WiFi.getMode();
  DebugPrintln("mode:");
  DebugPrintln(mode);
  WiFi.mode(WIFI_STA);
  WiFi.disconnect(false, true);
  delay(100);
  wl_status_t status = WiFi.begin(ssid, password);
  DebugPrintln("status:");
  DebugPrintln(status);
[... 184 lines cut ...]
  return true;
}

uint16_t rawData[279] = {9068, 4450, 734, 1484, 728, 464, 756, 438, 724, 468, 730, 1508, 700, 1486, 728, 464, 732, 460, 758, 470, 648, 1530, 760, 430, 756, 1458, 734, 490, 696, 466, 728, 464, 726, 466, 716, 474, 716, 480, 752, 440, 752, 438, 730, 464, 726, 1486, 724, 466, 734, 460, 652, 540, 730, 462, 754, 438, 734, 462, 756, 1454, 654, 536, 728, 1486, 712, 550, 698, 466, 756, 1456, 728, 464, 752, 20060, 730, 464, 728, 466, 730, 466, 650, 540, 756, 438, 652, 540, 730, 464, 710, 482, 752, 444, 754, 438, 728, 466, 756, 438, 732, 494, 730, 436, 732, 458, 656, 538, 756, 438, 728, 466, 730, 464, 752, 442, 724, 466, 754, 438, 754, 440, 752, 440, 724, 468, 724, 464, 758, 436, 730, 462, 758, 1456, 652, 540, 730, 1482, 756, 436, 734, 39994, 9082, 4438, 732, 1510, 730, 432, 654, 540, 726, 496, 700, 1512, 686, 1524, 700, 464, 726, 466, 726, 464, 734, 1476, 734, 460, 744, 1468, 730, 492, 700, 462, 760, 434, 754, 438, 756, 466, 694, 468, 746, 444, 734, 460, 728, 470, 744, 1466, 756, 438, 764, 430, 752, 440, 730, 464, 734, 486, 700, 464, 756, 1458, 728, 1486, 726, 1512, 694, 476, 730, 466, 730, 1480, 728, 460, 682, 19840, 762, 434, 756, 436, 652, 540, 656, 536, 730, 464, 730, 462, 758, 436, 738, 454, 758, 438, 712, 478, 728, 462, 760, 462, 730, 432, 656, 538, 728, 492, 730, 438, 732, 464, 728, 462, 736, 458, 652, 540, 730, 468, 728, 488, 728, 1486, 700, 464, 728, 464, 726, 466, 756, 432, 760, 464, 726, 1486, 732, 438, 648, 540, 758, 1454, 728}; // KELVINATOR
uint8_t state[16] = {0x31, 0x0A, 0x20, 0x50, 0x00, 0x00, 0x00, 0x50, 0x31, 0x0A, 0x20, 0x70, 0x00, 0x00, 0x40, 0x90};                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                          // uint8_t state[16] = {0x39, 0x0A, 0x20, 0x50, 0x00, 0x00, 0x00, 0xD0, 0x39, 0x0A, 0x20, 0x70, 0x00, 0x00, 0x40, 0x10};
void loop()
{

  irsend.sendRaw(rawData, 279, 38);
  loopCheckIR();
  delay(2000);
  int bits = irsend.defaultBits(NEC);
  DebugPrintln("bits");
  DebugPrintln(bits);

  irsend.send(NEC, 0xDDDD, bits);

  loopCheckIR();

  delay(2000);
  loopCheckIR();
}
`

When wifi and Bluetooth are running at the same time, the data transmitted by the infrared function is very unstable, and in many cases the air conditioner can not resolve it, and when the wifi connection is closed, it is very stable, and the air conditioner can receive the signal stably.

hzhh110, 2023-07-26:
[image]
[image]
[image]
[image]

hzhh110, 2023-07-26:
[image]
When the wifi is turned on, it often cannot be used normally. If the wifi Bluetooth is turned on, it cannot be controlled. If it is integrated into a large project, it is also difficult to control.I feel that the data will be seriously deviated, but if the wifi is turned off, it will be normal.

hzhh110, 2023-07-26:
[env:esp32-c3-devkitm-1]
platform = espressif32 @^5.2.0
board = esp32-c3-devkitm-1
board_build.flash_mode = dio
framework = arduino
lib_deps = 
	crankyoldgit/IRremoteESP8266 @^2.8.5
build_flags =
  -D MQTT_MAX_PACKET_SIZE=1024
  -D MQTT_KEEPALIVE=60
board_build.partitions = esp32cam2.csv

hzhh110, 2023-07-26:
[image]
Especially App.When communicating, it is basically inaccurate.

hassbian-ABC, 2023-07-26:
Using IRMQTTServer on c3 yields the same structure, with air conditioning not receiving signals most of the time. Library version 2.8.5

hzhh110, 2023-07-27:
> Using IRMQTTServer on c3 yields the same structure, with air conditioning not receiving signals most of the time. Library version 2.8.5

Do you have the same problem? Is there any way to fix this?

hassbian-ABC, 2023-07-27:
Yes, I have encountered the same problem, but there is no way to solve it @hzhh110

crankyoldgit, 2023-07-27:
It seems the only way to solve it on that architecture is for us to re-write the sending routines to use the [`rmt` feature/library](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/rmt.html).
The library doesn't use or support that at present.

hzhh110, 2023-07-27:
> It seems the only way to solve it on that architecture is for us to re-write the sending routines to use the [`rmt` feature/library](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/rmt.html).
> The library doesn't use or support that at present.

Thank you for the boss's reply, or the boss inside, at least there is a solution to the problem, I am afraid that the structure can not solve the problem directly?I wonder if there are any plans to adapt to the scheme this year?I don't know if the adaptation is complicated?

crankyoldgit, 2023-07-27:
It is something I want to do, but I just don't have the spare time or have a great enough need at present.
I think there have been attemps/forks to add `rmt` support, but I have not tracked their progress.

Potentially look at: @tuxBurner 's [`feature/ESP32_RMT`](https://github.com/tuxBurner/IRremoteESP8266/tree/feature/ESP32_RMT) branch.

tuxBurner, 2023-07-27:
Yes @crankyoldgit this fixes ble / wifi problems when usin infrared. Becaus the timing is not done on the cpu directly. Instead it is done in the RMT part. Downside was that the code is getting bigger because of the RMT part. But perhaps that changed since i started the Fork. I have to finish it and make a PR to the main repo.

hassbian-ABC, 2023-07-27:
I use [feature/ESP32_RMT](https://github.com/tuxBurner/IRremoteESP8266/tree/feature/ESP32_RMT) on C3， when use RMT_CHANNEL_2，I received this error prompt 
[image]

when use RMT_CHANNEL_0 or RMT_CHANNEL_1，I received this error prompt 

[image]

SDK 4.4.4

hassbian-ABC, 2023-07-27:
How can I fix this error @tuxBurner

tuxBurner, 2023-07-27:
@hassbian-ABC i am still at work i have to take a look at home

hzhh110, 2023-08-01:
> @hassbian-ABC i am still at work i have to take a look at home

uint16_t rawData[] = {9068, 4450, 734, 1484, 728, 464, 756, 438, 724, 468, 730, 1508, 700, 1486, 728, 464, 732, 460, 758, 470, 648, 1530, 760, 430, 756, 1458, 734, 490, 696, 466, 728, 464, 726, 466, 716, 474, 716, 480, 752, 440, 752, 438, 730, 464, 726, 1486, 724, 466, 734, 460, 652, 540, 730, 462, 754, 438, 734, 462, 756, 1454, 654, 536, 728, 1486, 712, 550, 698, 466, 756, 1456, 728, 464, 752, 20060, 730, 464, 728, 466, 730, 466, 650, 540, 756, 438, 652, 540, 730, 464, 710, 482, 752, 444, 754, 438, 728, 466, 756, 438, 732, 494, 730, 436, 732, 458, 656, 538, 756, 438, 728, 466, 730, 464, 752, 442, 724, 466, 754, 438, 754, 440, 752, 440, 724, 468, 724, 464, 758, 436, 730, 462, 758, 1456, 652, 540, 730, 1482, 756, 436, 734, 39994, 9082, 4438, 732, 1510, 730, 432, 654, 540, 726, 496, 700, 1512, 686, 1524, 700, 464, 726, 466, 726, 464, 734, 1476, 734, 460, 744, 1468, 730, 492, 700, 462, 760, 434, 754, 438, 756, 466, 694, 468, 746, 444, 734, 460, 728, 470, 744, 1466, 756, 438, 764, 430, 752, 440, 730, 464, 734, 486, 700, 464, 756, 1458, 728, 1486, 726, 1512, 694, 476, 730, 466, 730, 1480, 728, 460, 682, 19840, 762, 434, 756, 436, 652, 540, 656, 536, 730, 464, 730, 462, 758, 436, 738, 454, 758, 438, 712, 478, 728, 462, 760, 462, 730, 432, 656, 538, 728, 492, 730, 438, 732, 464, 728, 462, 736, 458, 652, 540, 730, 468, 728, 488, 728, 1486, 700, 464, 728, 464, 726, 466, 756, 432, 760, 464, 726, 1486, 732, 438, 648, 540, 758, 1454, 728}; // KELVINATOR

irsend.sendRaw(rawData, sizeof(rawData)/2, 38);

Similar Raw for KELVINATOR cannot be controlled
for void IRsend::sendRaw(const uint16_t buf[], const uint16_t len,
                     const uint16_t hz) {
if(len % 2 != 0) { 
   return;
}

crankyoldgit, 2023-08-01:
> irsend.sendRaw(rawData, sizeof(rawData)/2, 38);

First of all, I'm assuming the issues (crashes/failures) you are having are with the fork, not the release branch.

I'm not sure your code `sizeof(rawData)/2` does what you think it does. It probably does, but you might want to check that.

hzhh110, 2023-08-02:
@tuxBurner 's [feature/ESP32_RMT](https://github.com/tuxBurner/IRremoteESP8266/tree/feature/ESP32_RMT) branch.
for ac and for 
for void IRsend::sendRaw(const uint16_t buf[], const uint16_t len,
  const uint16_t hz) {
  if(len % 2 != 0) {
  return;
}

hzhh110, 2023-08-02:
@tuxBurner 's [feature/ESP32_RMT](https://github.com/tuxBurner/IRremoteESP8266/tree/feature/ESP32_RMT) branch.
for ac sending crashes  and for
for void IRsend::sendRaw(const uint16_t buf[], const uint16_t len,
const uint16_t hz) {
if(len % 2 != 0) {
return;
}

hzhh110, 2023-08-11:
> this

That's weird. Did you get any luck with that? for rmt mode

tuxBurner, 2023-09-11:
Sorry for the late Response i need to order a s2 to test the bug

hipal31, 2023-11-22:
@hzhh110 @crankyoldgit @tuxBurner Does anyone have solution of this?
I'm using this library in ESP-IDF V4.4 and using BLE-MESH as well.

tuxBurner, 2023-11-23:
OMG i totally forgot this.

To much to do at work :) 

But i ordered the parts and will check this at the weekend.

mrninhvn, 2024-08-08:
Building on the work initiated by @tuxBurner, I made a patch using the RMT implementation and Arduino Release v3.0.4.

https://github.com/mrninhvn/IRremoteESP/releases/tag/v2.8.6-rmt

It will need to be tested thoroughly to make sure all protocols work, welcome to try it out (Note: Only Tested with Platformio with ESP32-C3).

mistic100, 2024-10-07:
@mrninhvn just to let you know, this line does not compile
https://github.com/mrninhvn/IRremoteESP/blob/8f941e66e4e6689e34ad372db22b5c20cbb9707e/src/IRrecv.cpp#L499
I added to surround it with `#ifndef ESP32_RMT`

Beside that your fork works perfectly on the Seeeduino Xiao with Wifi enabled, thank you very much

I would be could to clean it up, perhaps remove any IDF related stuff, and make a pull request.