crankyoldgit/IRremoteESP8266 #2025: ESP32 S2 and Wifi = AC code sending interrupted

Insomniac80, 2023-08-17:
### Version/revision of the library used is 2.8.6

I have two IR leds in a row driven by a transistor to remote control my Technibel (rebranded) Air Conditioner.
ESP32 is a S2 mini.
Changing a sample code accordingly works fine, every command is recognized by the AC.

But as soon as I add also some WLan functionality (cannot post the code here as it is my multi class MQTT / WLan lib I use for all controllers) most IR codes I send are not recognized by the AC. My wlan code is fully single task written, sending cannot be interrupted by it. The cause must be under the hood.

Have now switched wo a D1 Mini ESP32 (dual core) with the same code (compiled for the new chip for sure) and using the same circuit, and now also with Wifi every command is accepted by the AC. Works like a charm.

My guess is that the single core of the S2 is having problems, and wlan (even If nothing is recived) interrupts most of the IR output, at least when sending longer sequences as the ones of the Technibel ACs.
Just wanted to mention it here, just for the case someone also thinks about using a S2 for it.

Btw, I was really impressed to directly find code for my AC which is so super easy to use and great documented! Thank you for this awesome library!

crankyoldgit, 2023-08-20:
> My guess is that the single core of the S2 is having problems, and wlan (even If nothing is recived) interrupts most of the IR output, at least when sending longer sequences as the ones of the [...]

Sadly, that's my belief too. Which is surprising as the ESP8266 is single core and behaves fine.
Until someone gets the library using RMT for sending, this issue will percist.

Ref #1922 & https://github.com/crankyoldgit/IRremoteESP8266/issues/2014#issuecomment-1653121814

fjulian79, 2024-02-22:
Damm .. same here. I have prototyped everything with a ESP32 Dev Kit C and my last step was to build the final HW around the S2 mini --> Epic fail! I which I would have seen this issue earlier. However, at least it's documented here. I will rebuild my final Hardware now and next time I will check the already known issues earlier.

Famous last words .. this lib is great, specially the Webserver example! Thanks a lot!! :-)