crankyoldgit/IRremoteESP8266 #2026: Panasonic CS-E12QKEW

duczz, 2023-08-28:
Hello,

you can add the "Panasonic CS-E12QKEW" to the compatibility List, its working fine with following Settings

Tasmota Log (in older tasmota versions "Model":"DKE" shows Model":3):
`14:10:29.104 MQT: stat/Klimaanlage_Buero/RESULT = {"IRHVAC":{"Vendor":"PANASONIC_AC","Model":"DKE","Command":"Control","Mode":"Auto","Power":"Off","Celsius":"On","Temp":22,"FanSpeed":"Min","SwingV":"High","SwingH":"Middle","Quiet":"Off","Turbo":"On","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1,"Clock":910,"iFeel":"Off","SensorTemp":"n/a"}}
`

Homeasssitant Settings (i think "hvac_model: 3" ist only the important thing):
```
#Büro
  - platform: tasmota_irhvac
    name: "Klimaanlage Buero"
    command_topic: "cmnd/Klimaanlage_Buero/irhvac"
    # Pick one of the following:
    # State is updated when the tasmota device receives an IR signal (includes own transmission and original remote)
    # useful when a normal remote is in use alongside the tasmota device, may be less reliable than the second option.
    #state_topic: "tele/Klimaanlage_Buero/RESULT"
    # State is updated when the tasmota device completes IR transmission, should be pretty reliable.
    #state_topic: "stat/your_tasmota_device/RESULT"
    # Uncomment if your 'available topic' of Tasmota IR device are different (if device in HA is disabled)
    #availability_topic: "tele/your_tasmota_device/LWT"
    temperature_sensor: sensor.wohnzimmer_temperatursensor_snzb_02_temperature
    humidity_sensor: sensor.wohnzimmer_temperatursensor_snzb_02_humidity #optional - default None
    #power_sensor: binaly_sensor.kitchen_ac_power #optional - default None
    vendor: "PANASONIC_AC"
    min_temp: 16 #optional - default 16 int value
    max_temp: 30 #optional - default 32 int value
    target_temp: 26 #optional - default 26 int value
    initial_operation_mode: "off" # optional - default "off" string value (one of the "supported_modes")
    #away_temp: 24 #optional - default 24 int value (preset_modes)
    precision: 1 #optional - default 1 int or float value. Can be set to 1, 0.5 or 0.1
    supported_modes:
      - "auto"
      - "heat"
      - "cool"
      - "dry"
      - "fan_only" # Use "fan_only" even if Tasmota shows "Mode":"Fan"
      - "off" #Turns the AC off - Should be in quotes
      # Some devices have "auto" and "fan_only" switched. If the following two lines are uncommented, "auto" and "fan" shoud be commented out
      #- "auto_fan_only" #if remote shows fan but tasmota says auto
      #- "fan_only_auto" #if remote shows auto but tasmota says fan
    supported_fan_speeds:
      # Some devices say max,but it is high, and auto which is max. If you uncomment the following two, you have to comment high and max
      # - "auto_max" #woud become max
      # - "max_high" #would become high
      #- "on"
      #- "off"
      - "auto"
      - "min"
      - "low"
      - "medium"
      - "high"
      - "max"
    supported_swing_list:
      - "off"
      - "vertical" #up to down
      - "horizontal" # Left to right
      #- "both"
    default_quiet_mode: "Off" #Quiet und Turbo sind in der Firmware vertauscht | optional - default "Off" string value
    default_turbo_mode: "On" #Quiet und Turbo sind in der Firmware vertauscht | optional - default "Off" string value
    default_econo_mode: "Off" #optional - default "Off" string value
    hvac_model: "3" #optional - default "1" string value
    celsius_mode: "On" #optional - default "On" string value
    default_light_mode: "Off" #optional - default "Off" string value
    default_filter_mode: "Off" #optional - default "Off" string value
    default_clean_mode: "Off" #optional - default "Off" string value
    default_beep_mode: "Off" #optional - default "Off" string value
    default_sleep_mode: "-1" #optional - default "-1" string value
    default_swingv: "high" #optional - default "" string value
    default_swingh: "middle" #optional - default "" string value 
    keep_mode_when_off: True #optional - default False boolean value : Must be True for MITSUBISHI_AC, ECOCLIM, etc.
```

NiKiZe, 2023-08-28:
Thanks for the report, do you also have a model of the remote?
I would like to make it clear that there is a difference between "works with" and "is using" the later is preferred but requires actually dumps of what the original remote sends to confirm.

duczz, 2023-08-29:
Sure here is the original remote code:
`
11:59:29.472 MQT: tele/Klimaanlage_Buero/RESULT = {"IrReceived":{"Protocol":"PANASONIC_AC","Bits":216,"Data":"0x0220E004000000060220E00400092C805106000EE0000081000081","Repeat":0,"IRHVAC":{"Vendor":"PANASONIC_AC","Model":"NKE","Command":"Control","Mode":"Auto","Power":"On","Celsius":"On","Temp":22,"FanSpeed":"Medium","SwingV":"Highest","SwingH":"Middle","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1,"iFeel":"Off","SensorTemp":"n/a"}}}
`

Information:
With your project Model 0 is working, but no swing and Econo is possible, all other features are working
With your project Model 3 is only Econo no working, all other features are working (RollingKey maybe?)

"Econo On" original remote code:
12:03:57.755 MQT: tele/Klimaanlage_Buero/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":486,"Hash":"0x9F9E8B11","Repeat":0}}

crankyoldgit, 2023-08-29:
> "Econo On" original remote code:
> 12:03:57.755 MQT: tele/Klimaanlage_Buero/RESULT = {"IrReceived":{"Protocol":"UNKNOWN","Bits":486,"Hash":"0x9F9E8B11","Repeat":0}}

FYI: This bit doesn't help us at all. See: [https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#my-remotes-signal-is-being-reported-as-an-unknown-code-what-does-that-mean-how-do-i-send-an-unknown-code](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#my-remotes-signal-is-being-reported-as-an-unknown-code-what-does-that-mean-how-do-i-send-an-unknown-code)