crankyoldgit/IRremoteESP8266 #2027: Help Needed (re: Mixiaoxiao / ESP8266-IR-HOMEKIT)

loyukfai, 2023-08-28:
I bought a pre-built WiFi IR module using the below codes. It doesn't work with my AC (Hitachi RAS-E13CAK). Would like to update the library but no idea how, any pointers would be much appreciated!

https://github.com/Mixiaoxiao/ESP8266-IR-HOMEKIT

NiKiZe, 2023-08-28:
There is a issue template that you should fill in when creating issues, why did you skip that?

That template both give you guidance on how. But also collects data that we need to help you. 

You can manually copy the template from https://github.com/crankyoldgit/IRremoteESP8266/blob/master/.github/ISSUE_TEMPLATE/problem-report.md?plain=1 read it carefully and fill in the requested data.

loyukfai, 2023-08-28:
Sorry! I'll do it again!

closed 2023-08-28 by loyukfai