crankyoldgit/IRremoteESP8266 #2037: Mitsubishi Heavy 88-bit A/C - Fan and Mode cannot be set

kodjo80, 2023-09-27:
### Version/revision of the library used
### _Typically located in the `library.json` & `src/IRremoteESP8266.h` files in the root directory of the library.
v2.8.6,  'master' 
I am using Wemos D1 Mini v 4.0
My AC is Mitsubishi Heavy DXK12Z6-W with remote RKX502A007P

### Describe the bug
When I try to set the operation mode or the fan the settings were not changed accordingly.
When I change the operation mode with kMitsubishiHeavyAuto, kMitsubishiHeavyHeat, kMitsubishiHeavyCool, kMitsubishiHeavyDry, kMitsubishiHeavyFan the resulting code did.t change as it should.
In the summary of the sent message the mode is not changed. I compared the codes with the IRrecvDump_v3 and they don't match.
Here comes the interesting:
When I change the fan mode with kMitsubishiHeavy88Fan/Auto, High, Med, Low, Econo, Turbo/ they change the operations mode /Heat, Cool, Dry, Auto/

I cannot check what mode is set on the A/C unit, because there is no indicators on it. I can only compare the remote codes with the ones produced by the library, but when I send a code with the ESP the fan is always on High or Auto /I suppose/.

I tried SmartIRRepeater, and it seems to work as expected, so I assume my hardware is setup correctly.

Best wishes,
Venelin

crankyoldgit, 2023-09-27:
Can you please provide the minimal working code example that demonstrates the bug/your issue please?
e.g.
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/TurnOnMitsubishiHeavyAc/TurnOnMitsubishiHeavyAc.ino

kodjo80, 2023-09-28:
Hm...
it seems that with the TurnOnMitsubishiHeavyAc.ino example code the codes are produced exactly as they should.
Maybe my implementation is not working properly.
I will continue with debugging.

I am baffled how did you manage to compile such a complex library. Thank you for your work, and sorry for wasting your time.

Best wishes
Venelin.

crankyoldgit, 2023-09-28:
> it seems that with the TurnOnMitsubishiHeavyAc.ino example code the codes are produced exactly as they should.

Did you change the code appropriately to `88` and not the `152` stuff? e.g. All the constants and the object declaration/instantiation.
e.g.
https://github.com/crankyoldgit/IRremoteESP8266/blob/a295f87e067bc8ccb22523ef035a05ea11d65e49/examples/TurnOnMitsubishiHeavyAc/TurnOnMitsubishiHeavyAc.ino#L32

> Maybe my implementation is not working properly.

That is quite likely.

Have you tried replaying the `rawData` using `sendRaw` See:
e.g.
```c++
uint16_t rawData[179] = {3228, 1572, 446, 350, 420, 1186, 444, 350, 420, 382, 418, 1184, 446, 350, 420, 1184, 444, 350, 420, 382, 418, 1184, 446, 1152, 446, 1154, 446, 348, 420, 1184, 446, 350, 420, 1184, 446, 1154, 444, 1154, 446, 348, 420, 382, 418, 382, 418, 382, 418, 1184, 446, 1154, 446, 350, 420, 1184, 446, 1154, 444, 350, 420, 382, 418, 1184, 444, 348, 420, 382, 418, 1182, 446, 350, 420, 382, 420, 1182, 446, 1154, 446, 350, 420, 1186, 444, 1154, 446, 1154, 444, 352, 420, 1184, 444, 1154, 422, 1178, 444, 1152, 424, 1176, 422, 1176, 424, 374, 418, 1184, 422, 376, 418, 380, 418, 382, 418, 384, 418, 382, 418, 382, 418, 1182, 422, 1178, 422, 1178, 422, 1178, 422, 376, 418, 382, 416, 382, 418, 384, 418, 382, 416, 384, 418, 380, 418, 382, 418, 1184, 422, 1178, 422, 1176, 422, 1178, 422, 376, 418, 1182, 422, 1178, 422, 376, 416, 382, 418, 1184, 422, 1178, 422, 376, 416, 1182, 422, 376, 418, 382, 418, 1182, 424, 1176, 424, 374, 418, 380, 418, 1184, 426}; // MITSUBISHI_HEAVY_88
IRsend irsend(IR_LED_GPIO);
irsend.sendRaw(rawData, 179, 38);  // Send a raw data capture at 38kHz.
```
and `state` via [`setRaw()`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRMitsubishiHeavy88Ac.html#abf01e448da9ec6e3b4512f58c3020299) & [`send();`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRMitsubishiHeavy88Ac.html#a707cb3ec3e3c18bedeb12205580d5048)
```c++
IRMitsubishiHeavy88Ac ac(IR_LED_GPIO);
uint8_t state[11] = {0xAD, 0x51, 0x3C, 0xD9, 0x26, 0x02, 0xFD, 0xF0, 0x0F, 0x99, 0x66};

ac.setRaw(state);
ac.send();
```

> I am baffled how did you manage to compile such a complex library.

It takes some effort and automation. ;-)