crankyoldgit/IRremoteESP8266 #2039: Feature Request: Support for esp32 Arduino 3.0.0

Jason2866, 2023-10-06:
Espressif Arduino 3.0.0 alpha for ESP32, ESP32-S2, ESP32-S3, ES32C3 and ESP32-C6 is released.
IRremoteESP8266 does not work with.

NiKiZe, 2023-10-06:
Can you be more specific? What exactly don't work?

Jason2866, 2023-10-07:
Trying to compile `IRrecvDumpV3` with Espressif Arduino 3.0.0 alpha is generating this compile errors
```
/Volumes/T7-Mac/IRremoteESP8266/src/IRrecv.cpp: In function 'void gpio_intr()':
/Volumes/T7-Mac/IRremoteESP8266/src/IRrecv.cpp:218:10: warning: '++' expression of 'volatile'-qualified type is deprecated [-Wvolatile]
  218 |   params.rawlen++;
      |   ~~~~~~~^~~~~~
/Volumes/T7-Mac/IRremoteESP8266/src/IRrecv.cpp:246:3: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
  246 |   timerAlarmEnable(timer);
      |   ^~~~~~~~~~~~~~~~
      |   timerAlarm
/Volumes/T7-Mac/IRremoteESP8266/src/IRrecv.cpp: In member function 'void IRrecv::enableIRIn(bool)':
/Volumes/T7-Mac/IRremoteESP8266/src/IRrecv.cpp:362:21: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
  362 |   timer = timerBegin(_timer_num, 80, true);
      |           ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/hans/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:84,
                 from /Users/hans/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,
                 from /Volumes/T7-Mac/IRremoteESP8266/src/IRrecv.h:10,
                 from /Volumes/T7-Mac/IRremoteESP8266/src/IRrecv.cpp:6:
/Users/hans/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-timer.h:35:14: note: declared here
   35 | hw_timer_t * timerBegin(uint32_t frequency);
      |              ^~~~~~~~~~
/Volumes/T7-Mac/IRremoteESP8266/src/IRrecv.cpp:371:3: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
  371 |   timerAlarmWrite(timer, MS_TO_USEC(params.timeout), ONCE);
      |   ^~~~~~~~~~~~~~~
      |   timerWrite
/Volumes/T7-Mac/IRremoteESP8266/src/IRrecv.cpp:375:23: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
  375 |   timerAttachInterrupt(timer, &read_timeout, false);
      |   ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/hans/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t * timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
Compiling .pio/build/esp32dev/lib0e3/src/IRutils.cpp.o
/Volumes/T7-Mac/IRremoteESP8266/src/IRrecv.cpp: In member function 'void IRrecv::disableIRIn()':
/Volumes/T7-Mac/IRremoteESP8266/src/IRrecv.cpp:401:3: error: 'timerAlarmDisable' was not declared in this scope
  401 |   timerAlarmDisable(timer);
      |   ^~~~~~~~~~~~~~~~~
/Volumes/T7-Mac/IRremoteESP8266/src/IRrecv.cpp: In member function 'void IRrecv::pause()':
/Volumes/T7-Mac/IRremoteESP8266/src/IRrecv.cpp:416:3: error: 'gpio_intr_disable' was not declared in this scope; did you mean 'esp_intr_disable'?
  416 |   gpio_intr_disable((gpio_num_t)params.recvpin);
      |   ^~~~~~~~~~~~~~~~~
      |   esp_intr_disable
/Volumes/T7-Mac/IRremoteESP8266/src/IRrecv.cpp: In member function 'void IRrecv::resume()':
/Volumes/T7-Mac/IRremoteESP8266/src/IRrecv.cpp:429:3: error: 'timerAlarmDisable' was not declared in this scope
  429 |   timerAlarmDisable(timer);
      |   ^~~~~~~~~~~~~~~~~
Archiving .pio/build/esp32dev/libcbb/libESPmDNS.a
/Volumes/T7-Mac/IRremoteESP8266/src/IRrecv.cpp:430:3: error: 'gpio_intr_enable' was not declared in this scope; did you mean 'esp_intr_enable'?
  430 |   gpio_intr_enable((gpio_num_t)params.recvpin);
      |   ^~~~~~~~~~~~~~~~
      |   esp_intr_enable
/Volumes/T7-Mac/IRremoteESP8266/src/IRrecv.cpp: In member function 'void IRrecv::crudeNoiseFilter(decode_results*, uint16_t)':
/Volumes/T7-Mac/IRremoteESP8266/src/IRrecv.cpp:506:37: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
  506 |         results->rawbuf[offset - 1] += addition;
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
Indexing .pio/build/esp32dev/libcbb/libESPmDNS.a
*** [.pio/build/esp32dev/lib0e3/src/IRrecv.cpp.o] Error 1
```

Jason2866, 2023-10-07:
I made a Platform to use Arduino 3.0.0 alpha1 with Platformio. Just use this for esp32 devices
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/2023.10.10/platform-espressif32.zip
```

tonhuisman, 2023-10-07:
The compilation issues above are already solved in the Tasmota `development` branch and I'm working on patching these in #2040

Not yet testing with hardware, just 'making it work' first 😅

tonhuisman, 2023-10-07:
The `gpio_intr_enable` and `gpio_intr_disable` issues are with the 2023.10.10 platform release, and don't occur when using 2023.10.03.

Jason2866, 2023-10-07:
@tonhuisman Platform 2023.10.10 uses the code base of Arduino 3.0.0 alpha1 release.
Every other(platform) builds before are not relevant. The are WiP builds and do not align with anything official.

Jason2866, 2023-10-07:
@tonhuisman Now the fun starts, it does compile now with your open PR. Try to receive something.

tonhuisman, 2023-10-07:
> Now the fun starts

Not sure if that's to blame on my work, as I just copied the mods made at Tasmota 😄 
It crashes at calling `timerAlarm`, so I'd expect that to cash on Tasmota as well when using Arduinoe 3.0, but as Tasmota is still using Arduino 2,0.14 it'll probably work as before...

Jason2866, 2023-10-07:
It crashes with Tasmota too. Earlier WiP builds (before alpha1) did not crash. 
Don't know if the introduced a bug or more changes where done which needs adopting.

s-hadinger, 2023-11-01:
I have fixed the crash and the timer issues here https://github.com/arendst/Tasmota/pull/19904

tonhuisman, 2023-11-01:
> I have fixed the crash and the timer issues here [arendst/Tasmota#19904](https://github.com/arendst/Tasmota/pull/19904)

Thanks!
I'll apply those changes in my PR soon.

avinashboy, 2024-05-30:
I'm also getting the same error on the ESP32 38-pin Development Board. Is there any update regarding this issue? What should I do? Could you please guide me on this problem?

s-hadinger, 2024-05-30:
We have made all the fixes in Tasmota but I lost track of which ones were backported here

tonhuisman, 2024-05-30:
@avinashboy Are you compiling #2040 with IDF 5.1? As only that branch (PR) is adjusted and (somewhat) tested with IDF 5.1, the main 2.8.6 release doesn't have that (yet).

Zhentao-Lin, 2024-05-31:
I am a teacher, 4 days ago, arduino-esp32 was officially updated to version 3.0.0 Release, then I found that IRremoteESP8266 compiler error, currently I can only let students use arduino-esp32 version 2.0.17. Will you update the library in the future to make it compatible with 3.0.0？Wish you a happy life.

avinashboy, 2024-05-31:
> @avinashboy Are you compiling #2040 with IDF 5.1? As only that branch (PR) is adjusted and (somewhat) tested with IDF 5.1, the main 2.8.6 release doesn't have that (yet).

Yes, I do

Zhentao-Lin, 2024-05-31:
Thank you very much. Looking forward to your update.

avinashboy, 2024-05-31:
> arduino-esp32

Thank you so much. It works if we downgrade the ESP32 board manager to 2.0.13.

avinashboy, 2024-05-31:
> > arduino-esp32
> 
> Thank you so much. It works if we downgrade the ESP32 board manager to 2.0.13.

avinashboy, 2024-05-31:
> I am a teacher, 4 days ago, arduino-esp32 was officially updated to version 3.0.0 Release, then I found that IRremoteESP8266 compiler error, currently I can only let students use arduino-esp32 [...]

Thank you so much. It works if we downgrade the ESP32 board manager to 2.0.13.

phphengenharia, 2024-06-25:
Hi Guys! First at all, congratulations by IRremoteESP8266! 

By the way, I tried esp32-arduino-libs-3.0.2@idf-release_v5.1 and I would like you let know the compile problems:

Using library IRremoteESP8266 version 2.8.6 by David Conran, Mark Szabo, Sebastien Warin, Roi Dayan, Massimiliano Pinto, Christian Nilsson in folder "C:\Users\phmota\Documents\Arduino\libraries\IRremoteESP8266"
"c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\tools\esp-x32\2302/bin/xtensa-esp32-elf-g++" -MMD -c "@c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32/flags/cpp_flags" -w -Os -DF_CPU=240000000L -DARDUINO=108010 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DARDUINO_PARTITION_default "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32:UploadSpeed=921600,CPUFreq=240,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=none,PSRAM=disabled,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_CDC_ON_BOOT=0 "@c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32/flags/defines" "-ID:\Google Drive\PROGRAMAÇÃO\Arduino\3 - WiFi\ESP32 - DS18B20 PIR Fancoletes" -iprefix "c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32/include/" "@c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32/flags/includes" "-Ic:\Users\phmota\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32/qio_qspi/include" -I"c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\cores\esp32" -I"c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\variants\esp32" -I"C:\Users\phmota\Documents\Arduino\libraries\OneWire" -I"C:\Users\phmota\Documents\Arduino\libraries\DallasTemperature" -I"c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\WiFi\src" -I"c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\Network\src" -I"c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\HTTPClient\src" -I"c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\NetworkClientSecure\src" -I"C:\Users\phmota\Documents\Arduino\libraries\NTPClient" -I"C:\Users\phmota\Documents\Arduino\libraries\OneWire\utility" -I"C:\Users\phmota\Documents\Arduino\libraries\DallasTemperature\utility" -I"C:\Users\phmota\Documents\Arduino\libraries\NTPClient\utility" -I"C:\Users\phmota\Documents\Arduino\libraries\IRremoteESP8266\src" -I"c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\EEPROM\src" -I"C:\Users\phmota\Documents\Arduino\libraries\WebServer\src" -I"c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\FS\src" "@C:\Users\phmota\AppData\Local\Temp\VMBuilds\Fancoletes ESP32\esp32_esp32\Release/build_opt.h" "@C:\Users\phmota\AppData\Local\Temp\VMBuilds\Fancoletes ESP32\esp32_esp32\Release/file_opts" "C:\Users\phmota\Documents\Arduino\libraries\IRremoteESP8266\src\IRac.cpp" -o "C:\Users\phmota\AppData\Local\Temp\VMBuilds\Fancoletes ESP32\esp32_esp32\Release\IRremoteESP8266\IRac.cpp.o"
"c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\tools\esp-x32\2302/bin/xtensa-esp32-elf-g++" -MMD -c "@c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32/flags/cpp_flags" -w -Os -DF_CPU=240000000L -DARDUINO=108010 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DARDUINO_PARTITION_default "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32:UploadSpeed=921600,CPUFreq=240,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=none,PSRAM=disabled,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_CDC_ON_BOOT=0 "@c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32/flags/defines" "-ID:\Google Drive\PROGRAMAÇÃO\Arduino\3 - WiFi\ESP32 - DS18B20 PIR Fancoletes" -iprefix "c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32/include/" "@c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32/flags/includes" "-Ic:\Users\phmota\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32/qio_qspi/include" -I"c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\cores\esp32" -I"c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\variants\esp32" -I"C:\Users\phmota\Documents\Arduino\libraries\OneWire" -I"C:\Users\phmota\Documents\Arduino\libraries\DallasTemperature" -I"c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\WiFi\src" -I"c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\libraries\Network\src" -I"c:\Users\phmota\AppData\Local\Arduino15\packages\esp32\hardware\esp32 [... cut ...]
   416 |   gpio_intr_disable((gpio_num_t)params.recvpin)
   |   ^~~~~~~~~~~~~~~~~
   |   esp_intr_disable
irrecv.cpp: In member function void IRrecv::resume()
 
irrecv.cpp: 429:3: error: 'timerAlarmDisable' was not declared in this scope
   429 |   timerAlarmDisable(timer)
   |   ^~~~~~~~~~~~~~~~~
 
irrecv.cpp: 430:3: error: 'gpio_intr_enable' was not declared in this scope; did you mean 'esp_intr_enable'?
   430 |   gpio_intr_enable((gpio_num_t)params.recvpin)
   |   ^~~~~~~~~~~~~~~~
   |   esp_intr_enable

Jason2866, 2024-06-25:
@phphengenharia  use the PR #2040 which adds the needed changes.

phphengenharia, 2024-06-25:
It compiled fine. Thanks.

Vorms, 2024-08-05:
I need also the update for Espressif ver 3.x

tonhuisman, 2024-08-05:
> I need also the update for Espressif ver 3.x

Repeated answer:
use the PR #2040 which adds the needed changes.

Abdull, 2024-08-07:
Steps to use PR https://github.com/crankyoldgit/IRremoteESP8266/pull/2040 in *Arduino IDE 2.3.2* under Linux (no IDE restart required):

```sh
##########
# prerequisites:
# * have Github "gh" cli tool installed, e.g. for Debian 12 with "sudo apt install gh" (while quite outdated, it works for these steps)
# * in Arduino IDE 2's Library Manager GUI, remove any currently installed "IRremoteESP8266" released library, e.g. remove the currently latest (as of this writing) release 2.8.6

##########
# clone *this* repository to Arduino IDE 2's library directory:
cd "${HOME}/Arduino/libraries/"
git clone https://github.com/crankyoldgit/IRremoteESP8266.git

# check out merge request #2040:
cd "${HOME}/Arduino/libraries/IRremoteESP8266/"
gh pr checkout 2040

# if this is the first time the gh command is used, follow setup instructions,
# e.g. "gh auth login"
# then execute "gh pr checkout 2040" again

# if everything went well, "git show --summary" should show the following latest commit:
# $ git show --summary
# commit db98aa001b8302ee827d1cd23367eb43480ba5ee (HEAD -> feature/C++20-compatibility)
# Author: Ton Huisman <tonhu2004@hotmail.com>
# Date:   Thu Nov 2 21:22:16 2023 +0100
#
#     [Lib] Add latest ESP32 fixes by @s-hadinger, add extra NULL safeguard, update Platform build
#
```

mm5agm, 2024-08-09:
Steps to use PR #2040 in Arduino IDE 2.3.2 under Linux (no IDE restart required):

Can someone provide a Windows solution?

tonhuisman, 2024-08-09:
For Windows the `git` commands are the same, with the same requirement: having the Github tools installed. The only difference would be the directory names, but you should be able to open a command prompt from Windows Explorer to get there.

Vorms, 2024-08-09:
Hello,
Why the main branch doesn't support the version 3.x ?

Thierry

kaituozhe, 2024-08-23:
It seems not working with [arduino-esp32](https://github.com/espressif/arduino-esp32)  V3.0.4
I meet the same problem when using [IRremoteESP8266](https://github.com/crankyoldgit/IRremoteESP8266)  v2.8.6 :

In function 'void gpio_intr()':
error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
  246 |   timerAlarmEnable(timer);
      |   ^~~~~~~~~~~~~~~~
      |   timerAlarm

Zhentao-Lin, 2024-08-23:
[](https://github.com/Freenove/Freenove_IR_Lib_for_ESP32)
I haven't been able to use it for a long time and I'm waiting for them to update until they can, until then, I have my own ESP32-based rmt encapsulated library, which can barely be used to identify infrared remote controls. But it can't be used for anything else. There is no infrared transmission for the time being.

Zhentao-Lin, 2024-08-23:
https://github.com/Freenove/Freenove_IR_Lib_for_ESP32

tonhuisman, 2024-08-23:
> I meet the same problem when using [IRremoteESP8266](https://github.com/crankyoldgit/IRremoteESP8266) v2.8.6 :

@kaituoshe Then most likely you didn't pull the PR code local, as described above.

NiKiZe, 2024-09-23:
I'm considering merging #2144, looking for feedback from anyone that could help test.

mistic100, 2024-10-06:
I tested #2144 on a Seeeduino Xiao ESP32C3 using [pioarduino platform](https://github.com/pioarduino/platform-espressif32), not problem so far.

(well no more that the base library on Arduino 2, I get garbage output when using Wifi, but that's another story. Without Wifi it's OK)

quicksketch, 2025-01-02:
I tested https://github.com/crankyoldgit/IRremoteESP8266/pull/2040 (maybe I should have gone with the smaller/simpler #2144?). But I can confirm that https://github.com/crankyoldgit/IRremoteESP8266/pull/2040 works great with my ESP32S3 and Arduino esp32 3.1.0.

If #2144 is better for me to provide feedback, I'm happy to test that one also.

2lian, 2025-03-20:
I runing this branch on my Xiao ESP32-C6 , it works. Thanks!
I'll report back if there are any issues.

AdarWa, 2025-04-10:
@2lian 
How did you manage to get it working with the ESP32-C6 XIAO?
Would you mind sharing your platformio.ini?

carlos0810, 2025-04-10:
i am using arduino 2.3.6 and  IRremoteESP8266 v2.8.6 but it show the same error:     

c:\Users\mypc\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp: In function 'void gpio_intr()':
c:\Users\mypc\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:246:3: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
  246 |   timerAlarmEnable(timer);
      |   ^~~~~~~~~~~~~~~~
      |   timerAlarm
c:\Users\mypc\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp: In member function 'void IRrecv::enableIRIn(bool)':
c:\Users\mypc\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:362:21: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
  362 |   timer = timerBegin(_timer_num, 80, true);
      |           ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\mypc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32/esp32-hal.h:98,
                 from C:\Users\mypc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32/Arduino.h:45,
                 from c:\Users\mypc\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.h:10,
                 from c:\Users\mypc\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:6:
C:\Users\mypc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32/esp32-hal-timer.h:35:13: note: declared here
   35 | hw_timer_t *timerBegin(uint32_t frequency);
      |             ^~~~~~~~~~
c:\Users\mypc\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:371:3: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
  371 |   timerAlarmWrite(timer, MS_TO_USEC(params.timeout), ONCE);
      |   ^~~~~~~~~~~~~~~
      |   timerWrite
c:\Users\mypc\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:375:23: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
  375 |   timerAttachInterrupt(timer, &read_timeout, false);
      |   ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\mypc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
c:\Users\mypc\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp: In member function 'void IRrecv::disableIRIn()':
c:\Users\mypc\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:401:3: error: 'timerAlarmDisable' was not declared in this scope
  401 |   timerAlarmDisable(timer);
      |   ^~~~~~~~~~~~~~~~~
c:\Users\mypc\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp: In member function 'void IRrecv::resume()':
c:\Users\mypc\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:429:3: error: 'timerAlarmDisable' was not declared in this scope
  429 |   timerAlarmDisable(timer);
      |   ^~~~~~~~~~~~~~~~~

tonhuisman, 2025-04-11:
You will need the source code from this PR to successfully compile with Arduino 3.x / IDF 5.x.
Please read back earlier comments (some may be slightly hidden) on what to do, it's been mentioned a few times before... 🤔

mschratt1980, 2025-07-19:
Timer function altered in new Version. For reference see the following site:

https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html

and 

https://docs.espressif.com/projects/arduino-esp32/en/latest/api/timer.html