crankyoldgit/IRremoteESP8266 #2042: Panasonic AC 128bit

mrninhvn, 2023-10-26:
### What brand/model IR demodulator are you using?
Panasonic CZ-RWS3

### Information
I am trying to decode a Panasonic  AC signal with 128bit length, format show as follow:
```
union PanasonicAc128Protocol {
  uint8_t raw[kPanasonicAc128StateLength];  ///< The state in IR code form.
  struct {
    // Byte 0-4
    uint8_t Header[5];
    // Byte 5
    uint8_t Type        :2;
    uint8_t Fan         :2;
    uint8_t             :1;
    uint8_t Mode        :3;
    // Byte 6
    uint8_t             :1;
    uint8_t Temp        :5;
    uint8_t             :1;
    uint8_t isAuto      :1;
    // Byte 7
    uint8_t             :8; // 0xC0
    // Byte 8-9
    uint8_t             :8; // 0x00
    uint8_t             :8; // 0x00
    // Byte 10
    uint8_t SwingV      :3;
    uint8_t             :5;
    // Byte 11 // Checksum
    uint8_t Check1      :8;
    // Byte 12
    uint8_t             :8; // 0x02
    // Byte 13
    uint8_t             :4;
    uint8_t             :1;
    uint8_t Sensor      :1;
    uint8_t Ion         :1;
    uint8_t             :1;
    // Byte 14
    uint8_t             :6;
    uint8_t Econo       :1;
    uint8_t             :1;
    // Byte 15
    uint8_t Check2      :8;
  };
};
```
I have problem with 2 byte checksum,  the strange thing is that the checksum changes every 0x0F depending on the temperature level and mode, with no rule.
Thís is a sample of full data: `0x40001481260D9AC00000003E02E7076D`

Any idea?

mrninhvn, 2023-10-27:
Here is full data
[CZ-RWS3 Datas.csv](https://github.com/crankyoldgit/IRremoteESP8266/files/13187473/CZ-RWS3.Datas.csv)

aboyuanna, 2025-09-04:
@mrninhvn Hi, I am currently facing the same issue. Have you resolved it?

mrninhvn, 2025-09-04:
> [@mrninhvn](https://github.com/mrninhvn) Hi, I am currently facing the same issue. Have you resolved it?

Yes, I managed to make it work, but in a dumb way.

```
void IRPanasonicAc128::calcChecksum(void) {
  _.Check1 = sumBytes(_.raw, 11, 0xCD);

  switch (getMode()){

    case kPanasonicAcAuto: {
      if (getTemp() < 20) _.Check1 += 0x0F;
      _.Check2 = sumBytes(_.raw, 15, 0xCE);
      if (getTemp() < 20) _.Check2 += 0x0F;
      uint8_t check2Temp = _.Check2;
      if (check2Temp >= 0x40 && check2Temp <= 0x4F) _.Check2 += 0x0F;
      if (check2Temp >= 0x70 && check2Temp <= 0x8F) _.Check2 -= 0x0F;
      break;
    }

    case kPanasonicAcHeat: {
      _.Check1 += 0x0F;
      _.Check1 += 0x0F;
      _.Check1 += 0x0F;
      if (getTemp() < 28) { _.Check1 += 0x0F; }
      if (getTemp() < 20) _.Check1 += 0x0F;
      _.Check2 = sumBytes(_.raw, 15, 0xCE);
      _.Check2 += 4*0x0F;
      if (getTemp() < 28) { _.Check2 += 0x0F; }
      if (getTemp() < 20) { _.Check2 += 0x0F; }
      uint8_t check2Temp = _.Check2;
      if (check2Temp >= 0x60 && check2Temp <= 0x7F) _.Check2 -= 0x0F;
      if (check2Temp >= 0x80 && check2Temp <= 0x8F) _.Check2 -= 2*0x0F;
      break;
    }

    case kPanasonicAcDry: {
      _.Check1 -= 4*0x0F;
      if (getTemp() < 20) _.Check1 += 0x0F;
      if (getTemp() > 27) _.Check1 -= 0x0F;
      _.Check2 = sumBytes(_.raw, 15, 0xCE);
      _.Check2 -= 4*0x0F;
      if (getTemp() < 20) _.Check2 += 0x0F;
      if (getTemp() > 27) _.Check2 -= 0x0F;
      uint8_t check2Temp = _.Check2;
      if (check2Temp >= 0x70 && check2Temp <= 0x8F) _.Check2 -= 0x0F;
      if (check2Temp >= 0x40 && check2Temp <= 0x4F) _.Check2 += 0x0F;
      break;
    }

    case kPanasonicAcCool: {
      if (getTemp() < 20) _.Check1 += 0x0F;
      if (getTemp() > 27) _.Check1 -= 0x0F;
      _.Check2 = sumBytes(_.raw, 15, 0xCE);
      if (getTemp() < 20) _.Check2 += 0x0F;
      if (getTemp() > 27) _.Check2 -= 0x0F;
      uint8_t check2Temp = _.Check2;
      if (check2Temp >= 0x40 && check2Temp <= 0x4F) _.Check2 += 0x0F;
      if (check2Temp >= 0x70 && check2Temp <= 0x8F) _.Check2 -= 0x0F;
      break;
    }

    case kPanasonicAcFan: {
      _.Check1 += 8*0x0F;
      _.Check2 = sumBytes(_.raw, 15, 0xCE);
      _.Check2 += 8*0x0F;
      if (_.Check2 >= 0x40 && _.Check2 <= 0x4F) _.Check2 += 0x0F;
      break;
    }

    default: {
      break;
    }
  }
  ESP_LOGW("IR", "Check1: %02X, Check2: %02X", _.Check1, _.Check2);
}
```