crankyoldgit/IRremoteESP8266 #2049: IRremoteESP8266 library sending wrong when send signal of  32 bits

Teslaxxx, 2023-12-08:
### Version/revision of the library used
 v2.8.6

### Describe the bug
 when store  signal of  32 bits of Samsung that read by **irrecv.decode(&results)** function  , then when sending that signal by **irsend.sendSAMSUNG(0xE0E0E01F, 32 );** 
then read it  by irrecv.decode(&results) function again , the results not same between signal i store and signal  i send 
please help me 
thanks

NiKiZe, 2023-12-08:
What is the full output that gave you the original 0xE0E0E01F ?

You can see in tests that we are confident that sending the data will be received the same, given default parameters https://github.com/crankyoldgit/IRremoteESP8266/blob/a295f87e067bc8ccb22523ef035a05ea11d65e49/test/ir_Samsung_test.cpp#L123

Maybe you can post full output from the dump program, (as requested in the issue template)

Teslaxxx, 2023-12-09:
Yes, that's right, I was sending lowercase letters, not capital letters , I thought they were the same because we send hex code

NiKiZe, 2023-12-09:
If this is in code, then it is your compiler that deals with 0x.. to be converted to integer. You might have a compiler that does this differently, but that's unlikely, ther emust be something else, or are you sending this as text somewhere?

crankyoldgit, 2023-12-09:
Can you please post the minimalist code that reproduces the issue you have?
We are fairly confident the library is correct, and we suspect you're doing something incorrect in **your** code.
Please also supply the output of the IRrecvDumpV3 program too. That will give us all the info about the physical  signal you created.