crankyoldgit/IRremoteESP8266 #2053: Library's Non-declared variables & too many arguments to function issues.

QnBarb, 2023-12-20:
### Version/revision of the library used
2.8.6

### Describe the bug
Variables not declared: timerAlarmEnable; timerAlarmWrite; timerAlarmDisable(timer); gpio_intr_disable((gpio_num_t)params.recvpin); gpio_intr_enable((gpio_num_t)params.recvpin)

Too many arguments to function: timer = timerBegin(_timer_num, 80, true); timerAttachInterrupt(timer, &read_timeout, false)

#### To Reproduce
Compile/Upload

#### Example code used
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRrecv.h>
#include <IRutils.h>

// an IR detector/demodulator is connected to GPIO pin 2
uint16_t RECV_PIN = 2;

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup() {
  Serial.begin(115200);
  irrecv.enableIRIn();  // Start the receiver
}

void dump(decode_results *results) {
  // Dumps out the decode_results structure.
  // Call this after IRrecv::decode()
  uint16_t count = results->rawlen;
  if (results->decode_type == UNKNOWN) {
    Serial.print("Unknown encoding: ");
  } else if (results->decode_type == NEC) {
    Serial.print("Decoded NEC: ");
  } else if (results->decode_type == SONY) {
    Serial.print("Decoded SONY: ");
  } else if (results->decode_type == RC5) {
    Serial.print("Decoded RC5: ");
  } else if (results->decode_type == RC5X) {
    Serial.print("Decoded RC5X: ");
  } else if (results->decode_type == RC6) {
    Serial.print("Decoded RC6: ");
  } else if (results->decode_type == RCMM) {
    Serial.print("Decoded RCMM: ");
  } else if (results->decode_type == PANASONIC) {
    Serial.print("Decoded PANASONIC - Address: ");
    Serial.print(results->address, HEX);
    Serial.print(" Value: ");
  } else if (results->decode_type == LG) {
    Serial.print("Decoded LG: ");
  } else if (results->decode_type == JVC) {
    Serial.print("Decoded JVC: ");
  } else if (results->decode_type == AIWA_RC_T501) {
    Serial.print("Decoded AIWA RC T501: ");
  } else if (results->decode_type == WHYNTER) {
    Serial.print("Decoded Whynter: ");
  } else if (results->decode_type == NIKAI) {
    Serial.print("Decoded Nikai: ");
  }
  serialPrintUint64(results->value, 16);
  Serial.print(" (");
  Serial.print(results->bits, DEC);
  Serial.println(" bits)");
  Serial.print("Raw (");
  Serial.print(count, DEC);
  Serial.print("): {");

  for (uint16_t i = 1; i < count; i++) {
    if (i % 100 == 0)
      yield();  // Preemptive yield every 100th entry to feed the WDT.
    if (i & 1) {
      Serial.print(results->rawbuf[i] * kRawTick, DEC);
    } else {
      Serial.print(", ");
      Serial.print((uint32_t) results->rawbuf[i] * kRawTick, DEC);
    }
  }
  Serial.println("};");
}

void loop() {
  if (irrecv.decode(&results)) {
    dump(&results);
    Serial.println("DEPRECATED: Please use IRrecvDumpV2.ino instead!");
    irrecv.resume();  // Receive the next value
  }
}

#### Expected behaviour
1. Compile.
2. Expect compile.

#### Output of raw data from [IRrecvDumpV2.ino]
c:\Users\QuangPC\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp: In function 'void gpio_intr()':
c:\Users\QuangPC\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:246:3: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
  246 |   timerAlarmEnable(timer);
      |   ^~~~~~~~~~~~~~~~
      |   timerAlarm
c:\Users\QuangPC\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp: In member function 'void IRrecv::enableIRIn(bool)':
c:\Users\QuangPC\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:362:21: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
  362 |   timer = timerBegin(_timer_num, 80, true);
      |           ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\QuangPC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\cores\esp32/esp32-hal.h:84,
                 from C:\Users\QuangPC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\cores\esp32/Arduino.h:36,
                 from c:\Users\QuangPC\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.h:10,
                 from c:\Users\QuangPC\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:6:
C:\Users\QuangPC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\cores\esp32/esp32-hal-timer.h:35:14: note: declared here
   35 | hw_timer_t * timerBegin(uint32_t frequency);
      |              ^~~~~~~~~~
c:\Users\QuangPC\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:371:3: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
  371 |   timerAlarmWrite(timer, MS_TO_USEC(params.timeout), ONCE);
      |   ^~~~~~~~~~~~~~~
      |   timerWrite
c:\Users\QuangPC\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:375:23: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
  375 |   timerAttachInterrupt(timer, &read_timeout, false);
      |   ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\QuangPC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\cores\esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t * timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
c:\Users\QuangPC\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp: In member function 'void IRrecv::disableIRIn()':
c:\Users\QuangPC\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:401:3: error: 'timerAlarmDisable' was not declared in this scope
  401 |   timerAlarmDisable(timer);
      |   ^~~~~~~~~~~~~~~~~
c:\Users\QuangPC\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp: In member function 'void IRrecv::pause()':
c:\Users\QuangPC\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:416:3: error: 'gpio_intr_disable' was not declared in this scope; did you mean 'esp_intr_disable'?
  416 |   gpio_intr_disable((gpio_num_t)params.recvpin);
      |   ^~~~~~~~~~~~~~~~~
      |   esp_intr_disable
c:\Users\QuangPC\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp: In member function 'void IRrecv::resume()':
c:\Users\QuangPC\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:429:3: error: 'timerAlarmDisable' was not declared in this scope
  429 |   timerAlarmDisable(timer);
      |   ^~~~~~~~~~~~~~~~~
c:\Users\QuangPC\Documents\Arduino\libraries\IRremoteESP8266\src\IRrecv.cpp:430:3: error: 'gpio_intr_enable' was not declared in this scope; did you mean 'esp_intr_enable'?
  430 |   gpio_intr_enable((gpio_num_t)params.recvpin);
      |   ^~~~~~~~~~~~~~~~
      |   esp_intr_enable

exit status 1

Compilation error: exit status 1

### What brand/model IR demodulator are you using?
Vishay TSSP93056.

### Circuit diagram and hardware used (if applicable)
Board used: ESP32C3DevKitM-1

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes

### Has this library/code previously worked as expected for you?
_No

### Other useful information
-

NiKiZe, 2023-12-20:
This seems to be related to C3 and features in the Arduino framework that has changed, see already existing issues.

NiKiZe, 2024-07-27:
Closing as duplicate of #2039

closed 2024-07-27 by NiKiZe