crankyoldgit/IRremoteESP8266 #2059: IRremoteESP8266 library compile error bk7231t

DevineCZ, 2024-01-30:
### Version/revision of the library used
2.8.6

### Describe the bug
When I compile in ESPHome 2023.12.9 in Home Assistant IRremoteESP826, I get an error message.

Compiling .pioenvs/samsung-ac-remote-control/lib84c/IRremoteESP8266/IRrecv.cpp.o
Compiling .pioenvs/samsung-ac-remote-control/lib84c/IRremoteESP8266/IRtext.cpp.o
Compiling .pioenvs/samsung-ac-remote-control/lib84c/IRremoteESP8266/IRtimer.cpp.o
Compiling .pioenvs/samsung-ac-remote-control/lib84c/IRremoteESP8266/IRutils.cpp.o
/piolibs/IRremoteESP8266/src/IRrecv.cpp:140:7: error: '_IRrecv' has not been declared
  140 | using _IRrecv::timer;
      | ^~~~~~~
/piolibs/IRremoteESP8266/src/IRrecv.cpp:174:3: error: expected unqualified-id before 'if'
  174 | if (params.rawlen) params.rcvstate = kStopState;
      | ^~
/piolibs/IRremoteESP8266/src/IRrecv.cpp:181:1: error: expected declaration before '}' token
  181 | }
      | ^
/piolibs/IRremoteESP8266/src/IRrecv.cpp:184:27: error: expected initializer before 'gpio_intr'
  184 | static void USE_IRAM_ATTR gpio_intr() {
      | ^~~~~~~~~
/piolibs/IRremoteESP8266/src/IRrecv.cpp: In member function 'void IRrecv::enableIRIn(bool)':
/piolibs/IRremoteESP8266/src/IRrecv.cpp:389:35: error: 'gpio_intr' was not declared in this scope
  389 | attachInterrupt(params.recvpin, gpio_intr, CHANGE);
      | ^~~~~~~~~
*** [.pioenvs/samsung-ac-remote-control/lib84c/IRremoteESP8266/IRrecv.cpp.o] Error 1

#### To Reproduce
In the ESPHome add-on in Home Assistant, try compiling the following:

esphome:
  name: samsung-ac-remote-control
  friendly_name: samsung-ac
  includes:
  - include/irsamsung.h
  libraries:
  - IRremoteESP8266
  build_path: build/samsung-ac-remote-control
  area: ''
  platformio_options: {}
  name_add_mac_suffix: false
  min_version: 2023.12.9
bk72xx:
  board: wb3s
  framework:
    version: 0.0.0
    loglevel: WARN
    debug: []
    sdk_silent: all
    gpio_recover: true
    options: {}
    source: null
  family: BK7231T
  component_id: bk72xx
wifi:
  ap:
    ssid: Samsung Ac Remote Control
    password: password
    ap_timeout: 1min
  domain: .local
  reboot_timeout: 15min
  power_save_mode: NONE
  fast_connect: false
  passive_scan: false
  enable_on_boot: true
  networks:
  - ssid: xxxxxx
    password: xxxxxx
    priority: 0.0
  use_address: samsung-ac-remote-control.local
captive_portal: {}
logger:
  baud_rate: 115200
  tx_buffer_size: 512
  deassert_rts_dtr: false
  hardware_uart: DEFAULT
  level: DEBUG
  logs: {}
api:
  password: password
  port: 6053
  reboot_timeout: 15min
ota:
  password: password
  safe_mode: true
  port: 8892
  reboot_timeout: 5min
  num_attempts: 10
status_led:
  pin:
    number: 9
    mode:
      output: true
      input: false
      open_drain: false
      pullup: false
      pulldown: false
      analog: false
    inverted: false
sensor:
- platform: template
  name: My Room Temperature
  accuracy_decimals: 1
  id: my_room_temp
  disabled_by_default: false
  force_update: false
  update_interval: 60s
climate:
- platform: custom
  lambda: !lambda |-
    auto samsungac = new SamsungAC();
    samsungac->set_sensor(id(my_room_temp));
    App.register_component(samsungac);
    return {samsungac};
  climates:
  - name: My Room AC
    disabled_by_default: false
    visual: {}

crankyoldgit, 2024-02-02:
> Processing samsung-ac-remote-control (board: wb3s; framework: arduino; platform: libretiny @ 1.4.1)

I think your problem is you're trying to build for a platform that this library doesn't support.
ie. _wb3s_ is not an _ESP8266_ or a _ESP32_.

See the main `README.md` for this repository/project/library:

https://github.com/crankyoldgit/IRremoteESP8266/blob/f67948f6c8b156ff178b943af52bd4d3e0a94a6a/README.md?plain=1#L10-L11

nagyrobi, 2024-06-26:
Fix in https://github.com/crankyoldgit/IRremoteESP8266/pull/2111