crankyoldgit/IRremoteESP8266 #2063: Fails to detect Hitachi AC Remote. Sending raw data also does not work

vgahlaut, 2024-02-08:
### Version/revision of the library used
2.8.6

### Describe the bug
I am using IRremoteESP8266 for an IRBlaster project. It is working fine with many ACs that we tested (Voltas, carrier, coolix) but not working one of the HITACHI AC I have.

Used IRrecvDumpV2 to captured Turn-On event on 38KHz Receiver (Vishay TSOP14638) and it reports Unknown protocol. Tried to send rawdata but did not work.

Used IRrecvDumpV2 to captured Turn-On event on 56KHz Receiver (Vishay TSSP4056) and it reports HITACHI_AC296 protocol. Tried to send rawdata but did not work. I tried sending this data by passing both 56KHz and 38KHz parameter in IRsend.sendRaw() and nothing works.

Tried calling IRHitachiAc296 functions to turn on AC but again it does not work. Here is the code. 

`    IRHitachiAc296 ac(13);
    ac.begin();
    ac.setFan(kHitachiAc296FanAuto);
    ac.setMode(kHitachiAc296Cool);
    ac.setTemp(25);
    ac.on();
    ac.send();`

Looks like false detection on 56KHz and No detection on 38KHz. But one thing that I am not able to understand why sending rawdata is not working. Using standard example send raw data. Code is also provided.

#### To Reproduce
Not sure how can it be reproduced without having same model of AC and Remote at other's end. On My end I use IRRecvDumpV2 to capture and IRSendDemo to send raw data

#### Example code used

For Recv
`#include <Arduino.h>
#include <assert.h>
#include <IRrecv.h>
#include <IRremoteESP8266.h>
#include <IRac.h>
#include <IRtext.h>
#include <IRutils.h>

// ==================== start of TUNEABLE PARAMETERS ====================
// An IR detector/demodulator is connected to GPIO pin 14
// e.g. D5 on a NodeMCU board.
// Note: GPIO 16 won't work on the ESP8266 as it does not have interrupts.
// Note: GPIO 14 won't work on the ESP32-C3 as it causes the board to reboot.
#ifdef ARDUINO_ESP32C3_DEV
const uint16_t kRecvPin = 10;  // 14 on a ESP32-C3 causes a boot loop.
#else  // ARDUINO_ESP32C3_DEV
const uint16_t kRecvPin = 12;
#endif  // ARDUINO_ESP32C3_DEV

// The Serial connection baud rate.
// i.e. Status message will be sent to the PC at this baud rate.
// Try to avoid slow speeds like 9600, as you will miss messages and
// cause other problems. 115200 (or faster) is recommended.
// NOTE: Make sure you set your Serial Monitor to the same speed.
const uint32_t kBaudRate = 115200;

// As this program is a special purpose capture/decoder, let us use a larger
// than normal buffer so we can handle Air Conditioner remote codes.
const uint16_t kCaptureBufferSize = 1024;

// kTimeout is the Nr. of milli-Seconds of no-more-data before we consider a
// message ended.
// This parameter is an interesting trade-off. The longer the timeout, the more
// complex a message it can capture. e.g. Some device protocols will send
// multiple message packets in quick succession, like Air Conditioner remotes.
// Air Coniditioner protocols often have a considerable gap (20-40+ms) between
// packets.
// The downside of a large timeout value is a lot of less complex protocols
// send multiple messages when the remote's button is held down. The gap between
// them is often also around 20+ms. This can result in the raw data be 2-3+
// times larger than needed as it has captured 2-3+ messages in a single
// capture. Setting a low timeout value can resolve this.
[... 193 lines cut ...]
                                            Mesg Desc.: Power: On, Mode: 6 (Heat), Temp: 24C, Fan: 3 (Medium)
uint16_t rawData[595] = {3306, 1698,  416, 1272,  416, 444,  416, 444,  418, 444,  414, 444,  418, 442,  418, 444,  416, 448,  418, 442,  418, 444,  416, 466,  418, 442,  418, 1268,  420, 440,  418, 444,  416, 450,  418, 444,  416, 444,  420, 440,  418, 444,  416, 444,  418, 440,  418, 444,  418, 450,  416, 444,  416, 444,  418, 444,  416, 444,  418, 442,  418, 440,  418, 1272,  416, 450,  418, 1304,  418, 1272,  416, 1268,  418, 1268,  418, 1270,  416, 1272,  418, 440,  420, 1274,  418, 1270,  418, 1268,  418, 1270,  418, 1268,  416, 1272,  414, 1272,  416, 1272,  416, 1278,  416, 466,  418, 442,  418, 444,  416, 440,  418, 442,  418, 444,  416, 444,  416, 452,  416, 444,  416, 444,  416, 1272,  414, 1272,  416, 444,  416, 444,  416, 1272,  416, 1278,  418, 1268,  418, 1270,  418, 442,  418, 440,  418, 1268,  418, 1270,  418, 440,  418, 450,  416, 444,  416, 1272,  416, 444,  418, 440,  418, 1268,  418, 444,  416, 444,  418, 1278,  416, 1272,  416, 444,  418, 1270,  418, 1270,  418, 442,  418, 1270,  418, 1268,  420, 484,  418, 1270,  418, 1268,  418, 442,  418, 440,  416, 1272,  416, 444,  416, 444,  416, 452,  416, 444,  416, 444,  416, 1272,  416, 1272,  416, 444,  416, 1272,  416, 1272,  416, 1278,  416, 442,  418, 442,  418, 440,  418, 442,  418, 440,  418, 1268,  416, 1272,  416, 450,  418, 1270,  418, 1270,  418, 1268,  416, 1272,  416, 1272,  416, 444,  418, 440,  416, 1282,  416, 444,  416, 444,  416, 444,  420, 440,  418, 440,  418, 444,  416, 474,  416, 452,  416, 1270,  418, 1268,  418, 1270,  418, 1268,  420, 1268,  416, 1272,  416, 1268,  418, 1278,  416, 444,  416, 444,  416, 444,  418, 442,  418, 444,  416, 444,  416, 444,  418, 450,  416, 1272,  416, 1272,  416, 1272,  416, 1272,  416, 1272,  416, 1272,  416, 1272,  416, 1280,  416, 440,  418, 444,  416, 444,  414, 444,  418, 440,  420, 444,  416, 444,  416, 450,  418, 1272,  416, 1268,  418, 1270,  418, 1270,  418, 1268,  416, 1272,  416, 1272,  416, 1280,  416, 444,  416, 444,  418, 444,  416, 444,  416, 444,  418, 442,  418, 440,  418, 450,  418, 1268,  418, 1270,  418, 1268,  416, 1272,  416, 1272,  416, 1272,  416, 1272,  416, 1280,  416, 444,  416, 444,  416, 444,  416, 444,  418, 442,  418, 442,  418, 440,  420, 446,  418, 1268,  418, 1272,  418, 1270,  416, 1272,  416, 1270,  418, 1270,  418, 1270,  418, 1278,  418, 442,  418, 1266,  418, 1272,  416, 444,  436, 1248,  418, 1268,  440, 424,  436, 430,  440, 1248,  440, 424,  436, 422,  436, 1264,  418, 444,  436, 424,  436, 1250,  436, 1258,  438, 1248,  440, 424,  436, 420,  440, 420,  438, 1252,  436, 1248,  438, 1248,  438, 1258,  438, 420,  418, 1270,  436, 1252,  436, 1252,  418, 440,  438, 424,  436, 422,  418, 450,  436, 424,  436, 424,  440, 420,  440, 422,  436, 424,  436, 424,  438, 420,  440, 428,  440, 1248,  436, 1252,  436, 1252,  436, 1250,  440, 1248,  436, 1252,  436, 1252,  436, 1256,  436, 424,  436, 424,  436, 424,  440, 420,  438, 424,  436, 422,  436, 422,  436, 428,  440, 1248,  438, 1248,  440, 1252,  436, 1252,  436, 1252,  436, 1250,  438, 1248,  438, 1258,  436, 424,  436, 424,  438, 420,  440, 420,  456, 402,  456, 402,  456, 406,  440, 426,  456, 1234,  436, 1252,  458, 1230,  436, 1252,  436, 1248,  456, 1232,  456, 1232,  456, 1240,  436, 1250,  436, 1252,  436, 424,  434, 424,  456, 404,  456, 404,  456, 406,  454, 412,  456, 406,  454, 406,  454, 1230,  456, 1234,  452, 1230,  456, 1232,  456, 1252,  456, 1224,  454};  // HITACHI_AC296
               uint8_t state[37] = {0x01, 0x10, 0x00, 0x40, 0xBF, 0xFF, 0x00, 0xCC, 0x33, 0x92, 0x6D, 0x13, 0xEC, 0x60, 0x9F, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x36, 0xC9, 0xF1, 0x0E, 0x00, 0xFF, 0x00, 0xFF, 0x00, 0xFF, 0x03, 0xFC};`

### What brand/model IR demodulator are you using?
IR Transmitter: Vishay TSAL4400
IR Receiver: Vishay TSOP14638 (38Khz)
IR Receiver: Vishay TSSP4056 (56KHz)

### Circuit diagram and hardware used (if applicable)
Generic ESP8266. Tx GPIO 13
RX GPIO: 12 (38Khz)
RX GPIO: 14 (56Khz)

### I have followed the steps in the [Troubleshooting Guide]
(https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes

### Has this library/code previously worked as expected for you?
No

### Other useful information
Attaching image of remote and model/specs of AC
[image]
[image]

crankyoldgit, 2024-02-09:
What you've described is not a failure to detect. You're describing a failure to transmit.

You need to check your sending circuit. Please try following the [troubleshooting guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide#sending-an-ir-signal)

vgahlaut, 2024-02-09:
Dear @crankyoldgit ,

I have already checked Transmit circuit multiple times. The same circuit is working absolutely fine for other protocols (Voltas, Carrier) and all those protocols are well detected with 38Khz receiver. This Hitachi one is showing unknown protocol with 38Khz receiver. 

Attaching Transmit circuit driven by GPIO13 of ESP8266. Can you please check and let me know what is wrong with the transmit circuit and why it works for other protocols that are correctly detected with 38KHz receiver.

[image]

crankyoldgit, 2024-02-09:
> Also tried to capture at 56KHz IR Receiver: Vishay TSSP4056 (56KHz) to see if that works but that again does not work
> ```
> Timestamp : 000009.476 Library : v2.8.6
>
> Protocol : HITACHI_AC296
> Code : [...]

The info you supplied shows it is working when using the 56k IR demodulator.

Try sending the 56k raw data that you collected decoded successfully with `sendRaw(rawData, 595, 56);`

Perhaps your remote & A/C operate at 56k, where as others have been reported to work at 38k.

The chances of a false-positive detection of the hex-code of a long AC message like that are vanishingly small. It may easily be the text message desription is wrong. But these protocols have plenty of checks built into them to detect 'errors' in the messages, and we check for those most of the time. i.e. If the protocol isn't followed, the message is rejected.

vgahlaut, 2024-02-09:
I already tried sending 56K raw data with `sendRaw(rawData, 595, 56);` and it did not work. If that could have worked I did not need to look for help because it could be clear that Remote is working on 56KHz and we need to settle both Tx and Rx on 56KHz but sending raw data is not working either on 38KHz or on 56KHz and that is the reason I am seeking your help because it seems to be something deep that I am not familiar with.

Sketch is also below.

`#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>

const uint16_t kIrLed = 13;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).

IRsend irsend(kIrLed);  // Set the GPIO to be used to sending the message.

// Example of data captured by IRrecvDumpV2.ino
uint16_t rawData[595] = {3306, 1698,  416, 1272,  416, 444,  416, 444,  418, 444,  414, 444,  418, 442,  418, 444,  416, 448,  418, 442,  418, 444,  416, 466,  418, 442,  418, 1268,  420, 440,  418, 444,  416, 450,  418, 444,  416, 444,  420, 440,  418, 444,  416, 444,  418, 440,  418, 444,  418, 450,  416, 444,  416, 444,  418, 444,  416, 444,  418, 442,  418, 440,  418, 1272,  416, 450,  418, 1304,  418, 1272,  416, 1268,  418, 1268,  418, 1270,  416, 1272,  418, 440,  420, 1274,  418, 1270,  418, 1268,  418, 1270,  418, 1268,  416, 1272,  414, 1272,  416, 1272,  416, 1278,  416, 466,  418, 442,  418, 444,  416, 440,  418, 442,  418, 444,  416, 444,  416, 452,  416, 444,  416, 444,  416, 1272,  414, 1272,  416, 444,  416, 444,  416, 1272,  416, 1278,  418, 1268,  418, 1270,  418, 442,  418, 440,  418, 1268,  418, 1270,  418, 440,  418, 450,  416, 444,  416, 1272,  416, 444,  418, 440,  418, 1268,  418, 444,  416, 444,  418, 1278,  416, 1272,  416, 444,  418, 1270,  418, 1270,  418, 442,  418, 1270,  418, 1268,  420, 484,  418, 1270,  418, 1268,  418, 442,  418, 440,  416, 1272,  416, 444,  416, 444,  416, 452,  416, 444,  416, 444,  416, 1272,  416, 1272,  416, 444,  416, 1272,  416, 1272,  416, 1278,  416, 442,  418, 442,  418, 440,  418, 442,  418, 440,  418, 1268,  416, 1272,  416, 450,  418, 1270,  418, 1270,  418, 1268,  416, 1272,  416, 1272,  416, 444,  418, 440,  416, 1282,  416, 444,  416, 444,  416, 444,  420, 440,  418, 440,  418, 444,  416, 474,  416, 452,  416, 1270,  418, 1268,  418, 1270,  418, 1268,  420, 1268,  416, 1272,  416, 1268,  418, 1278,  416, 444,  416, 444,  416, 444,  418, 442,  418, 444,  416, 444,  416, 444,  418, 450,  416, 1272,  416, 1272,  416, 1272,  416, 1272,  416, 1272,  416, 1272,  416, 1272,  416, 1280,  416, 440,  418, 444,  416, 444,  414, 444,  418, 440,  420, 444,  416, 444,  416, 450,  418, 1272,  416, 1268,  418, 1270,  418, 1270,  418, 1268,  416, 1272,  416, 1272,  416, 1280,  416, 444,  416, 444,  418, 444,  416, 444,  416, 444,  418, 442,  418, 440,  418, 450,  418, 1268,  418, 1270,  418, 1268,  416, 1272,  416, 1272,  416, 1272,  416, 1272,  416, 1280,  416, 444,  416, 444,  416, 444,  416, 444,  418, 442,  418, 442,  418, 440,  420, 446,  418, 1268,  418, 1272,  418, 1270,  416, 1272,  416, 1270,  418, 1270,  418, 1270,  418, 1278,  418, 442,  418, 1266,  418, 1272,  416, 444,  436, 1248,  418, 1268,  440, 424,  436, 430,  440, 1248,  440, 424,  436, 422,  436, 1264,  418, 444,  436, 424,  436, 1250,  436, 1258,  438, 1248,  440, 424,  436, 420,  440, 420,  438, 1252,  436, 1248,  438, 1248,  438, 1258,  438, 420,  418, 1270,  436, 1252,  436, 1252,  418, 440,  438, 424,  436, 422,  418, 450,  436, 424,  436, 424,  440, 420,  440, 422,  436, 424,  436, 424,  438, 420,  440, 428,  440, 1248,  436, 1252,  436, 1252,  436, 1250,  440, 1248,  436, 1252,  436, 1252,  436, 1256,  436, 424,  436, 424,  436, 424,  440, 420,  438, 424,  436, 422,  436, 422,  436, 428,  440, 1248,  438, 1248,  440, 1252,  436, 1252,  436, 1252,  436, 1250,  438, 1248,  438, 1258,  436, 424,  436, 424,  438, 420,  440, 420,  456, 402,  456, 402,  456, 406,  440, 426,  456, 1234,  436, 1252,  458, 1230,  436, 1252,  436, 1248,  456, 1232,  456, 1232,  456, 1240,  436, 1250,  436, 1252,  436, 424,  434, 424,  456, 404,  456, 404,  456, 406,  454, 412,  456, 406,  454, 406,  454, 1230,  456, 1234,  452, 1230,  456, 1232,  456, 1252,  456, 1224,  454};

void setup() {
  irsend.begin();
#if ESP8266
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
#else  // ESP8266
  Serial.begin(115200, SERIAL_8N1);
#endif  // ESP8266
}

void loop() {
  Serial.println("Sending a rawData capture from IRrecvDumpV2");
  irsend.sendRaw(rawData, 595, 56);  // Send a raw data capture at 38kHz.
  delay(2000);
}
`

vgahlaut, 2024-02-09:
Please note that I also tried to use standard commands offered by protocol 

`IRHitachiAc296 ac(13); ac.begin(); ac.setFan(kHitachiAc296FanAuto); ac.setMode(kHitachiAc296Cool); ac.setTemp(25); ac.on(); ac.send();`

This also did not work. But I just checked that in constructor IRHitachiAc296  also provide options "inverted" and "use_modulation" parameter apart from Tx pin. I have not played with these parameters yet. Not really sure what these parameter will do but will try all possible combination and will update the results here.

crankyoldgit, 2024-02-09:
Is your IR led still working for your other devices? e.g. You could have killed it. They are notoriously easy to kill and you won't notice it because the messages are not visible. Have you tested it is working (blinking) using a phone-camera?
How close is the receiving device to the sending device?
Is the room dark?

Don't try to using the Aircon class object yet, that is locked to 38khz. If the raw signal replay isn't working, there are far more fundamental problems.

Do you have a second ESP8266 to configure as a capture device?

Do you have an oscilloscope to connect to the output signal?

The TSAL4400 seems to be a blue-grey unit, most of the IR 940nm LEDs people have reported success with are the clear ones. I can't see any reason why it wouldn't work. But we had someone report similar issue to you (trouble transmitting) only to replace it with a cheap/generic/clear IR 940nm LED, and it worked smoothly after that.

vgahlaut, 2024-02-09:
1. IR LED is fully function and can control my Voltas AC both with raw data and class object. 
2. Yes, Its Phone camera shows pink/purple blinking when command is send
3. No. Didn't try in the dark room. I doubt that it will make any difference because currently keeping circuit only 3-4 feet away and its not working. However will try in dark room and update results.
4. I will assemble another ESP8266 PCB to capture data. Have only 1 at this time
5. No. I do not have oscilloscope to connect to output signal.
6. Ordered Everlight IR333-3C clear LED. Will have in a day or two. Will test with clear LED also and update results.

crankyoldgit, 2024-02-09:
Hmm. Maybe try moving it closer or futher away from the A/C.

The raw capture and send are about as dumb and simple as it gets. It's is just recording the signal and replaying it. The only thing it can't see is the modulation frequency. If it doesn't work, then something is typically wrong in the hardware somewhere or environment. If you're getting a noticable clear illumination of the IR led (via camera) with your test code, then it should just work. Potentially the signal might not be strong enough or too much.

Maybe try making a `irsend.calibrate(56000);` call (see [api doc](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRsend.html#ad1776aa6c699f9eeca1eef9bb4fe355b)) after the `irsend,begin();`. That can potentially fix some internal timing settings, especially if you're running the ESP at an unexpected or unusual clock rate.

That fact you've go the same exact circuit sending to other devices fine kind of rules out you using the library incorrectly with your code.

Just an obvious/dumb question to ask. The IR remote used in the capture does actually control the AC you're trying to send to, right?
Do you have the model number of the remote?

vgahlaut, 2024-02-09:
Dear @crankyoldgit 

Thank you so much for your help. Got raw data working. 

As you correctly suspected it was due to transmit power. Based on your suggestion I made the room completely dark and then tried it from my desk (around 7-8 feet) and in the dark it worked from my desk (wow). But when I turned on the lights, it again did not work. Then I took circuit near to AC around 3-4 feet keeping LED exactly pointing to the receiver side of AC and it worked now in fully lighted room. So that seems to conclude the issue is transmit power. 

Its 100mA Transmit LED and I do not want to give more than 70mA for a safer side. My final circuit will have 6-7 transmitters, transmitting at same time in 360 direction. I hope that will resolve the transmit power issue. 

Nothing wrong with library of course. Sorry to trouble you and tons of thanks for your support.