crankyoldgit/IRremoteESP8266 #2075: Carrier AC 84bit

RBroks, 2024-02-28:
Hello, is there plan to support Carrier 84-bit protocol in full, same as MITSUBISHI_AC?

NiKiZe, 2024-02-28:
Please follow the issue template, the questions in there is important, you will find information about what is required for a new protocol in the FAQ, which is linked to from the template.

The FAQ is also clear on what we do and don't.
We can not create a protocol without having any data on the protocol.
We will not have the time to reverse engineer protocols.
There is no point in having an issue open on this unless the data exists, so closing until such data is provided.

closed 2024-02-28 by NiKiZe

RBroks, 2024-02-29:
Could it help if i can read all remote functions and write them in file? I can read every single function and write down what did i  change. I am new to this stuff just wanted to know is there any options to help somehow, i am not so powerful to doit all by my self..