crankyoldgit/IRremoteESP8266 #2079: Problem with Zanussi A/C

stakanich, 2024-03-15:
### Version/revision of the library used
2.8.6

### Describe the bug
I try to power on a Zanussi A/C (made by GREE). I use a pair of 38kHz transmitter and receiver.  
I tried to use CommonAcControl example and it didn't work.  
Then I tried to dump messages from the original remote.   
When I dump them by using `resultToHumanReadableBasic` I get two possible options:   
```
Protocol  : GREE
Code      : 0x0009205000C000F0 (64 Bits)
Protocol  : GREE
Code      : 0x0809605000C00070 (64 Bits)
```
When I dump them with `resultToSourceCode` - raw data is always different (423 bits always) and state has two options too:  
```
uint8_t state[8] = {0x00, 0x09, 0x20, 0x50, 0x00, 0xC0, 0x00, 0xF0};
uint8_t state[8] = {0x08, 0x09, 0x60, 0x50, 0x00, 0xC0, 0x00, 0x70};
```
I tried to put these values into IRSendDemo like this:  
```
#include <IRsend.h>

const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).

IRsend irsend(kIrLed);  // Set the GPIO to be used to sending the message.

// Example of data captured by IRrecvDumpV2.ino
uint16_t rawData[423] = {9014, 4466,  674, 554,  658, 550,  650, 552,  652, 554,  650, 554,  652, 552,  652, 114,  
228, 114,  746, 554,  652, 1632,  678, 552,  652, 554,  652, 1632,  674, 556,  652, 114,  
226, 116,  750, 552,  650, 556,  650, 554,  654, 552,  652, 554,  652, 554,  654, 550,  
652, 1632,  676, 552,  656, 550,  652, 554,  650, 554,  652, 552,  654, 552,  652, 1632,  674, 
554,  652, 1650,  654, 556,  650, 558,  648, 1654,  654, 554,  652, 19962,  676, 556,  650, 552,  
652, 554,  650, 554,  654, 552,  652, 552,  652, 554,  654, 552,  652, 554,  652, 554,  650, 554,  652, 554,  
652, 554,  652, 550,  654, 1630,  676, 1654,  654, 556,  652, 528,  678, 528,  676, 528,  674, 532,  674, 530,  
676, 532,  672, 532,  674, 530,  678, 528,  676, 526,  676, 530,  676, 1634,  676, 1648,  656, 1652,  654, 1632,  
672, 39998,  8992, 4468,  672, 532,  674, 534,  672, 532,  678, 528,  678, 528,  676, 532,  674, 530,  676, 534,  
674, 1650,  658, 528,  678, 528,  674, 1654,  652, 536,  672, 530,  672, 536,  672, 532,  672, 532,  672, 534,  676, 
532,  672, 532,  674, 532,  672, 1652,  654, 536,  668, 534,  674, 530,  672, 556,  652, 556,  648, 556,  650, 1660,  
648, 1656,  648, 1658,  650, 556,  650, 558,  648, 1658,  650, 554,  650, 19988,  652, 556,  648, 558,  648, 556,  648, 
558,  646, 558,  646, 558,  648, 556,  648, 560,  646, 558,  646, 558,  648, 556,  646, 558,  646, 560,  650, 556,  
646, 562,  642, 562,  646, 560,  644, 560,  644, 564,  640, 564,  640, 566,  640, 566,  638, 568,  614, 596,  610, 
618,  590, 614,  590, 616,  588, 620,  588, 1720,  588, 1718,  588, 616,  588, 618,  588, 40064,  8926, 4552,  590, 
620,  584, 620,  588, 620,  584, 618,  588, 618,  590, 614,  590, 616,  590, 616,  590, 614,  588, 620,  586, 614,  
590, 618,  588, 616,  588, 618,  588, 616,  586, 618,  590, 618,  586, 618,  588, 618,  590, 620,  584, 620,  586, 
618,  592, 614,  588, 618,  588, 620,  588, 618,  586, 618,  588, 618,  586, 618,  588, 1722,  586, 622,  584, 1720,  
586, 620,  584, 1718,  592, 616,  586, 20052,  592, 614,  590, 616,  590, 618,  586, 618,  588, 618,  586, 618,  588, 
618,  586, 618,  588, 620,  586, 618,  590, 614,  590, 616,  588, 618,  586, 618,  588, 618,  590, 616,  588, 618,  
586, 618,  588, 620,  586, 620,  596, 608,  588, 618,  588, 620,  586, 620,  586, 618,  586, 620,  586, 620,  588, 
614,  586, 622,  588, 1718,  586, 618,  588, 1720,  586};
// Example Samsung A/C state captured from IRrecvDumpV2.ino
uint8_t samsungState[8] = {0x08, 0x09, 0x60, 0x50, 0x00, 0xC0, 0x00, 0x70};

void setup() {
  irsend.begin();
#if ESP8266
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
#else  // ESP8266
  Serial.begin(115200, SERIAL_8N1);
#endif  // ESP8266
}

void loop() {
  Serial.println("NEC");
  irsend.sendNEC(0x0809605000C00070);
  delay(2000);
  Serial.println("Sony");
  irsend.sendSony(0xa90, 12, 2);  // 12 bits & 2 repeats
  delay(2000);
  Serial.println("a rawData capture from IRrecvDumpV2");
  irsend.sendRaw(rawData, 67, 38);  // Send a raw data capture at 38kHz.
  delay(2000);
  Serial.println("a Samsung A/C state from IRrecvDumpV2");
  irsend.sendSamsungAC(samsungState);
  delay(2000);
}
```
But it didn't work too. I tried to power up my transmitter by using breadboard power module with 12V-1A power adapter (I put a switch on a 5V output option). Still no effect.  

Where do I make a mistake? What is wrong?
Thanks in advance.

#### Expected behaviour
Power on A/C.

### What brand/model IR demodulator are you using?
https://aliexpress.com/item/1005003804455377.html?sku_id=12000027223801982

### Circuit diagram and hardware used (if applicable)
GND->GND
VCC->5v
DAT->GPIO23

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes

### Has this library/code previously worked as expected for you?
No

NiKiZe, 2024-03-15:
How is the transistor connected to drive the transmitter IR LED?, Do you have any circuit diagram of that "transmitt module"?

crankyoldgit, 2024-03-15:
`irsend.sendRaw(rawData, 423, 38);`

You forgot to change the size here.

stakanich, 2024-03-16:
> `irsend.sendRaw(rawData, 423, 38);`
> 
> You forgot to change the size here.

I fixed it and it still doesnt't work.

I tried to use `DumbIRRepeater` and `SmartIRRepeater` examples. It says that retransmits data but my A/C doesn't react.
Then I modified `DumbIRRepeater` to this:  
```
#include <Arduino.h>
#include <IRsend.h>
#include <IRrecv.h>
#include <IRremoteESP8266.h>
#include <IRutils.h>

// ==================== start of TUNEABLE PARAMETERS ====================

// The GPIO an IR detector/demodulator is connected to. Recommended: 14 (D5)
// Note: GPIO 16 won't work on the ESP8266 as it does not have interrupts.
// Note: GPIO 14 won't work on the ESP32-C3 as it causes the board to reboot.
#ifdef ARDUINO_ESP32C3_DEV
const uint16_t kRecvPin = 22;  // 14 on a ESP32-C3 causes a boot loop.
#else  // ARDUINO_ESP32C3_DEV
const uint16_t kRecvPin = 22;
#endif  // ARDUINO_ESP32C3_DEV

// GPIO to use to control the IR LED circuit. Recommended: 4 (D2).
const uint16_t kIrLedPin = 23;

// The Serial connection baud rate.
// NOTE: Make sure you set your Serial Monitor to the same speed.
const uint32_t kBaudRate = 115200;

// As this program is a special purpose capture/resender, let's use a larger
// than expected buffer so we can handle very large IR messages.
// i.e. Up to 512 bits.
const uint16_t kCaptureBufferSize = 1024;

// kTimeout is the Nr. of milli-Seconds of no-more-data before we consider a
// message ended.
const uint8_t kTimeout = 50;  // Milli-Seconds

// kFrequency is the modulation frequency all messages will be replayed at.
const uint16_t kFrequency = 38000;  // in Hz. e.g. 38kHz.

uint16_t rawData[423] = {9014, 4466,  674, 554,  658, 550,  650, 552,  652, 554,  650, 554,  652, 552,  652, 114,  
228, 114,  746, 554,  652, 1632,  678, 552,  652, 554,  652, 1632,  674, 556,  652, 114,  
226, 116,  750, 552,  650, 556,  650, 554,  654, 552,  652, 554,  652, 554,  654, 550,  
652, 1632,  676, 552,  656, 550,  652, 554,  650, 554,  652, 552,  654, 552,  652, 1632,  674, 
554,  652, 1650,  654, 556,  650, 558,  648, 1654,  654, 554,  652, 19962,  676, 556,  650, 552,  
652, 554,  650, 554,  654, 552,  652, 552,  652, 554,  654, 552,  652, 554,  652, 554,  650, 554,  652, 554,  
652, 554,  652, 550,  654, 1630,  676, 1654,  654, 556,  652, 528,  678, 528,  676, 528,  674, 532,  674, 530,  
676, 532,  672, 532,  674, 530,  678, 528,  676, 526,  676, 530,  676, 1634,  676, 1648,  656, 1652,  654, 1632,  
672, 39998,  8992, 4468,  672, 532,  674, 534,  672, 532,  678, 528,  678, 528,  676, 532,  674, 530,  676, 534,  
674, 1650,  658, 528,  678, 528,  674, 1654,  652, 536,  672, 530,  672, 536,  672, 532,  672, 532,  672, 534,  676, 
532,  672, 532,  674, 532,  672, 1652,  654, 536,  668, 534,  674, 530,  672, 556,  652, 556,  648, 556,  650, 1660,  
648, 1656,  648, 1658,  650, 556,  650, 558,  648, 1658,  650, 554,  650, 19988,  652, 556,  648, 558,  648, 556,  648, 
558,  646, 558,  646, 558,  648, 556,  648, 560,  646, 558,  646, 558,  648, 556,  646, 558,  646, 560,  650, 556,  
646, 562,  642, 562,  646, 560,  644, 560,  644, 564,  640, 564,  640, 566,  640, 566,  638, 568,  614, 596,  610, 
618,  590, 614,  590, 616,  588, 620,  588, 1720,  588, 1718,  588, 616,  588, 618,  588, 40064,  8926, 4552,  590, 
620,  584, 620,  588, 620,  584, 618,  588, 618,  590, 614,  590, 616,  590, 616,  590, 614,  588, 620,  586, 614,  
590, 618,  588, 616,  588, 618,  588, 616,  586, 618,  590, 618,  586, 618,  588, 618,  590, 620,  584, 620,  586, 
618,  592, 614,  588, 618,  588, 620,  588, 618,  586, 618,  588, 618,  586, 618,  588, 1722,  586, 622,  584, 1720,  
586, 620,  584, 1718,  592, 616,  586, 20052,  592, 614,  590, 616,  590, 618,  586, 618,  588, 618,  586, 618,  588, 
618,  586, 618,  588, 620,  586, 618,  590, 614,  590, 616,  588, 618,  586, 618,  588, 618,  590, 616,  588, 618,  
586, 618,  588, 620,  586, 620,  596, 608,  588, 618,  588, 620,  586, 620,  586, 618,  586, 620,  586, 620,  588, 
614,  586, 622,  588, 1718,  586, 618,  588, 1720,  586};

// ==================== end of TUNEABLE PARAMETERS ====================

[... 55 lines cut ...]
}
```

I tried to check my LED by this: it will send data and receiver will try to receive it.
The output:
```
000003.490: A message that was 427 entries long was retransmitted.
000006.947: A message that was 419 entries long was retransmitted.
000010.406: A message that was 419 entries long was retransmitted.
000013.862: A message that was 421 entries long was retransmitted.
000017.322: A message that was 417 entries long was retransmitted.
000020.780: A message that was 421 entries long was retransmitted.
000024.237: A message that was 421 entries long was retransmitted.
000027.697: A message that was 415 entries long was retransmitted.
⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮19 entries long was retransmitted.
000034.616: A message that was 417 entries long was retransmitted.
000038.072: A message that was 417 entries long was retransmitted.
000041.531: A message that was 415 entries long was retransmitted.
000044.987: A message that was 421 entries long was retransmitted.
000048.448: A message that was 417 entries long was retransmitted.
```

@NiKiZe , I am not sure that there is a transistor here but I don't know a lot about electronic.
[image]
[image]

stakanich, 2024-03-16:
Both receiver and transmitter have LED's on their boards. They are connected to a data-line as far as I understand. And both of them blink in sequence with `DumbIRRepeater` example. So it looks like they work.

NiKiZe, 2024-03-16:
I did not see any driver transistor on that board, which is why I was asking. Since without it you won't have enough "power" on the transmitter LED, or at least not unless it is right next to the receiver.

stakanich, 2024-03-16:
> I did not see any driver transistor on that board, which is why I was asking. Since without it you won't have enough "power" on the transmitter LED, or at least not unless it is right next to the receiver.

5 volts from 5 volts pin on a esp32 wouldn't be enough? O_o I thought that as far as I power vcc pin of transmitter with this power pin - it will work...

stakanich, 2024-03-16:
Oh...I think, I see now. VCC pin on transmitter doesn't has a line! It is useless!

mr-miky, 2024-12-14:
Could you tell me the model of the remote control?

YAN1F6 ??

If this is the case, the transmission protocol has more bits than the library thinks...

Aermec has this remote control which is actually GREE.

Attached files to be displayed with PulseView
[IR_capture.zip](https://github.com/user-attachments/files/18137623/IR_capture.zip)

mr-miky, 2024-12-14:
I notice something strange here:

uint16_t rawData[423] = {9014, 4466,  674, 554,  658, 550,  650, 552,  652, 554,  650, 554,  652, 552,  652, _**114, 
228, 114**_,  746, 554,  652, 1632,  678, 552,  652, 554,  652, 1632,  674, 556,  652, **_114,  
226, 116_**,

After a mark of 650 us there should be a space of 550 or 1600 us depending if you transmit a bit 0 or 1

114 + 228 + 114  + 746 = 1202  as  650 + 550 !

try to correct those 2 strange sequences

652, (114 + 228 + 114 + 100) , (746 -100), 554 

652, (114 + 226 + 116 + 100),  /750 - 100) , 552

Obviously the rawData[] to send are minus 4.

irsend.sendRaw(rawData, 423 - 4 , 38);  // Send a raw data capture at 38kHz.