crankyoldgit/IRremoteESP8266 #2083: Unable to include C library

karantejassp, 2024-04-13:
### Version/revision of the library used
2.8.6

### Describe the bug
I'm trying to compile example files for receiving IR signals using the Arduino Cloud IDE. The inlude for many header files is missing while trying to compile. The error message is like the following: 
```
In file included from /mnt/create-efs/webide/9e/87/9e879e4336aea3a9e53fc1faf5565892:karantejas/libraries_v2/IRremoteESP8266/src/IRac.h:15:0,

from /mnt/create-efs/webide/9e/87/9e879e4336aea3a9e53fc1faf5565892:karantejas/libraries_v2/IRremoteESP8266/src/IRac.cpp:7:

/mnt/create-efs/webide/9e/87/9e879e4336aea3a9e53fc1faf5565892:karantejas/libraries_v2/IRremoteESP8266/src/ir_Argo.h:14:10: fatal error: set: No such file or directory

#include <set>

^~~~~

compilation terminated.

```
I actually faced this issue with IDE on my laptop. To check if it was a issue with my laptop, I used the Arduino cloud.

#### To Reproduce
_1. Open the Arduino Cloud IDE._
_2. Upload the IRremoteESP8266 Library._
_3. Compile the IRecvDump3 example._

#### Example code used
IRecvDump3

NiKiZe, 2024-04-13:
So you have this issue both with local Arduino studio, and cloud?

Which device are you trying to build for?

karantejassp, 2024-04-13:
Hey @NiKiZe, 
Yeah I'm facing it for both Arduino Studio and cloud.
I'm using a Arduino Uno board, but I haven't yet plugged the device, I'm just trying to compile the code.

NiKiZe, 2024-04-13:
> I'm using a Arduino Uno board, 

So reading https://github.com/crankyoldgit/IRremoteESP8266/blob/master/README.md
> This library enables you to send and receive infra-red signals on an ESP8266 or an ESP32 using the Arduino framework

Try using a library suitable for your device, more info in FAQ and other issues, please try the search function ;)

closed 2024-04-13 by NiKiZe