crankyoldgit/IRremoteESP8266 #2095: Feature Request: De'Longhi Pinguino PAC N82 ECO IR Protocol

MrGoodbody, 2024-05-18:
Hi,
I recently had to look at the protocol for a "De'Longhi Pinguino PAC N82 ECO" A/C. 
I couldn't find any documentation for it and had to analyze it manually. 
In the repositiory you can already find a Delonghi protocol that does not work for the "N series" (at least the N82). 
Unfortunately I am not a programmer, so I can only help with the information I have gathered. I am attaching the picture of the remote control and an ODS document with the necessary information.
Gladly help if you need more information.

[De'Longhi Pinguino PAC N82 ECO IR Protocol.ods](https://github.com/crankyoldgit/IRremoteESP8266/files/15362908/De.Longhi.Pinguino.PAC.N82.ECO.IR.Protocol.ods)

The remote keeps track of all settings and every buttonpush send all information to the ac
[image]

NiKiZe, 2024-05-18:
The issue template have several parts of information that we want you to provide, some of it is to collect raw data, we will want to have the issue directly.

Other information should be in an online spreadsheet so we don't need a special program, is Google sheets an alternative you could use?

MrGoodbody, 2024-05-18:
I'm aware of the problem template, but since I can't share a spreadsheet anonymously and don't want to reveal my identity, I can only provide a screenshot of the spreadsheet.

[image]

perhaps my request was too imprecisely defined. the protocol is know (NEC_LIKE), i would like to request a device integration.
Do you still need raw data? unfortunately i am not able to do this

NiKiZe, 2024-05-18:
Without raw data it is difficult to verify, and also we won't get the normal cases for our automated tests.

MrGoodbody, 2024-05-18:
OK, I understand, unfortunately I am not able to obtain the data for various reasons. Sorry to have wasted your time. Thanks anyway.

closed 2024-05-18 by MrGoodbody

NiKiZe, 2024-05-18:
I do not think this is waste of time, you have excellent information, and hopefully someone will be able to provide the raw data, and once that happens the information you have provided will make the rest a breeze. (You have provided the most difficult parts!)