crankyoldgit/IRremoteESP8266 #2101: The implementation and declaration of some functions differs from the function call

MarkKoshelev, 2024-06-04:
I encountered this when my AC Haier was not working correctly.
src/IRac.cpp
1460: void IRac::haier176(IRHaierAC176 *ac, const haier_ac176_remote_model_t model,
                    const bool on, const stdAc::opmode_t mode,
                    const bool celsius, const float degrees,
                    const stdAc::fanspeed_t fan,
                    const stdAc::swingv_t swingv,
                    const stdAc::swingh_t swingh,
                    const bool turbo, const bool quiet, const bool filter,
                    const int16_t sleep)

3289:     haier176(&ac, (haier_ac176_remote_model_t)send.model, send.power,
               send.mode, send.celsius, send.degrees, send.fanspeed,
               send.swingv, send.swingh, send.turbo, send.filter, send.sleep);

The call of function  haier176 dont have "const bool quiet" argument

closed 2024-06-05 by crankyoldgit

crankyoldgit, 2024-06-06:
@MarkKoshelev I think I've fixed it. Can you please confirm it now operates as expected in the master branch?

MarkKoshelev, 2024-07-02:
> @MarkKoshelev I think I've fixed it. Can you please confirm it now operates as expected in the master branch?

You have fixed ony one function (haier176). But the same bug is in a nubber of functions.
I see them in haierYrwo2 also. 
Please check all functions declaration and implementation. All haierXXX function have this bug.

NiKiZe, 2024-07-03:
> You have fixed ony one function (haier176). But the same bug is in a nubber of functions. I see them in haierYrwo2 also. Please check all functions declaration and implementation. All haierXXX [...]

Feel free to create one or more PRs 😉

MarkKoshelev, 2024-07-03:
Hi

I use tasmota project code. It uses IRremoteESP8266 v2.8.4.

After the same fixes locally is OK with haier176.

Its look like correctly with the another haier AC now.

PR #2102 (2024-06-05): Fix missing quiet parameter of haier176
Try to fix #2101
commit 0b0fe778 (2024-06-05): Fix missing quiet parameter of haier176 (#2102)

Try to fix #2101
commit 702ec8b3 (2024-07-03): Update haier160 & HaierYRWO2 to use quiet in the common class. (#2115)

Quiet mode operation was missing in the function calls of `haierYrwo2()` and `haier160()` thus passing the wrong arguments to the function.
Add the `quiet` parameter in fixes this.

Fixes #2101