crankyoldgit/IRremoteESP8266 #2116: Using any other mode than "Cool" causes AC to turn off.

Kulasekaran-148, 2024-07-06:
### Version/revision of the Library used
2.8.6

### Describe the bug
Using any other mode than "Cool" causes AC to turn off. I understand different modes will require other parameters to be of a particular value in ACs. For eg. if we're using fan mode, then fanspeed will get set to max in some ACs. 

#### To Reproduce
use the AC.setMode() function and pass in different mode value other than "Cool" and also at the same time try and configure other parameters like AC.setTemp() with a different temperature value rather than the one the AC remote sets when you press the mode button on the AC remote.

#### Example code used
```
        ac_daikin216.setPower(gwy_ac_control_t.control.power);
        ac_daikin216.setTemp(gwy_ac_control_t.control.temp);
        if (gwy_ac_control_t.control.swingH)
            gwy_ac_control_t.control.swingH = kDaikinSwingOn;
        ac_daikin216.setSwingHorizontal(gwy_ac_control_t.control.swingH);
        if (gwy_ac_control_t.control.swingV)
            gwy_ac_control_t.control.swingV = kDaikinSwingOn;
        ac_daikin216.setSwingVertical(gwy_ac_control_t.control.swingV);
        ac_daikin216.setFan(ac_daikin216.convertFan((stdAc::fanspeed_t)gwy_ac_control_t.control.fanSpeed));
        ac_daikin216.setMode(ac_daikin216.convertMode((stdAc::opmode_t)gwy_ac_control_t.control.mode_val));
        ac_daikin216.send();
```

### Expected behaviour
I'm expecting the AC to change mode when AC.setMode is used.

### What brand/model IR demodulator are you using?
TSOP38538

### I have followed the steps in the [Troubleshooting Guide]
Yes.

### Has this library/code previously worked as expected for you?
Yes.

### Other useful information
None.

crankyoldgit, 2024-07-06:
Can you please simplify your code to plain literals so we can see what values you are passing. Also what brand/model AC & remote are you using? 

And finally, could you please capture the equivalent signals from the original remote using the IRrecvDumpV2 or V3 example code. That way we can see what signal your ac requires/expects.

Kulasekaran-148, 2024-07-06:
Sure, I can provide you in a few mins, but here are some details that I can give you right away. 

- I'm trying to control Daikin and Hitachi. I know from using the IR-recv-dump example, that my ACs are expecting Daikin216 and Hitachi296 protocols.

The problem is what is the exact procedure I need to follow to change modes in the above two ACs. 

When using the AC.setMode() function, Am I allowed to call other functions like AC.setTemp or AC.setFan before finally called the AC.send() function ?

crankyoldgit, 2024-07-06:
Yes you are able to do that

Kulasekaran-148, 2024-07-06:
### AC details
- AC Brand : Daikin
- AC Remote model : ARC484B32
- AC Protocol being used : Daikin216

### Here's the AC setting set using original remote before blasting the command
- Power : ON
- Mode : Cool
- Temp : 26
- Fan : Max
- SwingH : ON

### Here's the IR RAW value that is being blasted from my device : 
```
1, 1732, 868, 217, 645, 219, 218, 219, 218, 213, 225, 206, 677, 102, 387, 90, 38, 88, 276, 205, 230, 217, 223, 219, 721, 179, 347, 87, 29, 92, 665, 217, 688, 165, 232, 173, 695, 216, 761, 89, 742, 198, 831, 87, 751, 131, 188, 218, 317, 119, 46, 91, 796, 91, 352, 90, 251, 214, 391, 157, 240, 223, 234, 118, 2091, 221, 785, 99, 636, 206, 220, 217, 220, 217, 220, 243, 193, 244, 193, 245, 192, 244, 192, 246, 192, 243, 192, 245, 191, 214, 222, 214, 223, 214, 222, 214, 223, 245, 191, 220, 221, 241, 194, 243, 194, 243, 194, 217, 220, 243, 193, 218, 221, 216, 218, 219, 217, 215, 222, 214, 647, 216, 220, 217, 221, 215, 220, 217, 219, 217, 220, 217, 220, 217, 14839, 1752, 842, 216, 645, 218, 219, 218, 218, 218, 221, 216, 643, 215, 221, 216, 221, 216, 220, 216, 220, 216, 646, 218, 219, 218, 643, 220, 641, 217, 220, 217, 645, 219, 643, 220, 641, 217, 644, 214, 648, 215, 221, 216, 220, 217, 647, 216, 218, 218, 218, 245, 193, 244, 191, 246, 192, 245, 191, 220, 219, 217, 219, 243, 193, 218, 218, 245, 192, 218, 218, 219, 218, 219, 218, 219, 218, 219, 218, 219, 217, 215, 221, 215, 224, 218, 216, 216, 221, 221, 216, 215, 221, 216, 221, 216, 220, 217, 220, 217, 220, 216, 220, 217, 219, 217, 219, 218, 218, 218, 218, 218, 218, 219, 218, 219, 220, 217, 217, 214, 222, 220, 217, 214, 223, 214, 222, 214, 222, 215, 221, 216, 222, 214, 221, 216, 221, 216, 221, 216, 221, 216, 220, 217, 219, 218, 221, 216, 220, 216, 220, 217, 219, 217, 219, 217, 220, 217, 219, 217, 219, 218, 218, 218, 219, 218, 219, 218, 218, 219, 218, 219, 218, 219, 217, 214, 222, 220, 226, 221, 224, 218, 221, 221, 215, 216, 220, 217, 220, 217, 220, 217, 221, 217, 220, 216, 222, 215, 219, 218, 219, 218, 219, 218, 218, 218, 229, 219, 213, 219, 218, 219, 219, 218, 217, 219, 217, 219, 217, 214, 224, 218, 219, 218, 218, 218, 218, 218, 218, 219, 218, 219, 218, 245, 191, 219, 218, 214, 222, 215, 221, 242, 195, 215, 223, 219, 215, 216, 220, 217, 219, 217, 219, 218, 218, 218, 643, 215, 647, 216, 221, 221, 218, 213, 221, 218, 219, 218, 217, 218, 220, 216, 219, 218, 223, 245, 190, 242, 195, 242, 194, 243, 194, 216, 220, 243, 196, 246, 188, 243, 193, 217, 219, 244, 617, 215, 222, 215, 222, 215, 647, 217, 219, 217, 645, 219, 643, 215,
```

### Here's the AC settings set using the device
- Power : ON
- Mode : Heat
- Temp 25
- Fan : Max
- SwingH : ON

### Expected behavior of the AC
- To switch to Heat mode

### Actual behavior of the AC 
- AC is getting turned off

Kulasekaran-148, 2024-07-06:
OK I think I made some errors during my testing. It seems to be working now. I'll close this issue for now and then reopen in case I face it in the future again. Thanks @crankyoldgit for replying very soon. I never expected the reply to come in this soon. By the your repo has been a life saver for me in my project. Keep up the good work. 👍

closed 2024-07-06 by Kulasekaran-148

NiKiZe, 2024-07-06:
If you have issues that you think is due to incorrect commands, be sure to include data from your original remote captured with `IRrecvDumpV3`, we most likely are not helped by the capture of what the library generates.