crankyoldgit/IRremoteESP8266 #2117: Solution to AC remote giving different data for same button pressed

AMHKM-1, 2024-07-13:
Hi everyone,

I'm not sure if this is common knowledge or a new discovery, but I'm confident this will help many of you.

For the past two days, I was trying to get the hex code and raw data sent by my IR remote. Unfortunately, the same button (e.g., the temperature up button) kept giving different data each time. This made me think something was wrong with my VS1838B receiver. Here’s what I discovered:

AC remotes are not like other remotes. Typical remotes have a single code assigned to a button, but AC remotes send codes depending on the state of the AC. For example, my AC with cool mode on and temperature set to 25°C has a specific IR signal assigned to it, while cool mode at 26°C has another code. So, if your AC is in cool mode at 24°C and you press the temperature up button, the hex code for 25°C cool mode is sent. If you press the same button again, another hex code is sent, even though you pressed the same button.

Practically, this means you need to capture every hex code for each specific AC state (temperature and mode) you want to implement. Feel free to ask any questions. A big thanks to crankyoldgit for this amazing library!

NiKiZe, 2024-07-14:
Yes that is expected.
There is an issue template that you are expected to fill when creating an issue. It also has links to the FAQ.

Why are you even opening an issue? (What is it that you want us to do?)

closed 2024-07-14 by NiKiZe

crankyoldgit, 2024-07-14:
> I'm not sure if this is common knowledge or a not, but I'm confident this will help many of you.

Yep. Fairly common knowledge. See: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#a-note-on-collecting-data