crankyoldgit/IRremoteESP8266 #2123: [XIAO_ESP32C6] IRrecv.cpp:246:3: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?

cadesalaberry, 2024-07-27:
### Version/revision of the library used
2.8.6

### Describe the bug
When running the verify command on Arduino IDE, it refuses to compile. It seems to be related to recent modifications in the arduino framework as mentioned here: https://github.com/crankyoldgit/IRremoteESP8266/issues/2053

#### To Reproduce
- Open Arduino IDE
- Pick `XIAO_ESP32C6` as the target board
- Click on Verify

#### Example code used
```c
#include <IRremoteESP8266.h>
#include <IRrecv.h>
#include <IRutils.h>

// Define the IR receiver pin
const uint16_t kRecvPin = 10;

// IR Receiver setup
IRrecv irrecv(kRecvPin);
decode_results results;

void setup() {
  Serial.begin(115200);

  Serial.print("Starting IR receiver... ");
   // Initialize IR Receiver
  irrecv.enableIRIn();
  Serial.print("OK");
}

void loop() {
  // Check if an IR signal is received
  if (irrecv.decode(&results)) {
    Serial.println(resultToHumanReadableBasic(&results));
    irrecv.resume(); // Receive the next value
  }
}
```

#### Expected behaviour
I would expect the verify step to successfully compile.

#### Output of raw data from [IRrecvDumpV2.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) or [V3](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV3/) (if applicable)
```
/Users/my_user/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp: In function 'void gpio_intr()':
/Users/my_user/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp:246:3: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
  246 |   timerAlarmEnable(timer);
      |   ^~~~~~~~~~~~~~~~
      |   timerAlarm
/Users/my_user/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp: In member function 'void IRrecv::enableIRIn(bool)':
/Users/my_user/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp:362:21: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
  362 |   timer = timerBegin(_timer_num, 80, true);
      |           ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/my_user/Library/Arduino15/packages/esp32/hardware/esp32/3.0.2/cores/esp32/esp32-hal.h:84,
                 from /Users/my_user/Library/Arduino15/packages/esp32/hardware/esp32/3.0.2/cores/esp32/Arduino.h:36,
                 from /Users/my_user/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.h:10,
                 from /Users/my_user/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp:6:
/Users/my_user/Library/Arduino15/packages/esp32/hardware/esp32/3.0.2/cores/esp32/esp32-hal-timer.h:35:13: note: declared here
   35 | hw_timer_t *timerBegin(uint32_t frequency);
      |             ^~~~~~~~~~
/Users/my_user/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp:371:3: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
  371 |   timerAlarmWrite(timer, MS_TO_USEC(params.timeout), ONCE);
      |   ^~~~~~~~~~~~~~~
      |   timerWrite
/Users/my_user/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp:375:23: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
  375 |   timerAttachInterrupt(timer, &read_timeout, false);
      |   ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/my_user/Library/Arduino15/packages/esp32/hardware/esp32/3.0.2/cores/esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
/Users/my_user/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp: In member function 'void IRrecv::disableIRIn()':
/Users/my_user/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp:401:3: error: 'timerAlarmDisable' was not declared in this scope
  401 |   timerAlarmDisable(timer);
      |   ^~~~~~~~~~~~~~~~~
/Users/my_user/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp: In member function 'void IRrecv::pause()':
/Users/my_user/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp:416:3: error: 'gpio_intr_disable' was not declared in this scope; did you mean 'esp_intr_disable'?
  416 |   gpio_intr_disable((gpio_num_t)params.recvpin);
      |   ^~~~~~~~~~~~~~~~~
      |   esp_intr_disable
/Users/my_user/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp: In member function 'void IRrecv::resume()':
/Users/my_user/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp:429:3: error: 'timerAlarmDisable' was not declared in this scope
  429 |   timerAlarmDisable(timer);
      |   ^~~~~~~~~~~~~~~~~
/Users/my_user/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp:430:3: error: 'gpio_intr_enable' was not declared in this scope; did you mean 'esp_intr_enable'?
  430 |   gpio_intr_enable((gpio_num_t)params.recvpin);
      |   ^~~~~~~~~~~~~~~~
      |   esp_intr_enable

exit status 1

Compilation error: exit status 1
```

### Circuit diagram and hardware used (if applicable)

XIAO_ESP32C6

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes_

### Has this library/code previously worked as expected for you?
_No_

### Other useful information

I am a bit rusty in C, it dates back to my Electrical Engineering Classes in 2015.
I am trying to build a volume controller for my Devialet speaker to control it with an AppleTV remote, copying the behaviour of a working [PoC](https://github.com/cadesalaberry/phantom-remote-control/tree/feature/use-official-devialet-api) in NodeJS.

cadesalaberry, 2024-07-27:
I went ahead an tried another branch of the library from this PR: https://github.com/crankyoldgit/IRremoteESP8266/pull/2040
Doing the following (On a Mac M2) seems to have solved my problem:
```
cd ~/Documents/Arduino/libraries/
rm -rf IRremoteESP8266/
git clone git@github.com:tonhuisman/IRremoteESP8266.git
cd IRremoteESP8266/
git switch feature/C++20-compatibility
```
Then I restarted my IDE, pressed "Verify" which led to a successful build 🎉 

@tonhuisman do you know what is missing to merge your PR?

NiKiZe, 2024-07-27:
Duplicate of #2039 #2122
Search: https://github.com/search?q=repo%3Acrankyoldgit%2FIRremoteESP8266+timerAlarmEnable&type=issues
Please always search first, and don't create duplicate issues.

closed 2024-07-27 by NiKiZe

cadesalaberry, 2024-07-27:
For others coming this way, here is my `PlatformIO` config to build this library for XIAO_ESP32C6:
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32-c6-devkitm-1]
board = esp32-c6-devkitm-1
platform = espressif32
platform_packages =
	platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.3
	platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/esp32-arduino-libs.git#idf-release/v5.1
framework = arduino
monitor_speed = 115200
lib_deps =
  https://github.com/tonhuisman/IRremoteESP8266/archive/feature/C++20-compatibility.zip

```

I also had to edit the `esp32-c6-devkitm-1.json` file to add `arduino` in the supported frameworks:
```json
// ~/.platformio/platforms/expressif32/boards/esp32-c6-devkitc-1.json
{
  "build": {
    "core": "esp32",
    "f_cpu": "160000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "mcu": "esp32c6",
    "variant": "esp32c6"
  },
  "connectivity": ["wifi"],
  "debug": {
    "openocd_target": "esp32c6.cfg"
  },
  "frameworks": ["espidf", "arduino"],
  "name": "Espressif ESP32-C6-DevKitM-1",
  "upload": {
    "flash_size": "4MB",
    "maximum_ram_size": 327680,
    "maximum_size": 4194304,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitm-1/index.html",
  "vendor": "Espressif"
}
```