crankyoldgit/IRremoteESP8266 #2127: Support for LG AKB75735421 (for models newer than 2022)

UnforeseenOcean, 2024-08-08:
### Version/revision of the library used
v2.8.6

### Describe the bug
It seems like this remote sends a signal recognized as two separate remotes, causing confusion during replay. This may require some investigation.

#### To Reproduce
1. Get a AKB75735421 remote (for cooling/heating, as it says on the back; Region: Korea) Since I have one, I can help you with dumping data from it. (Note: This remote is used for many other air conditioner models, including ceiling cartridges that use shared outside compressor unit.)
2. Try to use the built-in IR remote dump sketch, any version past v2 has the same symptom.

#### Example code used
Refer to built-in example, `IRrecvDumpV2.ino` or `IRrecvDumpV3.ino` with the input pin changed from 14 to 13.

#### Expected behaviour
A single cohesive data is extracted from the stream, or the remote model is properly handled.

#### Output of raw data
This is the result using the cheap decoder. Action of turning on the air purifier function while the air conditioner is running in cooling mode at 30C, with wind strength set to maximum, automatic drying, robot cleaning, intelligent power saving and horizontal sweep function enabled:
```
Timestamp : 000120.623
Library   : v2.8.6

Protocol  : LG2
Code      : 0x88C000C (28 Bits)
Mesg Desc.: Model: 2 (AKB75215403), Power: Off
uint16_t rawData[59] = {3158, 9944,  506, 1596,  476, 554,  504, 560,  480, 560,  478, 1600,  504, 536,  480, 560,  506, 534,  478, 1610,  478, 1602,  474, 558,  526, 534,  504, 534,  478, 560,  480, 560,  474, 558,  498, 556,  500, 558,  498, 556,  528, 534,  504, 536,  504, 536,  502, 536,  474, 556,  500, 1602,  504, 1610,  478, 560,  478, 560,  478};  // LG2 88C000C
uint32_t address = 0x88;
uint32_t command = 0xC000;
uint64_t data = 0x88C000C;

Timestamp : 000120.849
Library   : v2.8.6

Protocol  : LG2
Code      : 0x88FF3EF (28 Bits)
Mesg Desc.: Model: 3 (AKB74955603), Power: Off
uint16_t rawData[59] = {3184, 10054,  496, 1580,  532, 532,  504, 536,  504, 534,  502, 1586,  476, 558,  500, 570,  502, 544,  506, 1584,  478, 1610,  478, 1610,  478, 1610,  476, 1610,  506, 1610,  478, 1600,  480, 1618,  476, 552,  530, 536,  478, 1602,  480, 1590,  504, 1586,  502, 1616,  506, 1590,  474, 556,  504, 1600,  528, 1576,  504, 1608,  504, 1584,  478};  // LG2 88FF3EF
uint32_t address = 0x88;
uint32_t command = 0xFF3E;
uint64_t data = 0x88FF3EF;
```
NOTE: This is from ONE remote and reads consistently as two remotes. No other remotes were involved. And that's the problem.

Result with a different decoder, probably TSOP34438. The same action of turning on the air purifier function while the air conditioner is running in cooling mode at 30C, with wind strength set to maximum, automatic drying, robot cleaning, intelligent power saving, and horizontal sweep function enabled has been performed.
```
Timestamp : 000150.229
Library   : v2.8.6

Protocol  : LG2
Code      : 0x88C000C (28 Bits)
Mesg Desc.: Model: 2 (AKB75215403), Power: Off
uint16_t rawData[59] = {3172, 9928,  472, 1628,  506, 552,  478, 562,  478, 562,  478, 1608,  478, 560,  480, 562,  478, 562,  478, 1602,  478, 1608,  480, 560,  478, 562,  478, 562,  478, 562,  478, 560,  480, 560,  478, 560,  478, 562,  452, 586,  478, 562,  478, 560,  454, 586,  478, 562,  478, 560,  454, 1626,  478, 1600,  480, 560,  478, 562,  478};  // LG2 88C000C
uint32_t address = 0x88;
uint32_t command = 0xC000;
uint64_t data = 0x88C000C;

Timestamp : 000150.453
Library   : v2.8.6

Protocol  : LG2
Code      : 0x88FF3AB (28 Bits)
Mesg Desc.: Model: 3 (AKB74955603), Power: Off
uint16_t rawData[59] = {3106, 9976,  472, 1628,  506, 560,  478, 562,  478, 562,  478, 1606,  506, 552,  478, 550,  506, 550,  504, 1602,  478, 1590,  510, 1598,  478, 1598,  532, 1554,  534, 1572,  480, 1602,  514, 1562,  534, 524,  480, 556,  534, 1582,  480, 1600,  480, 1602,  478, 558,  558, 1548,  480, 556,  560, 1538,  480, 568,  480, 1598,  560, 1538,  480};  // LG2 88FF3AB
uint32_t address = 0x88;
uint32_t command = 0xFF3A;
uint64_t data = 0x88FF3AB;
```

### What brand/model IR demodulator are you using?
Unknown, presumed to be TSOP34438 and xx1838. Extracted from a cheap IR receiver module and a TV.

### Circuit diagram and hardware used (if applicable)
(Note: ESP32 is shown due to Wokwi's limitations, but the board used is Lolin D1, which is an ESP8266 board. Pin assignment is identical.)
[image]

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes, same setup has worked fine with other 36KHz remotes despite the potentially problematic receiver

### Has this library/code previously worked as expected for you?
Unsure, however it did work fine for other protocols (NEC)

NiKiZe, 2024-08-08:
Yould you be more specific on exactly which data is incorrect, or which features you think is missing for your remote?

Adding a remote model is the same as adding "detailed AC support" covered in the FAQ.

Also make a search for `LG2` and check existing issues and discussions so we don't have duplicates.

UnforeseenOcean, 2024-08-08:
> Specify which data is incorrect or which features are probably missing for the remote
The data for additional functions (listed in the original issue) are not decoded properly and returns "Power: Off" instead.

> Check for duplicates by searching for "LG2" in the issues tab
Done, it does not seem to have any match above 90%.

If you need data from this remote, tell me what to do and I will do it and post the result here or on a separate repository or Gist.

UnforeseenOcean, 2024-08-08:
I am currently dumping (nearly) all combinations of the functions, this may take some time.