crankyoldgit/IRremoteESP8266 #2133: Argo WREM2 v1 commands not recognized

mateuszdrab, 2024-08-29:
### Version/revision of the library used
v2.8.6

### Describe the bug
I also seem to be experiencing decoding issues with a Technibel WREM2 v1 remote (similarly to https://github.com/crankyoldgit/IRremoteESP8266/issues/1859)

Symptom is that all IR communications received by Tasmota (and hence the library) are decoded as iFeel Temperature Reports instead of actual command messages. The provided reading for the sensor is erroneous (when misinterpreting command messages, otherwise the iFeel readings from respective reports are accurate and correct), suggesting that the message bytes are not an actual iFeel report but an actual command which is misinterpreted by the library.

#### Example code used
This issue can be reproduced on both Tasmota (tasmota-ir, version 14.2 and when using [IRrecvDumpV2.ino].(https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) pulled from master.

#### Expected behaviour
Commands decoded correctly and not as iFeel reports

#### Output of raw data from [IRrecvDumpV2.ino]
```
iFeel enabled unless otherwise indicated
Temperature at remote 20
Target temp is 10
Cooling mode is on

# Turn off

Timestamp : 000036.015
Library   : v2.8.6

Protocol  : ARGO
Code      : 0xACF58039 (32 Bits)
Mesg Desc.: Model: 1 (WREM2), Sensor Temp: 20C
uint16_t rawData[183] = {6422, 3190,  408, 872,  428, 848,  428, 2126,  402, 2150,  408, 872,  404, 2152,  402, 874,  402, 2152,  402, 2154,  406, 872,  428, 2126,  404, 876,  404, 2150,  430, 2124,  432, 2122,  432, 2124,  404, 874,  404, 874,  426, 852,  428, 850,  412, 864,  406, 872,  404, 874,  400, 2154,  428, 2128,  402, 876,  402, 874,  400, 2154,  404, 2150,  406, 2150,  400, 878,  400, 876,  402, 874,  416, 2136,  404, 2152,  402, 876,  402, 874,  402, 874,  428, 852,  402, 874,  404, 872,  400, 876,  402, 876,  404, 872,  430, 846,  402, 2152,  406, 2150,  406, 2150,  404, 874,  432, 846,  426, 850,  428, 850,  400, 878,  398, 876,  404, 874,  404, 874,  400, 2152,  432, 2124,  428, 2128,  432, 846,  426, 852,  400, 2154,  404, 874,  426, 2128,  428, 2128,  404, 872,  430, 848,  426, 2128,  404, 872,  414, 2140,  432, 848,  400, 876,  426, 850,  404, 872,  402, 876,  400, 876,  430, 848,  404, 872,  404, 874,  402, 2154,  402, 876,  428, 2126,  406, 872,  426, 2128,  426, 852,  404, 872,  430, 2124,  404, 874,  430, 846,  426, 2128,  400};  // ARGO
uint8_t state[4] = {0xAC, 0xF5, 0x80, 0x39};

# Turn on

Timestamp : 000038.077
Library   : v2.8.6

Protocol  : ARGO
Code      : 0xACF58039 (32 Bits)
Mesg Desc.: Model: 1 (WREM2), Sensor Temp: 20C
uint16_t rawData[183] = {6418, 3192,  406, 872,  432, 846,  402, 2154,  430, 2124,  432, 846,  404, 2150,  406, 870,  408, 2148,  406, 2148,  428, 850,  402, 2152,  430, 848,  404, 2150,  428, 2126,  404, 2150,  406, 2148,  404, 876,  428, 846,  430, 848,  428, 848,  428, 852,  428, 848,  402, 876,  426, 2128,  432, 2124,  406, 870,  404, 874,  430, 2126,  432, 2124,  404, 2150,  406, 872,  406, 870,  428, 848,  402, 2154,  430, 2126,  428, 850,  426, 848,  432, 846,  428, 852,  426, 850,  430, 848,  404, 874,  404, 874,  404, 874,  404, 872,  426, 2126,  404, 2152,  404, 2152,  430, 846,  434, 846,  404, 870,  408, 872,  430, 848,  404, 870,  402, 874,  432, 844,  406, 2150,  406, 2150,  432, 2120,  402, 878,  428, 848,  430, 2122,  406, 870,  406, 2150,  406, 2150,  406, 870,  430, 848,  400, 2154,  406, 870,  406, 2148,  406, 872,  432, 846,  404, 872,  428, 848,  404, 874,  404, 872,  402, 874,  402, 2154,  426, 852,  402, 2152,  406, 876,  424, 2128,  428, 850,  402, 2152,  406, 874,  400, 878,  402, 2150,  404, 2152,  428, 850,  404, 2150,  406};  // ARGO
uint8_t state[4] = {0xAC, 0xF5, 0x80, 0x39};

# Target temp +1 (11)
Timestamp : 000049.843
Library   : v2.8.6

Protocol  : ARGO
Code      : 0xACF5C039 (32 Bits)
Mesg Desc.: Model: 1 (WREM2), Sensor Temp: 28C
uint16_t rawData[183] = {6416, 3192,  436, 846,  404, 874,  404, 2148,  406, 2150,  406, 872,  430, 2122,  432, 846,  432, 2124,  432, 2124,  428, 850,  426, 2128,  402, 876,  428, 2124,  402, 2152,  408, 2150,  406, 2148,  430, 848,  400, 874,  432, 846,  434, 846,  426, 850,  432, 846,  428, 2126,  404, 2150,  404, 2152,  432, 846,  406, 872,  428, 2124,  432, 2124,  432, 2122,  406, 874,  404, 872,  428, 848,  402, 2152,  430, 2126,  430, 848,  404, 872,  404, 874,  404, 872,  430, 848,  430, 844,  406, 874,  402, 876,  400, 878,  430, 846,  404, 2150,  406, 2150,  428, 2126,  402, 874,  404, 874,  428, 850,  424, 852,  402, 876,  404, 870,  432, 846,  404, 874,  400, 2152,  430, 2126,  430, 2124,  404, 872,  428, 848,  428, 2126,  432, 846,  426, 2130,  402, 2152,  406, 872,  426, 850,  402, 2152,  430, 848,  402, 2152,  430, 846,  404, 874,  430, 844,  402, 878,  424, 850,  430, 848,  404, 874,  428, 2124,  428, 850,  428, 2126,  406, 870,  406, 2150,  402, 876,  428, 2124,  432, 848,  404, 872,  426, 2126,  430, 2124,  430, 2124,  406, 2150,  432};  // ARGO
uint8_t state[4] = {0xAC, 0xF5, 0xC0, 0x39};

# Target temp +1 (12)
Timestamp : 000053.669
Library   : v2.8.6

Protocol  : ARGO
Code      : 0xACF5003A (32 Bits)
Mesg Desc.: Model: 1 (WREM2), Sensor Temp: 4C
uint16_t rawData[183] = {6418, 3194,  404, 874,  404, 874,  404, 2148,  432, 2122,  434, 844,  428, 2126,  402, 876,  426, 2130,  404, 2152,  404, 872,  426, 2128,  404, 872,  406, 2148,  406, 2152,  428, 2128,  402, 2152,  402, 876,  404, 874,  404, 870,  404, 874,  404, 872,  400, 878,  398, 878,  426, 850,  400, 874,  406, 2150,  426, 852,  430, 2124,  430, 2124,  434, 2124,  404, 872,  404, 874,  406, 872,  402, 2152,  404, 2150,  432, 846,  430, 844,  432, 846,  430, 844,  432, 848,  402, 874,  430, 848,  404, 874,  402, 874,  400, 876,  426, 2128,  428, 2128,  406, 2150,  426, 852,  402, 874,  400, 878,  428, 850,  404, 874,  404, 874,  428, 848,  428, 848,  400, 2154,  404, 2150,  432, 2124,  406, 2150,  406, 872,  402, 2152,  406, 872,  400, 2154,  404, 2148,  402, 874,  428, 850,  428, 2126,  432, 846,  406, 2148,  404, 874,  404, 874,  400, 878,  400, 878,  400, 878,  426, 850,  426, 852,  404, 2150,  430, 848,  404, 2150,  404, 874,  402, 2152,  402, 2152,  404, 2152,  404, 874,  400, 2154,  404, 2148,  432, 2124,  406, 872,  404, 874,  404};  // ARGO
uint8_t state[4] = {0xAC, 0xF5, 0x00, 0x3A};

# Target temp +1 (13)
Timestamp : 000056.058
Library   : v2.8.6

Protocol  : ARGO
Code      : 0xACF5403A (32 Bits)
Mesg Desc.: Model: 1 (WREM2), Sensor Temp: 12C
[... 121 lines cut ...]
uint8_t state[4] = {0xAC, 0xF5, 0x80, 0x99};

# Cool mode with iFeel off
Timestamp : 001089.476
Library   : v2.8.6

Protocol  : ARGO
Code      : 0xACF5AA4B (32 Bits)
Mesg Desc.: Model: 1 (WREM2), Sensor Temp: 25C
uint16_t rawData[67] = {6420, 3190,  402, 876,  402, 876,  424, 2130,  400, 2154,  402, 876,  400, 2154,  404, 876,  402, 2152,  404, 2152,  404, 876,  400, 2154,  426, 852,  402, 2152,  400, 2156,  402, 2154,  404, 2152,  402, 876,  398, 2154,  404, 874,  402, 2154,  428, 848,  404, 2150,  402, 876,  428, 2126,  404, 2152,  400, 2156,  402, 878,  426, 2126,  428, 850,  398, 878,  400, 2154,  402, 876,  402};  // ARGO
uint8_t state[4] = {0xAC, 0xF5, 0xAA, 0x4B};
```

### What brand/model IR demodulator are you using?
VS1838b, however the captures are consistent. Also tried another no brand IR receiver module and same issue persists.

### Circuit diagram and hardware used (if applicable)
I tried this on 3 different builds, with 2 NodeMCUs and 1 pre-built (YTF) blaster with Tasmota on it.
The receiver hooks directly to D5, GND and 3V3.

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes

### Has this library/code previously worked as expected for you?
No, tried versions down to 2.8.0

mbronk, 2024-08-30:
###### Disclaimer: I do *not* own a `WREM2` device, so all the below is based on the captures in this thread and my past work on this protocol when adding a `WREM3` AC protocol support (which has similarities but slightly different).

> //quote from #1859
At a quick glance, there are 90bits in the raw message, and we are claiming there is only 32. I'm guessing the code that determines if it is a short "sensor" message is faulty. i.e. It incorrectly thinks it is a sensor message.

@crankyoldgit , @NiKiZe  - These captures do indeed look like a `90-bit` proto message, and they fail to get detected because of the logic expecting `96 bits`, and falling back to the short/iFeel message for **anything** shorter.

The original state definition seems to have these stray `6 bits` also defined as "nothingness":
https://github.com/crankyoldgit/IRremoteESP8266/blob/850a45f0dfda9d57001fbda6e3b11f62ff1550ff/src/ir_Argo.h#L67
... so I figured to try to just cut the message size `96->90` and it *seems to* have improved the situation. 
Take a look at #2136 for  _experimental_ ❗  changes that address it _(maybe 😁 )_.

----

@mateuszdrab - You may want to try to compile with the experimental changes from #2136 and see if it improves anything.
One thing I have noticed is the `iFeel` (room temperatures) decoded by the lib seem to be `+1°C` ***higher*** than what you report with the data. Could you double check (also in the iFeel reporting mode, when no button is pressed on the remote, but it periodically sends the room temp updates)?
Is the room temp reading you report coming from the remote itself, or a separate ambient sensor you're using?

mateuszdrab, 2024-08-30:
Damn, you're a star 😀

The result speak for themselves

```
Timestamp : 000023.491
Library   : v2.8.6

Protocol  : ARGO
Code      : 0xACF5809906E000FF1F000203 (90 Bits)
Mesg Desc.: Model: 1 (WREM2), Power: Off, Mode: 0 (Cool), Fan: 3 (Max), Temp: 10C, Sensor Temp: 24C, Max: Off, IFeel: Off, Night: Off
uint16_t rawData[183] = {6390, 3192,  430, 848,  424, 852,  432, 2124,  424, 2132,  426, 850,  422, 2134,  424, 852,  432, 2124,  424, 2132,  426, 852,  432, 2122,  424, 852,  432, 2124,  424, 2132,  426, 2130,  428, 2126,  432, 846,  426, 850,  422, 854,  430, 848,  426, 850,  432, 844,  428, 850,  424, 2130,  426, 2128,  430, 848,  426, 852,  432, 2122,  424, 2132,  426, 850,  422, 854,  430, 2126,  432, 844,  430, 2126,  432, 2124,  424, 852,  430, 848,  426, 850,  424, 854,  428, 848,  424, 852,  432, 844,  430, 848,  424, 852,  432, 846,  428, 2128,  430, 2124,  434, 2122,  424, 852,  432, 846,  428, 850,  424, 854,  430, 846,  426, 850,  422, 854,  430, 848,  426, 2130,  430, 2126,  430, 2124,  422, 2132,  426, 2130,  428, 2126,  430, 2124,  422, 2132,  426, 2128,  430, 2126,  432, 2124,  424, 2130,  426, 2128,  430, 848,  426, 852,  432, 844,  428, 848,  424, 852,  432, 846,  428, 850,  424, 854,  430, 846,  426, 850,  434, 844,  430, 848,  424, 2130,  428, 850,  424, 854,  430, 846,  426, 850,  422, 854,  430, 848,  426, 2128,  428, 2128,  430};  // ARGO
uint8_t state[12] = {0xAC, 0xF5, 0x80, 0x99, 0x06, 0xE0, 0x00, 0xFF, 0x1F, 0x00, 0x02, 0x03};

Timestamp : 000030.882
Library   : v2.8.6

Protocol  : ARGO
Code      : 0xACF5809906E000FF1F208203 (90 Bits)
Mesg Desc.: Model: 1 (WREM2), Power: On, Mode: 0 (Cool), Fan: 3 (Max), Temp: 10C, Sensor Temp: 24C, Max: Off, IFeel: Off, Night: Off
uint16_t rawData[183] = {6420, 3164,  426, 850,  422, 854,  462, 2092,  432, 2122,  424, 852,  432, 2124,  422, 854,  462, 2092,  434, 2122,  458, 820,  454, 2102,  424, 854,  430, 2126,  454, 2100,  458, 2096,  428, 2126,  432, 846,  460, 816,  424, 854,  464, 814,  426, 852,  432, 844,  430, 848,  490, 2066,  426, 2128,  452, 826,  458, 818,  454, 2100,  458, 2098,  462, 816,  422, 854,  452, 2102,  456, 822,  462, 2092,  454, 2100,  458, 820,  454, 824,  460, 816,  456, 820,  452, 824,  460, 818,  422, 854,  462, 814,  458, 818,  454, 822,  428, 2126,  454, 2102,  456, 2098,  458, 818,  432, 844,  462, 816,  458, 820,  454, 824,  460, 816,  458, 820,  452, 824,  426, 2128,  462, 2092,  454, 2102,  456, 2098,  458, 2096,  462, 2092,  454, 2102,  456, 2098,  426, 2128,  464, 2092,  456, 2100,  456, 2098,  460, 2094,  452, 824,  460, 818,  456, 820,  462, 814,  458, 818,  454, 824,  460, 816,  456, 820,  452, 2102,  456, 822,  462, 816,  458, 820,  454, 2102,  424, 854,  452, 824,  460, 818,  456, 822,  462, 814,  458, 2098,  460, 2094,  454, 2102,  424};  // ARGO
uint8_t state[12] = {0xAC, 0xF5, 0x80, 0x99, 0x06, 0xE0, 0x00, 0xFF, 0x1F, 0x20, 0x82, 0x03};

Timestamp : 000052.403
Library   : v2.8.6

Protocol  : ARGO
Code      : 0xACF5809906E000FF1FA08201 (90 Bits)
Mesg Desc.: Model: 1 (WREM2), Power: On, Mode: 0 (Cool), Fan: 3 (Max), Temp: 10C, Sensor Temp: 24C, Max: Off, IFeel: On, Night: Off
uint16_t rawData[183] = {6388, 3196,  428, 850,  424, 852,  430, 2124,  424, 2132,  426, 852,  432, 2124,  424, 854,  430, 2124,  422, 2132,  426, 852,  434, 2122,  424, 852,  432, 2124,  422, 2134,  424, 2130,  428, 2126,  432, 846,  428, 850,  422, 854,  430, 848,  426, 852,  432, 844,  428, 848,  424, 2130,  428, 2128,  430, 848,  424, 852,  432, 2124,  424, 2132,  426, 850,  422, 854,  430, 2126,  432, 846,  428, 2128,  430, 2124,  422, 854,  428, 848,  424, 852,  432, 846,  428, 850,  424, 854,  430, 846,  426, 850,  422, 854,  430, 848,  426, 2130,  428, 2128,  430, 2124,  422, 854,  484, 794,  424, 852,  432, 844,  428, 850,  424, 852,  430, 846,  426, 850,  422, 2132,  426, 2130,  428, 2126,  430, 2124,  422, 2132,  426, 2130,  428, 2126,  430, 2124,  434, 2122,  426, 2130,  428, 2126,  430, 2124,  434, 2122,  424, 852,  432, 844,  428, 850,  422, 854,  430, 846,  426, 852,  432, 844,  428, 848,  424, 2130,  428, 848,  424, 2132,  426, 852,  432, 2122,  424, 852,  432, 846,  428, 850,  424, 854,  430, 846,  426, 2128,  430, 2126,  432, 846,  428};  // ARGO
uint8_t state[12] = {0xAC, 0xF5, 0x80, 0x99, 0x06, 0xE0, 0x00, 0xFF, 0x1F, 0xA0, 0x82, 0x01};

Timestamp : 000283.833
Library   : v2.8.6

Protocol  : ARGO
Code      : 0xACF5AA4B (32 Bits)
Mesg Desc.: Model: 1 (WREM2), Sensor Temp: 25C
uint16_t rawData[67] = {6388, 3194,  430, 848,  426, 852,  432, 2122,  424, 2130,  428, 850,  424, 2132,  426, 850,  434, 2122,  424, 2130,  426, 850,  424, 2132,  426, 852,  422, 2132,  424, 2130,  428, 2128,  430, 2124,  422, 854,  430, 2126,  434, 844,  428, 2126,  432, 846,  426, 2128,  430, 846,  426, 2130,  428, 2126,  432, 2122,  424, 854,  432, 2124,  424, 854,  430, 846,  426, 2130,  428, 848,  424};  // ARGO
uint8_t state[4] = {0xAC, 0xF5, 0xAA, 0x4B};

```

Regarding iFeel, the temp reported was initially 20 and that is what matched the screen on the remote. 
Then, I was doing few more tests in another room where the temperature was different, perhaps I missed the current reading.
I've verified the readings now for iFeel and they all match what's on the screen of the remote.

I can compile Tasmota with that version of the lib and test a bit more if you like?

mateuszdrab, 2024-08-30:
Just built Tasmota and I can confirm it works fine there as well, codes are being decoded correctly and Home Assistant is aware of the updates from the remote (I'm using the [Tasmota-IRHVAC](https://github.com/hristo-atanasov/Tasmota-IRHVAC) component)

mbronk, 2024-08-31:
@mateuszdrab Glad it worked for you 🥳 , thank you for confirming!

1. If you have a spare few minutes, would you be able to capture a few more snapshots, changing other knobs and report if everything is still A-OK? 
    - **Not critical**, just thought it might be better to have comprehensive baseline coverage _(i.e. different modes, turbo, swing...)_ in case there would be future changes/fixes.
    - If you feel comfortable with that, I'd suggest committing them directly as test cases, somewhere here:
https://github.com/mbronk/IRremoteESP8266/blob/95e5bc445455696fd66d8f232219ae253450f2a2/test/ir_Argo_test.cpp#L1757-L1769
(note I've captured the last known-good samples from you as UT cases already).

2. While the original issue was for decode-side only, can you confirm if the IR code ***transmit*** works as well and the HVAC reacts as expected?

mbronk, 2024-08-31:
> Can someone please check if the sending of the 90 bits works as expected?

Note though that as far as I can tell, my changes do ***not*** affect the `send` path. I was mostly looking to confirm if I didn't accidentally miss anything, but AFAIK the internal **state** structure is still very much byte-aligned, so would send as full 12 bytes (=96 bit).

- It would take a bigger refactor (ex. pair the structures with a counter w/ their true `bitwise` sizes and a new flavor of https://github.com/crankyoldgit/IRremoteESP8266/blob/850a45f0dfda9d57001fbda6e3b11f62ff1550ff/src/IRsend.cpp#L409-L415) to send bit sequences having `len > 64 && len % 8 != 0`...
- IMHO it's a valid change to complement the decode, but *if*(❓)  sending longer codes works... I wonder if there's much value _(other than semantical correctness)_ in making such a change for symmetry (*if it ain't broke... 😉*)

mateuszdrab, 2024-08-31:
> @mateuszdrab Glad it worked for you 🥳 , thank you for confirming!
>
> 1. If you have a spare few minutes, would you be able to capture a few more snapshots, changing other knobs and report if [...]

Sure, I can capture some more commands from the remote and feed it back to you or put them in the E2E tests.

As we speak about more captures, here are a few more (fan speeds this time).

None of the options for `supported_fan_speeds` in Tasmota-IRHVAC match the middle fan speed. 

The IRrecvDumpV2 dump shows fan speed is correctly decoded as auto, min, med, max. 

```
Timestamp : 000011.337
Library   : v2.8.6

Protocol  : ARGO
Code      : 0xACF580F205E000BF00A06601 (90 Bits)
Mesg Desc.: Model: 1 (WREM2), Power: On, Mode: 0 (Cool), Fan: 2 (Med), Temp: 14C, Sensor Temp: 19C, Max: Off, IFeel: On, Night: Off
uint16_t rawData[183] = {6388, 3196,  432, 844,  452, 824,  428, 2126,  424, 2130,  452, 824,  462, 2092,  458, 820,  456, 2100,  462, 2092,  458, 818,  424, 2132,  462, 814,  460, 2094,  456, 2098,  464, 2090,  460, 2096,  454, 822,  430, 846,  462, 816,  426, 850,  458, 820,  456, 822,  432, 846,  428, 2126,  424, 852,  456, 2100,  462, 816,  426, 850,  458, 2096,  454, 2100,  428, 2126,  424, 2132,  464, 2092,  458, 818,  424, 2130,  430, 846,  460, 816,  426, 852,  424, 852,  434, 844,  454, 824,  462, 814,  426, 850,  458, 820,  456, 822,  454, 2102,  428, 2128,  456, 2100,  430, 848,  428, 850,  424, 852,  456, 822,  432, 846,  462, 814,  428, 848,  460, 818,  458, 2098,  464, 2092,  458, 2096,  454, 2100,  428, 2126,  424, 2132,  464, 814,  462, 2092,  424, 852,  456, 822,  454, 824,  430, 848,  428, 850,  424, 852,  456, 822,  454, 824,  430, 848,  460, 818,  424, 852,  422, 854,  432, 846,  430, 2126,  424, 852,  456, 2098,  464, 814,  428, 2126,  424, 2132,  430, 846,  428, 848,  426, 2130,  432, 2124,  428, 848,  426, 2130,  454, 822,  452};  // ARGO
uint8_t state[12] = {0xAC, 0xF5, 0x80, 0xF2, 0x05, 0xE0, 0x00, 0xBF, 0x00, 0xA0, 0x66, 0x01};

Timestamp : 000016.748
Library   : v2.8.6

Protocol  : ARGO
Code      : 0xACF580EA05E000BF00A04601 (90 Bits)
Mesg Desc.: Model: 1 (WREM2), Power: On, Mode: 0 (Cool), Fan: 1 (Min), Temp: 14C, Sensor Temp: 19C, Max: Off, IFeel: On, Night: Off
uint16_t rawData[183] = {6388, 3194,  422, 854,  432, 844,  430, 2124,  426, 2130,  454, 822,  430, 2124,  426, 852,  424, 2130,  432, 2124,  426, 850,  424, 2132,  430, 846,  430, 2126,  426, 2130,  432, 2124,  428, 2128,  456, 822,  432, 844,  462, 814,  428, 848,  426, 852,  424, 854,  432, 844,  430, 2126,  426, 852,  424, 2132,  430, 846,  428, 2128,  422, 854,  454, 2102,  426, 2128,  424, 2132,  430, 2126,  426, 852,  424, 2132,  430, 848,  462, 816,  426, 852,  424, 854,  432, 846,  430, 848,  428, 850,  426, 852,  422, 854,  432, 846,  430, 2124,  426, 2130,  436, 2118,  428, 850,  426, 852,  456, 822,  432, 846,  430, 848,  428, 850,  424, 852,  422, 854,  432, 2124,  426, 2128,  424, 2132,  430, 2124,  426, 2130,  432, 2124,  428, 850,  424, 2130,  432, 846,  428, 848,  426, 850,  424, 852,  424, 854,  432, 846,  430, 848,  428, 850,  426, 852,  434, 844,  432, 846,  430, 848,  428, 850,  458, 2098,  452, 824,  428, 2126,  424, 852,  424, 2132,  462, 2092,  426, 852,  424, 854,  432, 846,  430, 2126,  426, 852,  424, 2132,  430, 848,  426};  // ARGO
uint8_t state[12] = {0xAC, 0xF5, 0x80, 0xEA, 0x05, 0xE0, 0x00, 0xBF, 0x00, 0xA0, 0x46, 0x01};

Timestamp : 000020.589
Library   : v2.8.6

Protocol  : ARGO
Code      : 0xACF580E205E000BF00A02601 (90 Bits)
Mesg Desc.: Model: 1 (WREM2), Power: On, Mode: 0 (Cool), Fan: 0 (Auto), Temp: 14C, Sensor Temp: 19C, Max: Off, IFeel: On, Night: Off
uint16_t rawData[183] = {6388, 3194,  456, 820,  432, 846,  430, 2126,  426, 2130,  454, 824,  430, 2126,  424, 852,  422, 2132,  430, 2124,  426, 852,  422, 2132,  430, 848,  426, 2128,  456, 2098,  430, 2126,  426, 2130,  432, 846,  430, 846,  428, 848,  426, 852,  424, 854,  454, 822,  430, 846,  428, 2128,  424, 854,  432, 2122,  428, 850,  426, 852,  424, 854,  454, 2102,  428, 2126,  424, 2132,  462, 2092,  426, 850,  424, 2130,  464, 814,  428, 850,  424, 852,  424, 852,  432, 844,  430, 848,  428, 850,  426, 852,  456, 822,  432, 844,  430, 2124,  426, 2130,  432, 2122,  428, 850,  426, 852,  456, 820,  432, 844,  462, 814,  428, 850,  426, 852,  424, 854,  432, 2122,  428, 2126,  424, 2132,  430, 2124,  426, 2130,  454, 2100,  428, 850,  426, 2130,  454, 824,  430, 848,  428, 850,  426, 852,  424, 852,  432, 844,  430, 846,  428, 848,  426, 852,  424, 852,  432, 844,  464, 814,  428, 850,  426, 2130,  454, 822,  430, 2126,  426, 852,  424, 2132,  430, 2126,  424, 852,  424, 852,  454, 2100,  428, 850,  424, 852,  424, 2130,  462, 814,  428};  // ARGO
uint8_t state[12] = {0xAC, 0xF5, 0x80, 0xE2, 0x05, 0xE0, 0x00, 0xBF, 0x00, 0xA0, 0x26, 0x01};

Timestamp : 000023.186
Library   : v2.8.6

Protocol  : ARGO
Code      : 0xACF580FA05E000BF00A08601 (90 Bits)
Mesg Desc.: Model: 1 (WREM2), Power: On, Mode: 0 (Cool), Fan: 3 (Max), Temp: 14C, Sensor Temp: 19C, Max: Off, IFeel: On, Night: Off
uint16_t rawData[183] = {6388, 3194,  422, 854,  432, 846,  430, 2126,  424, 2130,  432, 846,  462, 2094,  424, 852,  434, 2122,  430, 2126,  424, 852,  434, 2120,  428, 848,  426, 2128,  434, 2122,  428, 2126,  424, 2130,  430, 846,  430, 848,  428, 850,  424, 852,  434, 844,  454, 824,  430, 848,  426, 2128,  434, 844,  454, 2102,  426, 850,  424, 2130,  430, 2124,  460, 2094,  434, 2120,  430, 2126,  424, 2130,  452, 824,  462, 2094,  424, [... cut ...]
Do you think it's a mapping issue or should `med-high` be supported in upstream code?

Another slight issue, sending `IRHVAC {"Vendor":"ARGO","Model":"WREM2","Command":"IFeel Report","SensorTemp":21}` from Tasmota turns off the unit. Sending as WREM3 works, but I am not sure if the unit acknowledges it since those are silent messages anyway. Considering my unit is temperamental even with its own remote, I can't 100% confirm anything but will know since without iFeel the compressor never runs continuously.

mateuszdrab, 2024-08-31:
Well, I added support for `med-high` in my local copy of Tasmota-IRHVAC and it works now... 

Can polish it up later and build on top of https://github.com/hristo-atanasov/Tasmota-IRHVAC/pull/154 to make it show as Low, Medium, High nicely in the UI with proper icons and translations.

mbronk, 2024-08-31:
> Another slight issue, sending IRHVAC {"Vendor":"ARGO","Model":"WREM2","Command":"IFeel Report","SensorTemp":21} from Tasmota turns off the unit.

Looking at the code... it's quite possible (and would likely be a separate issue if confirmed).
- See, if you're going through the generic **A/C** interface, you're likely hitting this code:
https://github.com/crankyoldgit/IRremoteESP8266/blob/850a45f0dfda9d57001fbda6e3b11f62ff1550ff/src/IRac.cpp#L3091-L3094
which does ***not*** understand the `"Command":"IFeel Report"` part and *probably* sends out a canned `control` command using all-defaults.
- I'm actually partially to blame for this one :grin: as when I was adding support for iFeel in WREM3, I was pretty adamanant on trying to _not_ affect any legacy use case, so kept WREM2 mostly as-is. Looks like it might benefit from a variant of sth like this though: 
https://github.com/crankyoldgit/IRremoteESP8266/blob/850a45f0dfda9d57001fbda6e3b11f62ff1550ff/src/IRac.cpp#L3067-L3071
- Before we go any further I suggest to peel the onion 🧅  first and confirm if the `sendSensorTemp()` even works for `ARGO/WREM2`. 
  - Since you use the llb through MQTT/Tasmota and the generic `IRac` interface, there are quite a few oher things that could have gone bad between the layers _(my notes above are NOT result of any code analysis, rather an echo from the past)_. It would be ideal to repro using raw `IRremoteESP8266` first...
  - For example, would you be able to run a low-level example code (ex. `TurnOnArgoAc.ino`), issuing only a single command `ac.sendSensorTemp(...)` and report if the unit seems to react to it?

----

Separately: Aside fan speed and iFeel temp,do other commands you send to the device seem to work?

mateuszdrab, 2024-08-31:
I think that's the case
I remember the other day I was scanning through the code and saw something that point me to think this is the path the code takes unless WREM3 is involved which is why I'm sending my temp reports as WREM3. Can't find that code now, I have terrible memory.

Whether they work, I don't know for sure but it looks like the unit is working despite removing the remote out of the room and feeding it temperature from another thermometer. 

P.S
There's a guy nearby on Facebook Market selling what seems to be Argo Utilisse 13 with a snapped liquid pie which is a WREM3 model that I'm thinking to pick up for cheap.
He says he lost the remote so this library might be a life saver but the unit also has WiFi I think. 
Which unit/model do you have?

mbronk, 2024-08-31:
> Whether they work, I don't know for sure but it looks like the unit is working despite removing the remote out of the room and feeding it a temperature from another thermometer.

WREM2 and WREM3 iFeel IR commands are quite different (differ even in length), so I doubt they'll *just work™️*, but perhaps the vendor did include some forward-compat in their control unit...
- BTW. my unit has a 'feature' that if it doesn't receive a iFeel report for a while it silently falls back to built-in sensor/disabling iFeel. I'd suggest testing w/ iFeel temp being WAY different from actual ambient temp and see if the unit reacts.

> Which unit/model do you have?

I have the Ulisse 13 *Wifi* model (note there are also flavors w/o wifi) - see at the top of #1912 for photos of the unit and the remote.
- It is basically all-controlled through wifi (I've written an [openHAB addon](https://next.openhab.org/addons/bindings/argoclima/) for it, I believe there's also a HA version), *except* iFeel which **has to** be sent via infrared (for which I use Tasmota-IR and this lib). Works quote stable, but the wifi part has [its quirks](https://next.openhab.org/addons/bindings/argoclima/#advanced-modes) to set up properly.
- As much as I have tested, `IRremoteESP8266` can do _everything_ the original remote can, so wifi-less operation is certainly possible.

mateuszdrab, 2024-08-31:
> WREM2 and WREM3 iFeel IR commands are quite different (differ even in length), so I doubt they'll *just work™️*, but perhaps the vendor did include some forward-compat in their control unit...
> - [...]

Yeah I think you're right. It's not working the same way as it was with the remote in the room.

Regarding fall back, I'm sure this is what happens, it does the same with iFeel off - runs compressor for a while and shuts off. That's probably because the temperature sensor is satisfied in the unit.

I'll compile the solution from above to send a fake iFeel message and see if it returns to operating as expected.

Unless I can recompile Tasmota with some quick fix/hardcode something to fix sending the WREM2 iFeel messages?

mbronk, 2024-08-31:
> Unless I can recompile Tasmota with some quick fix/hardcode something to fix sending the WREM2 iFeel messages?

Might be difficult. Tasmota uses [`IRac` interface](https://github.com/arendst/Tasmota/blob/d1bd31b34166a35a6ed5940ff1587a6005e02f7d/tasmota/tasmota_xdrv_driver/xdrv_05_irremote_full.ino#L528-L529) which you'd have to bypass/hard-wire completely to get to the code this way.
I'd still consider a one-off Arduino sketch the easiest. Your next best alternative would be just to hack/update the `IRac.cpp` itself to branch off `send.command`, probably sth similar to this:
```cpp
IRArgoAC ac(_pin, _inverted, _modulation);  //line 3091
if (send.command == stdAc::ac_command_t::kSensorTempReport) {
  ac.sendSensorTemp(static_cast<uint8_t>(roundf(sensorTempC)));
} else {
  argo(&ac, send.power, send.mode, degC, sensorTempC, send.fanspeed,
  send.swingv, send.iFeel, send.turbo, send.sleep);
}
```

Then again, even if it works, it probably deserves its own issue (enhancement request), as it is quite distinct from the decode errors you've been seeing initially.

mateuszdrab, 2024-08-31:
I did the below:

- added argoWrem2_iFeelReport
```
void IRac::argoWrem2_iFeelReport(IRArgoAC *ac, const float sensorTemp)
{
  ac->begin();
  ac->setMessageType(argoIrMessageType_t::IFEEL_TEMP_REPORT);
  ac->setSensorTemp(static_cast<uint8_t>(roundf(sensorTemp)));
  ac->send();
}
```

Added a switch case for this inAC under sendARGO
```#if SEND_ARGO
  case ARGO:
  {
    if (send.model == argo_ac_remote_model_t::SAC_WREM3)
    {
      IRArgoAC_WREM3 ac(_pin, _inverted, _modulation);
      switch (send.command)
      {
      case stdAc::ac_command_t::kSensorTempReport:
        argoWrem3_iFeelReport(&ac, sensorTempC);
        break;
      case stdAc::ac_command_t::kConfigCommand:
        /// @warning: this is ABUSING current **common** parameters:
        ///           @c clock and @c sleep as config key and value
        ///           Hence, value pre-validation is performed (safe-mode)
        ///           to avoid accidental device misconfiguration
        argoWrem3_ConfigSet(&ac, send.clock, send.sleep, true);
        break;
      case stdAc::ac_command_t::kTimerCommand:
        argoWrem3_SetTimer(&ac, send.power, send.clock, send.sleep);
        break;
      case stdAc::ac_command_t::kControlCommand:
      default:
        argoWrem3_ACCommand(&ac, send.power, send.mode, degC, sensorTempC,
                            send.fanspeed, send.swingv, send.iFeel, send.quiet, send.econo,
                            send.turbo, send.filter, send.light);
        break;
      }
      OUTPUT_DECODE_RESULTS_FOR_UT(ac);
    }
    else
    {
      IRArgoAC ac(_pin, _inverted, _modulation);
      switch (send.command)
      {
      case stdAc::ac_command_t::kSensorTempReport:
        argoWrem2_iFeelReport(&ac, sensorTempC);
        break;
      case stdAc::ac_command_t::kControlCommand:
      default:
        argo(&ac, send.power, send.mode, degC, sensorTempC, send.fanspeed,
             send.swingv, send.iFeel, send.turbo, send.sleep);
        OUTPUT_DECODE_RESULTS_FOR_UT(ac);
        break;
      }
      break;
    }
  }
```

Seems to work, different message is sent; however, it is not decoded by my other receiver. Looks a bit too long.

```
Timestamp : 000179.610
Library   : v2.8.6

Protocol  : UNKNOWN
Code      : 0xE8CF0C18 (33 Bits)
uint16_t rawData[65] = {6430, 3290,  446, 870,  460, 842,  452, 2158,  458, 2152,  450, 860,  448, 2160,  476, 832,  442, 2174,  452, 2154,  450, 858,  438, 2174,  450, 858,  448, 2162,  454, 2156,  460, 2150,  452, 2158,  468, 844,  452, 856,  450, 858,  460, 2158,  446, 864,  454, 852,  466, 850,  446, 2162,  452, 860,  502, 806,  456, 852,  444, 866,  462, 864,  444, 852,  444, 862,  444};  // UNKNOWN E8CF0C18

Timestamp : 000219.730
Library   : v2.8.6

Protocol  : ARGO
Code      : 0xACF59A3B (32 Bits)
Mesg Desc.: Model: 1 (WREM2), Sensor Temp: 23C
uint16_t rawData[67] = {6418, 3166,  426, 850,  456, 820,  432, 2122,  426, 2130,  452, 824,  460, 2096,  432, 846,  428, 2126,  434, 2122,  426, 850,  424, 2132,  462, 816,  426, 2130,  430, 2124,  456, 2098,  428, 2126,  454, 822,  452, 2102,  424, 854,  454, 2100,  458, 2098,  450, 826,  460, 818,  456, 2098,  428, 2128,  432, 2124,  458, 818,  422, 2132,  460, 2096,  432, 2124,  458, 820,  432, 844,  452};  // ARGO
uint8_t state[4] = {0xAC, 0xF5, 0x9A, 0x3B};
```

Will use the sketch now to see if it's the same.

mateuszdrab, 2024-09-03:
Quick update

I've ran the IRsend sketch with rawdata of the iFeel message for 24 hours and the AC has been running fine so I switched to Tasmota with my modified code to send the iFeel WREM2 reports and it's been working fine. 

I did make an adjustment to the transmitter circuit so perhaps that's why, I suspect the previous setup might not have been transmitting strong enough.

Either way, once the fix is merged I'll PR my modification to allow sending WREM2 iFeel updates.