crankyoldgit/IRremoteESP8266 #2155: IR Dump: Senville (Midea or Coolix) fan speed

krisrondeau, 2024-10-23:
When I use IRdump v3, I successfully dumped my IR code of my Senville HeatPump.
It identify as Coolix AC.

I get all my codes correctly except for Fan Speed.

This HeatPump have 6 fan speeds: auto, low, min, med, high, max.

When I use IRdump v3 it gives me 24 bits uint64 codes for all remote buttons. Strange thing is that it gives me the exact code for fan speed high and max!

what can I do to fix it?

crankyoldgit, 2024-10-23:
You can start by showing us the data from the dumps.

krisrondeau, 2024-10-24:
**Heat Mode, 25c, Fan at 80%:**
Timestamp : 017893.148
Library   : v2.8.6

Protocol  : COOLIX
Code      : 0xB23FCC (24 Bits)
Mesg Desc.: Power: On, Mode: 3 (Heat), Fan: 1 (Max), Temp: 25C, Zone Follow: Off, Sensor Temp: Off
uint16_t rawData[225] = {4400, 4382,  546, 1604,  546, 552,  522, 1604,  546, 1604,  546, 554,  520, 552,  522, 1602,  546, 552,  524, 554,  520, 1604,  546, 554,  520, 554,  520, 1604,  570, 1580,  546, 552,  542, 1584,  570, 528,  532, 544,  546, 1580,  570, 1580,  546, 1602,  570, 1580,  570, 1582,  572, 1576,  570, 1582,  572, 1578,  572, 526,  548, 526,  550, 526,  548, 524,  552, 524,  550, 522,  552, 1574,  576, 1574,  576, 522,  552, 522,  552, 1574,  576, 1574,  576, 522,  552, 524,  552, 522,  552, 524,  550, 1576,  572, 1578,  572, 524,  524, 550,  526, 1602,  546, 1604,  544, 5198,  4398, 4388,  540, 1610,  540, 554,  520, 1608,  540, 1610,  540, 554,  520, 556,  518, 1616,  536, 554,  518, 558,  514, 1614,  534, 558,  514, 562,  512, 1636,  486, 1662,  486, 590,  482, 1668,  482, 590,  486, 588,  484, 1666,  484, 1664,  486, 1664,  486, 1664,  486, 1664,  486, 1666,  484, 1664,  508, 1642,  486, 612,  484, 592,  482, 590,  486, 588,  484, 590,  484, 590,  484, 1642,  508, 1642,  508, 590,  486, 590,  484, 1644,  484, 1664,  486, 612,  460, 614,  462, 612,  462, 616,  458, 1690,  460, 1690,  458, 616,  458, 614,  460, 1690,  458, 1690,  460, 5282,  4290, 4522,  406, 1742,  406, 1744,  406, 696,  380, 1742,  406, 720,  356, 1770,  380, 720,  352, 1794,  354, 722,  352, 1798,  352, 878,  196};  // COOLIX B23FCC
uint64_t data = 0xB23FCC;

**Heat Mode, 25c, Fan at 80%:**
Timestamp : 017982.809
Library   : v2.8.6

Protocol  : BOSCH144
Code      : 0xB24D3FC0CC33B24D3FC0CC33D56400000039 (144 Bits)
Mesg Desc.: Power: On, Mode: 6 (Heat), Fan: 5 (High), Temp: 25C, Quiet: Off
uint16_t rawData[299] = {4350, 4434,  496, 1656,  494, 578,  494, 1658,  492, 1654,  496, 580,  494, 580,  494, 1654,  494, 580,  494, 582,  494, 1652,  498, 578,  494, 582,  492, 1656,  494, 1658,  490, 582,  492, 1656,  492, 582,  494, 582,  606, 1540,  496, 1654,  494, 1654,  494, 1654,  496, 1654,  494, 1654,  496, 1654,  492, 1656,  494, 582,  494, 578,  494, 580,  494, 580,  492, 582,  494, 578,  494, 1654,  494, 1656,  494, 580,  494, 580,  494, 1654,  496, 1658,  492, 580,  492, 582,  494, 580,  494, 580,  494, 1654,  496, 1654,  494, 582,  492, 580,  494, 1656,  494, 1654,  494, 5248,  4376, 4408,  518, 1632,  498, 578,  496, 1652,  496, 1654,  494, 580,  494, 582,  490, 1654,  472, 604,  472, 602,  472, 1676,  472, 602,  494, 578,  496, 1654,  496, 1654,  496, 578,  494, 1656,  494, 582,  494, 578,  494, 1656,  496, 1654,  494, 1656,  494, 1658,  492, 1656,  494, 1658,  492, 1656,  492, 1658,  494, 580,  494, 582,  494, 580,  492, 584,  490, 584,  490, 586,  488, 1662,  488, 1662,  488, 584,  490, 586,  490, 1660,  488, 1662,  486, 586,  488, 586,  488, 586,  490, 584,  490, 1662,  484, 1664,  486, 586,  490, 586,  486, 1664,  486, 1664,  486, 5260,  4340, 4448,  482, 1668,  482, 1668,  482, 594,  480, 1668,  480, 594,  482, 1666,  482, 592,  482, 1668,  480, 614,  460, 1670,  480, 1692,  458, 614,  460, 614,  458, 1690,  460, 614,  462, 614,  458, 616,  460, 614,  458, 614,  462, 614,  460, 614,  458, 616,  460, 616,  456, 616,  460, 614,  460, 616,  458, 616,  458, 616,  458, 614,  458, 618,  458, 616,  460, 616,  460, 616,  456, 616,  458, 616,  458, 616,  458, 616,  458, 616,  458, 616,  458, 616,  458, 614,  458, 618,  456, 1694,  458, 1692,  456, 1694,  456, 620,  454, 620,  456, 1694,  456};  // BOSCH144
uint8_t state[18] = {0xB2, 0x4D, 0x3F, 0xC0, 0xCC, 0x33, 0xB2, 0x4D, 0x3F, 0xC0, 0xCC, 0x33, 0xD5, 0x64, 0x00, 0x00, 0x00, 0x39};

crankyoldgit, 2024-10-24:
> I also remarked that uint16_t rawData is never the same! Is it normal?

Yes. See [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#user-content-Why_is_the_raw_data_for_a_button_or_AC_state_always_different_for_each_capture).

krisrondeau, 2024-10-24:
I don't know enough about IR codes to tell...

But strangely, I made some other tests and found that some times it identify as BOSCH144!

And that way I get all my 6 fan speeds!

Sometimes it's better to not asking questions! :-p

krisrondeau, 2024-10-24:
What I get from IRrecvDumpV3 is:

**Protocol  : BOSCH144
Code      : 0xB24DFF00BC43B24DFF00BC43D514000000E9 (144 Bits)
Mesg Desc.: Power: On, Mode: 6 (Heat), Fan: 1 (Low), Temp: 30C, Quiet: Off
uint16_t rawData[299] = {4402, 4384,  544, 1630,  520, 552,  520, 1632,  516, 1632,  520, 554,  518, 556,  518, 1630,  518, 556,  520, 554,  520, 1630,  520, 554,  520, 556,  520, 1630,  518, 1632,  518, 558,  516, 1632,  520, 1628,  520, 1628,  520, 1630,  520, 1630,  526, 1624,  520, 1630,  520, 1630,  520, 1628,  520, 554,  518, 556,  520, 556,  520, 554,  520, 556,  518, 556,  520, 554,  518, 558,  516, 1630,  520, 554,  518, 1632,  518, 1632,  516, 1630,  520, 1630,  522, 554,  518, 556,  520, 554,  518, 1630,  520, 554,  522, 554,  520, 554,  518, 558,  518, 1628,  520, 1628,  522, 5196,  4376, 4410,  518, 1630,  518, 580,  494, 1632,  518, 1632,  518, 580,  492, 582,  492, 1632,  518, 582,  494, 560,  516, 1630,  520, 556,  516, 556,  520, 1630,  520, 1630,  518, 556,  518, 1632,  518, 1632,  516, 1632,  518, 1632,  520, 1632,  516, 1632,  516, 1632,  518, 1634,  514, 1634,  518, 556,  516, 558,  516, 558,  516, 558,  518, 556,  516, 558,  516, 560,  514, 560,  514, 1636,  514, 562,  512, 1636,  514, 1640,  510, 1658,  492, 1656,  494, 582,  490, 584,  492, 584,  490, 1660,  490, 584,  490, 584,  490, 586,  488, 588,  484, 1662,  486, 1666,  484, 5258,  4338, 4446,  484, 1668,  482, 1664,  486, 590,  484, 1666,  484, 592,  484, 1664,  484, 592,  484, 1666,  482, 592,  484, 592,  482, 592,  484, 1666,  484, 590,  486, 1666,  482, 592,  482, 592,  484, 590,  482, 594,  482, 590,  484, 592,  482, 592,  482, 592,  484, 592,  480, 592,  484, 592,  484, 590,  482, 592,  482, 592,  482, 594,  482, 592,  482, 592,  484, 590,  484, 590,  484, 592,  482, 594,  480, 592,  482, 592,  484, 614,  460, 616,  458, 614,  462, 1690,  458, 1690,  460, 1690,  458, 616,  458, 1690,  460, 616,  458, 614,  462, 1690,  458};  // BOSCH144
uint8_t state[18] = {0xB2, 0x4D, 0xFF, 0x00, 0xBC, 0x43, 0xB2, 0x4D, 0xFF, 0x00, 0xBC, 0x43, 0xD5, 0x14, 0x00, 0x00, 0x00, 0xE9};**

How can I send this with IRsend??

I tried: irsend.sendBosch144 (0xB24DFF00BC43B24DFF00BC43D514000000E9);

it gives me this error: 
**invalid conversion from 'long long unsigned int' to 'const unsigned char*' [-fpermissive]**

I see this in IRsend.h:
  **void sendBosch144(const unsigned char data[],
                    const uint16_t nbytes = kBosch144StateLength,
                    const uint16_t repeat = kNoRepeat);**

So I assume I should send const unsigned char, but i don't find the way to do it!

Thanks again!

closed 2024-10-24 by krisrondeau