crankyoldgit/IRremoteESP8266 #2169: Cannot build with esp32 platform 3.0.0 or higher

Ks89, 2024-12-23:
### Version/revision of the library used
2.8.6

### Describe the bug
Cannot build my sketch with esp32 platform 3.0.0 or higher. With 2.0.17 it's ok.

#### To Reproduce
Install esp32 platform 3.1.0 and build the project for esp32s2 dev module

#### Example code used
My project https://github.com/home-anthill/devices/tree/develop/device-ac-lg

#### Expected behaviour
To build the project without errors.

#### Output of raw data from [IRrecvDumpV2.ino]
```
/Users/ks89/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp: In function 'void gpio_intr()':
/Users/ks89/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp:246:3: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
  246 |   timerAlarmEnable(timer);
      |   ^~~~~~~~~~~~~~~~
      |   timerAlarm
/Users/ks89/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp: In member function 'void IRrecv::enableIRIn(bool)':
/Users/ks89/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp:362:21: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
  362 |   timer = timerBegin(_timer_num, 80, true);
      |           ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/ks89/Library/Arduino15/packages/esp32/hardware/esp32/3.1.0/cores/esp32/esp32-hal.h:98,
                 from /Users/ks89/Library/Arduino15/packages/esp32/hardware/esp32/3.1.0/cores/esp32/Arduino.h:36,
                 from /Users/ks89/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.h:10,
                 from /Users/ks89/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp:6:
/Users/ks89/Library/Arduino15/packages/esp32/hardware/esp32/3.1.0/cores/esp32/esp32-hal-timer.h:35:13: note: declared here
   35 | hw_timer_t *timerBegin(uint32_t frequency);
      |             ^~~~~~~~~~
/Users/ks89/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp:371:3: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
  371 |   timerAlarmWrite(timer, MS_TO_USEC(params.timeout), ONCE);
      |   ^~~~~~~~~~~~~~~
      |   timerWrite
/Users/ks89/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp:375:23: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
  375 |   timerAttachInterrupt(timer, &read_timeout, false);
      |   ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/ks89/Library/Arduino15/packages/esp32/hardware/esp32/3.1.0/cores/esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
/Users/ks89/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp: In member function 'void IRrecv::disableIRIn()':
/Users/ks89/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp:401:3: error: 'timerAlarmDisable' was not declared in this scope
  401 |   timerAlarmDisable(timer);
      |   ^~~~~~~~~~~~~~~~~
/Users/ks89/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp: In member function 'void IRrecv::resume()':
/Users/ks89/Documents/Arduino/libraries/IRremoteESP8266/src/IRrecv.cpp:429:3: error: 'timerAlarmDisable' was not declared in this scope
  429 |   timerAlarmDisable(timer);
      |   ^~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

Am I missing something?

NiKiZe, 2024-12-23:
Duplicate of #2039 #2137 and others

https://github.com/search?q=repo%3Acrankyoldgit%2FIRremoteESP8266+timerAlarmEnable%28timer%29&type=issues

But could you test if #2144 is a working for you?

closed 2024-12-23 by NiKiZe

Ks89, 2024-12-23:
@NiKiZe Yes with #2144  I can build the project