crankyoldgit/IRremoteESP8266 #2174: Support for Carrier 40MAHB

kmgwalker, 2025-01-10:
I'm writing to share a successful decoding of the Carrier 40MAHB protocol.  I have not yet massaged the code into a form suitable for incorporation into the IRremoteESP8266 library, and I don't anticipate having time to do so in the near future.  But the attached code excerpt successfully controls power, temperature, mode, fan and "swing" of the unit.

Notes:

* The unit uses a close cousin of the Bosch 144-bit protocol for temperature/fan/mode.
* My unit is set for Fahrenheit, not Celsius.  This entails an extra temperature bit in the protocol, corresponding roughly to 0.5 deg C.
* The "swing" settings (as well as power off) use a 96-bit Coolix-style format.
* The related Bosch format has a "quiet" bit, and I've retained this in my code, but I have not tested it.

I've attached the relevant parts of my working code.

[can't figure out how to attach, so I'm pasting it in]

`

/**** 
	relevant parts of working code (plus a few irrelevant parts)
*****/

#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Bosch.h>

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>

#include <map>
using namespace std;

String version_str = "0.5a";

// pin definitions
const int LED_PIN = 0;
const int IR_LED_PIN = 4;

// first semi-general version ///////////
  // Title of owner's manual: Carrier 40MAHB ductless unit split system (sizes 06 09 12 18 24 30 36)

void carrier_ir_send(uint8_t* buf, int len = 18);

// non-redundant bytes from 144-bit protocol
	// note: switching modes also changes fan (for modes auto and dry only?) and temp (just for fan mode?) bits; need to understand that better
union preCarrier144Protocol {
  uint8_t raw[6];
  struct {
    // section 1
    uint8_t unk1       :5;   // unknown
    uint8_t fan1       :3;   // fan speed bits part 1
    
    uint8_t unk2       :2;   // unknown
    uint8_t mode1      :2;   // operation mode bits part 1
    uint8_t temp1      :4;   // desired temperature part 1

    // section 3 (section 2 is a repeat of section 1)
    uint8_t mode2      :1;   // operation mode bits part 2
    uint8_t fan2       :6;   // fan speed bits part 2
    uint8_t unk3       :1;   // unknown (also part of fan speed bits?)
    
    uint8_t unk4       :5;   // unknown
    uint8_t temp2      :1;   // desired temperature part 2
    uint8_t unk5       :1;   // unknown
    uint8_t unk5q      :1;   // unknown (maybe silent mode)
    
    uint8_t unk6       :4;   // unknown
    uint8_t temp3      :1;   // desired temperature part 3
    uint8_t unk7       :3;   // unknown
    
    uint8_t unk8       :8;   // unknown
  };
};
[... 197 lines cut ...]
  if (!no_send) send_main();
  return 0;
}
int CMS::set_swing(uint8_t new_swing, uint8_t no_send) {
  if (new_swing > 6) return 1;    // ** fix: hard-coded
  swing = new_swing;
  if (!no_send) send_swing();
  return 0;
}

IRBosch144AC bosch_ac(IR_LED_PIN);  // used only for sending msgs/bytes, not constructing msgs
//IRsend irsend(IR_LED_PIN);    // try to use IRsend directly instead?

// send msg to Carrier minisplit/AC; use Bosch class (perhaps should use IRsend directly and remove Bosch stuff)
void carrier_ir_send(uint8_t* buf, int len) {
  bosch_ac.setRaw(buf, len);
  bosch_ac.send();
  Serial.println(cms.dump_state().c_str());
}

/*
	[...]
*/

`

kmgwalker, 2025-01-10:
Pasted code looks a little messed-up.  If someone can explain a better way to do this, I'll try again.

tonhuisman, 2025-01-11:
> Pasted code looks a little messed-up. If someone can explain a better way to do this, I'll try again.

You have used a single back-tick to start and end a code block, but for multiple-line code blocks, triple-back-ticks (first one optionally having a code-high-light name, like `c++`) must be used. You're close 😉 
You can use the Preview tab above the comment field to see the result.

kmgwalker, 2025-01-11:
Thank you, tonhuisman, for the helpful response.  I missed the Preview tab.  Here's the code again:

```

/**** 
	relevant parts of working code (plus a few irrelevant parts)
*****/

#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Bosch.h>

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>

#include <map>
using namespace std;

String version_str = "0.5a";

// pin definitions
const int LED_PIN = 0;
const int IR_LED_PIN = 4;

// first semi-general version ///////////
  // Title of owner's manual: Carrier 40MAHB ductless unit split system (sizes 06 09 12 18 24 30 36)

void carrier_ir_send(uint8_t* buf, int len = 18);

// non-redundant bytes from 144-bit protocol
	// note: switching modes also changes fan (for modes auto and dry only?) and temp (just for fan mode?) bits; need to understand that better
union preCarrier144Protocol {
  uint8_t raw[6];
  struct {
    // section 1
    uint8_t unk1       :5;   // unknown
    uint8_t fan1       :3;   // fan speed bits part 1
    
    uint8_t unk2       :2;   // unknown
    uint8_t mode1      :2;   // operation mode bits part 1
    uint8_t temp1      :4;   // desired temperature part 1

    // section 3 (section 2 is a repeat of section 1)
    uint8_t mode2      :1;   // operation mode bits part 2
    uint8_t fan2       :6;   // fan speed bits part 2
    uint8_t unk3       :1;   // unknown (also part of fan speed bits?)
    
    uint8_t unk4       :5;   // unknown
    uint8_t temp2      :1;   // desired temperature part 2
    uint8_t unk5       :1;   // unknown
    uint8_t unk5q      :1;   // unknown (maybe silent mode)
    
    uint8_t unk6       :4;   // unknown
    uint8_t temp3      :1;   // desired temperature part 3
    uint8_t unk7       :3;   // unknown
    
    uint8_t unk8       :8;   // unknown
  };
};

struct temp_bits {
  uint8_t temp1      :4;   // desired temperature part 1
  uint8_t temp2      :1;   // desired temperature part 2
  uint8_t temp3      :1;   // desired temperature part 3
};
struct fan_bits {
  uint8_t fan1       :3;   // fan speed bits part 1
  uint8_t fan2       :6;   // fan speed bits part 2
};
struct mode_bits {
[... 186 lines cut ...]
  if (!no_send) send_main();
  return 0;
}
int CMS::set_swing(uint8_t new_swing, uint8_t no_send) {
  if (new_swing > 6) return 1;    // ** fix: hard-coded
  swing = new_swing;
  if (!no_send) send_swing();
  return 0;
}

IRBosch144AC bosch_ac(IR_LED_PIN);  // used only for sending msgs/bytes, not constructing msgs
//IRsend irsend(IR_LED_PIN);    // try to use IRsend directly instead?

// send msg to Carrier minisplit/AC; use Bosch class (perhaps should use IRsend directly and remove Bosch stuff)
void carrier_ir_send(uint8_t* buf, int len) {
  bosch_ac.setRaw(buf, len);
  bosch_ac.send();
  Serial.println(cms.dump_state().c_str());
}

/*
	[...]
*/

```

tonhuisman, 2025-01-11:
> Here's the code again:

You could have edited the original comment, now you have 2x the same (rather long) code-fragment, 1 poorly readable, and 1 readable... 🤔