crankyoldgit/IRremoteESP8266 #2175: Daikin ARC484B53 Remote support

lyt-hipal, 2025-01-17:
Currentlt Daikin ARC484B53 Remote is not supported in library. I've check with master branch and latest version v2.8.6. Please let me know what informwation you'll need to add support for this remote.

NiKiZe, 2025-01-17:
You will need to fill in all the data that is requested by the issue template. What made you think it didn't apply in this case?

lyt-hipal, 2025-01-17:
### Version/revision of the library used
v2.8.6, or 'master' as at 13 January 2025.

### Describe the bug
Currently Daikin ARC484B53 Remote is not supported in library. 

#### To Reproduce
I used [IRrecvDumpV3](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRrecvDumpV3) example to decode signal from Daikin ARC484B53 remote. It indiacted an unknown protocol.

#### Example code used
[IRrecvDumpV3](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRrecvDumpV3)

#### Expected behaviour
It should receive a daikin protocol so thatv it can be used for sending IR signals using library.

#### Output of raw data from [IRrecvDumpV3.ino](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) or [V3](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV3/) (if applicable)

I (3760) IRrecvDumpV3: Timestamp : 000020.161

I (3761) IRrecvDumpV3: Library   : v2.8.6

I (3763) IRrecvDumpV3: Protocol  : UNKNOWN
Code      : 0x1B88B4E (130 Bits)

I (3764) IRrecvDumpV3: uint16_t rawData[259] = {3476, 1714,  448, 462,  426, 1294,  452, 458,  420, 462,  426, 456,  422, 1298,  446, 464,  424, 458,  420, 1300,  456, 454,  424, 1296,  448, 1300,  508, 374,  450, 1298,  448, 462,  426, 1294,  452, 1298,  448, 462,  426, 456,  422, 460,  428, 454,  424, 1296,  448, 460,  428, 454,  422, 460,  418, 464,  424, 458,  420, 462,  426, 456,  422, 458,  418, 464,  424, 456,  420, 462,  426, 1294,  452, 458,  420, 462,  426, 454,  422, 1326,  420, 462,  426, 456,  420, 460,  428, 1294,  452, 458,  420, 462,  426, 1294,  452, 1296,  448, 434,  454, 456,  422, 460,  428, 454,  424, 1296,  448, 462,  426, 456,  422, 460,  428, 454,  424, 458,  420, 462,  426, 456,  422, 460,  428, 454,  424, 458,  420, 462,  426, 456,  422, 460,  428, 454,  424, 458,  418, 462,  426, 456,  420, 460,  428, 454,  424, 458,  420, 462,  426, 456,  422, 460,  428, 454,  422, 458,  418, 464,  424, 458,  420, 462,  426, 456,  422, 460,  418, 464,  424, 458,  420, 434,  454, 456,  422, 460,  428, 454,  424, 458,  420, 462,  426, 456,  420, 460,  428, 454,  422, 460,  418, 462,  424, 458,  420, 460,  426, 456,  422, 458,  418, 464,  424, 456,  420, 462,  426, 1294,  452, 1296,  450, 1300,  456, 454,  424, 458,  420, 462,  426, 456,  422, 460,  428, 454,  424, 458,  420, 462,  426, 456,  422, 460,  428, 454,  424, 458,  420, 462,  426, 456,  422, 460,  428, 454,  424, 458,  418, 462,  426, 456,  422, 1298,  446, 462,  426, 1294,  450, 460,  418, 464,  424};  // UNKNOWN 1B88B4E

I (8725) IRrecvDumpV3: Timestamp : 000025.126

I (8727) IRrecvDumpV3: Library   : v2.8.6

I (8728) IRrecvDumpV3: Protocol  : UNKNOWN
Code      : 0xFA6B47AC (130 Bits)

I (8730) IRrecvDumpV3: uint16_t rawData[259] = {3474, 1718,  454, 428,  448, 1298,  448, 462,  426, 456,  422, 460,  428, 1292,  452, 430,  448, 462,  426, 1294,  450, 458,  418, 1302,  454, 1294,  450, 460,  418, 1302,  454, 456,  422, 1298,  448, 1302,  454, 454,  422, 460,  418, 464,  424, 458,  420, 1300,  456, 454,  424, 458,  418, 462,  426, 456,  420, 460,  428, 454,  422, 458,  420, 462,  424, 456,  422, 460,  428, 454,  424, 1296,  448, 462,  426, 456,  422, 460,  428, 1292,  452, 456,  420, 462,  426, 456,  422, 458,  418, 1302,  454, 456,  422, 1298,  446, 1302,  454, 456,  422, 460,  428, 454,  422, 458,  420, 1300,  454, 454,  424, 458,  418, 464,  424, 458,  420, 462,  426, 454,  422, 460,  418, 462,  424, 458,  420, 460,  428, 454,  424, 458,  418, 464,  424, 456,  420, 460,  428, 454,  424, 458,  420, 462,  426, 456,  420, 460,  428, 454,  424, 458,  420, 462,  426, 456,  422, 432,  456, 454,  424, 458,  420, 462,  426, 456,  420, 460,  428, 454,  422, 458,  420, 462,  424, 456,  422, 460,  428, 454,  424, 458,  420, 462,  426, 456,  422, 460,  418, 464,  424, 458,  420, 462,  426, 456,  422, 458,  418, 464,  424, 456,  420, 462,  426, 456,  422, 460,  418, 1302,  454, 1294,  450, 1298,  448, 462,  426, 456,  420, 462,  426, 454,  422, 460,  418, 462,  424, 458,  420, 462,  426, 454,  422, 460,  418, 464,  424, 458,  420, 462,  426, 456,  422, 458,  418, 464,  424, 456,  420, 1300,  456, 454,  424, 1296,  448, 462,  426, 1294,  452, 458,  418, 464,  426};  // UNKNOWN FA6B47AC

I (10193) IRrecvDumpV3: Timestamp : 000026.593

I (10194) IRrecvDumpV3: Library   : v2.8.6

I (10196) IRrecvDumpV3: Protocol  : UNKNOWN
Code      : 0xC0B22CFA (130 Bits)

I (10197) IRrecvDumpV3: uint16_t rawData[259] = {3474, 1716,  446, 436,  452, 1296,  450, 460,  428, 454,  424, 458,  420, 1300,  456, 426,  452, 458,  420, 1302,  454, 454,  422, 1298,  448, 1302,  454, 454,  422, 1298,  448, 462,  426, 1294,  450, 1298,  446, 462,  426, 456,  422, 460,  428, 454,  424, 1296,  448, 460,  428, 454,  422, 458,  418, 462,  426, 456,  420, 460,  428, 454,  424, 458,  420, 462,  426, 456,  422, 460,  428, 1292,  454, 456,  420, 460,  428, 454,  424, 1296,  450, 460,  426, 454,  424, 458,  418, 1300,  456, 1292,  452, 458,  420, 1300,  456, 1292,  452, 456,  420, 462,  426, 454,  424, 458,  418, 1302,  454, 454,  422, 458,  418, 464,  424, 456,  422, 460,  426, 454,  424, 458,  418, 462,  426, 456,  420, 462,  426, 454,  422, 460,  418, 462,  424, 456,  422, 460,  428, 454,  424, 458,  418, 462,  426, 456,  420, 460,  428, 454,  424, 458,  420, 462,  426, 456,  422, 460,  428, 454,  424, 458,  420, 462,  426, 456,  420, 460,  418, 464,  424, 458,  420, 462,  426, 454,  422, 460,  418, 462,  426, 456,  422, 460,  428, 454,  424, 4 [... cut ...]
### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes

### Has this library/code previously worked as expected for you?
_Yes. v2.8.6 is working with other AC remotes.

### Other useful information
Please find captured IR Code in [Spreadsheet](https://docs.google.com/spreadsheets/d/1wc2wrYs95U4f_pCh9OCgGWrGQ4bmzlR6w919qY3ifWk/edit?usp=sharing)

lyt-hipal, 2025-01-30:
@NiKiZe Can you help?

lyt-hipal, 2025-02-06:
@crankyoldgit can you help me with this?