crankyoldgit/IRremoteESP8266 #2176: TCL 112 Protocol Turbo Wrong Position on TAC-12/XA73

Geravandi, 2025-01-18:
The Turbo setting on my TCL TAC-12/XA73 Split Airco (don't know the model of the remote) is stored in a different bit.  
In the original protocol the Turbo mode is set in bit 6 of Byte 6. While on my remote Turbo is stored in bit 7 of Byte 6.

Tested it by editing my local copy of "ir_Tcl.h" and it seems to work fine. 

**Actually, most (but not all) features of the library work on the TCL TAC-12/XA73 (On/Off, Temp, Mode, Fan (including Quite and (now) Turbo), Light, Health, SwingV (but only Auto and Off), On of the things that are not working/missing are the Mid-Low and Mid-High Fan speeds. But these seems to be send via the kTcl112AcSpecial message type like the Quiet setting.

**Modified code:** 
// Byte 6
uint8_t Mode           :4;
uint8_t Health          :1;
uint8_t                      :1;
uint8_t Turbo           :1;
uint8_t                      :1;

**Original Code:**
// Byte 6
uint8_t Mode            :4;
uint8_t Health           :1;
uint8_t Turbo            :1;
uint8_t                       :2;

**IR Dump:**

Summery:
On..: 0x23, 0xCB, 0x26, 0x01, 0x00, 0x24, **0x41**, 0x06, 0x05, 0x00, 0x00, 0x00, 0x80, 0x05
Off..: 0x23, 0xCB, 0x26, 0x01, 0x00, 0x24, **0x01**, 0x06, 0x00, 0x00, 0x00, 0x00, 0x80, 0xC0

**Turbo On:**

Timestamp : 004934.289
Library   : v2.8.6

Protocol  : TCL112AC
Code      : 0x23CB26020040C0018300000000A9 (112 Bits)
Mesg Desc.: Model: 1 (TAC09CHSD), Type: 2, Quiet: Off
uint16_t rawData[227] = {3136, 1556,  528, 1054,  528, 1052,  530, 264,  552, 264,  552, 264,  552, 1054,  530, 264,  552, 264,  554, 994,  588, 972,  610, 264,  552, 1054,  530, 264,  552, 266,  552, 1054,  528, 1054,  530, 264,  552, 1054,  528, 1054,  528, 266,  552, 264,  552, 1034,  548, 264,  552, 264,  552, 264,  552, 1054,  528, 264,  552, 264,  552, 264,  552, 264,  554, 264,  552, 264,  552, 266,  552, 264,  554, 264,  554, 264,  552, 264,  554, 264,  552, 264,  552, 264,  552, 264,  552, 264,  554, 264,  552, 266,  552, 266,  552, 264,  552, 1032,  550, 264,  552, 264,  552, 266,  552, 264,  552, 264,  552, 264,  552, 264,  552, 1032,  550, 1032,  550, 1032,  550, 266,  552, 264,  554, 264,  554, 264,  554, 264,  578, 228,  586, 232,  586, 1006,  550, 1032,  550, 266,  550, 266,  550, 266,  550, 266,  552, 266,  550, 1034,  548, 268,  548, 268,  548, 268,  548, 268,  548, 268,  548, 268,  548, 270,  548, 268,  548, 270,  546, 270,  546, 272,  546, 270,  546, 272,  546, 272,  544, 274,  542, 276,  542, 296,  520, 298,  518, 300,  516, 300,  516, 302,  516, 302,  516, 302,  516, 300,  516, 300,  516, 302,  516, 300,  516, 302,  516, 300,  516, 302,  516, 302,  516, 302,  516, 1066,  514, 302,  516, 302,  516, 1068,  516, 300,  516, 1068,  514, 302,  516, 1068,  514};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x02, 0x00, 0x40, 0xC0, 0x01, 0x83, 0x00, 0x00, 0x00, 0x00, 0xA9};

Timestamp : 004934.482
Library   : v2.8.6

Protocol  : TCL112AC
Code      : 0x23CB260100244106050000008005 (112 Bits)
Mesg Desc.: Model: 1 (TAC09CHSD), Type: 1, Power: On, Mode: 1 (Heat), Temp: 25C, Fan: 5 (High), Swing(V): 0 (Auto), Swing(H): Off, Econo: Off, Health: Off, Turbo: Off, Light: On, On Timer: Off, Off Timer: Off
uint16_t rawData[227] = {3110, 1560,  524, 1058,  524, 1060,  522, 272,  544, 272,  544, 272,  544, 1060,  522, 274,  544, 274,  544, 1060,  522, 1060,  522, 274,  542, 1062,  520, 274,  542, 274,  542, 1062,  520, 1062,  522, 296,  520, 1062,  520, 1062,  520, 296,  520, 296,  520, 1062,  520, 296,  520, 298,  520, 1064,  518, 298,  518, 300,  518, 298,  518, 298,  518, 300,  518, 300,  518, 300,  518, 300,  518, 300,  516, 302,  516, 300,  518, 300,  516, 300,  516, 302,  518, 298,  516, 300,  516, 300,  516, 1066,  516, 300,  516, 300,  516, 1066,  516, 300,  518, 300,  516, 1066,  516, 300,  516, 300,  516, 300,  516, 300,  516, 300,  518, 1066,  516, 302,  516, 300,  518, 1066,  516, 1066,  516, 300,  516, 300,  516, 300,  516, 300,  516, 300,  516, 1066,  516, 300,  518, 1066,  516, 300,  516, 300,  516, 302,  516, 300,  516, 300,  516, 302,  516, 300,  516, 302,  516, 300,  518, 300,  516, 300,  516, 300,  518, 300,  518, 300,  516, 300,  516, 300,  516, 302,  516, 300,  516, 300,  516, 300,  516, 300,  516, 300,  516, 300,  516, 300,  516, 300,  516, 300,  516, 300,  516, 302,  516, 300,  516, 302,  516, 300,  516, 302,  516, 300,  516, 300,  516, 300,  516, 300,  516, 1066,  516, 1066,  516, 300,  518, 1066,  516, 300,  516, 300,  516, 302,  516, 300,  516, 302,  516};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x24, 0x41, 0x06, 0x05, 0x00, 0x00, 0x00, 0x80, 0x05};

**Turbo Off:**
Timestamp : 004982.004
Library   : v2.8.6

Protocol  : TCL112AC
Code      : 0x23CB260200402001830000000009 (112 Bits)
Mesg Desc.: Model: 1 (TAC09CHSD), Type: 2, Quiet: Off
uint16_t rawData[227] = {3134, 1556,  528, 934,  648, 1052,  530, 264,  554, 264,  552, 264,  552, 1054,  528, 264,  552, 264,  552, 1042,  542, 1054,  528, 264,  552, 1052,  530, 264,  552, 264,  552, 1048,  536, 1054,  528, 264,  554, 1054,  528, 1054,  528, 264,  552, 264,  552, 1054,  528, 266,  552, 264,  554, 264,  552, 1052,  530, 264,  554, 264,  552, 264,  552, 264,  552, 264,  552, 266,  552, 266,  552, 264,  554, 264,  552, 264,  552, 266,  552, 264,  552, 264,  552, 264,  552, 264,  552, 264,  552, 264,  552, 264,  552, 264,  552, 264,  554, 1052,  530, 264,  552, 264,  554, 264,  552, 264,  552, 264,  552, 264,  552, 1052,  530, 264,  552, 264,  554, 1032,  550, 264,  552, 266,  552, 266,  552, 264,  552, 264,  554, 264,  552, 264,  552, 1032,  550, 1032,  550, 266,  552, 264,  552, 264,  552, 264,  554, 264,  552, 1032,  550, 264,  552, 264,  552, 266,  552, 266,  552, 264,  554, 264,  552, 264,  552, 264,  552, 264,  552, 264,  552, 264,  552, 264,  552, 264,  552, 264,  552, 264,  552, 264,  552, 264,  552, 266,  552, 266,  552, 264,  552, 264,  552, 264,  552, 264,  554, 264,  552, 264,  554, 264,  552, 264,  552, 264,  554, 262,  554, 262,  556, 262,  554, 264,  554, 1030,  576, 230,  588, 230,  588, 1006,  576, 230,  588, 230,  562, 264,  554, 262,  552};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x02, 0x00, 0x40, 0x20, 0x01, 0x83, 0x00, 0x00, 0x00, 0x00, 0x09};

Timestamp : 004982.188
Library   : v2.8.6

Protocol  : TCL112AC
Code      : 0x23CB2601002401060000000080C0 (112 Bits)
Mesg Desc.: Model: 1 (TAC09CHSD), Type: 1, Power: On, Mode: 1 (Heat), Temp: 25C, Fan: 0 (Auto), Swing(V): 0 (Auto), Swing(H): Off, Econo: Off, Health: Off, Turbo: Off, Light: On, On Timer: Off, Off Timer: Off
uint16_t rawData[227] = {3138, 1532,  552, 1030,  554, 1030,  576, 230,  588, 230,  586, 230,  586, 1006,  578, 230,  588, 228,  588, 1006,  576, 1006,  576, 230,  586, 1008,  550, 264,  552, 264,  552, 1032,  550, 1032,  550, 266,  552, 1032,  550, 1032,  550, 266,  550, 266,  552, 1032,  550, 266,  550, 266,  550, 1032,  550, 266,  550, 266,  550, 266,  550, 266,  550, 268,  550, 268,  550, 268,  550, 268,  550, 268,  550, 268,  550, 268,  548, 268,  548, 268,  548, 268,  548, 268,  548, 268,  548, 268,  550, 1034,  546, 270,  548, 270,  548, 1034,  546, 270,  548, 268,  548, 1036,  546, 270,  548, 270,  548, 270,  548, 270,  548, 270,  546, 270,  548, 268,  548, 270,  548, 1038,  544, 1038,  546, 270,  546, 270,  548, 270,  546, 270,  548, 270,  546, 270,  546, 270,  548, 270,  546, 272,  546, 272,  546, 272,  546, 272,  546, 272,  546, 272,  546, 272,  546, 272,  546, 272,  544, 272,  544, 272,  544, 272,  544, 272,  544, 272,  544, 274,  544, 272,  546, 272,  544, 274,  544, 274,  544, 274,  542, 274,  542, 276,  542, 274,  542, 274,  542, 298,  520, 296,  520, 296,  520, 296,  520, 298,  520, 298,  520, 298,  520, 298,  518, 298,  518, 298,  518, 298,  518, 298,  518, 1064,  518, 298,  518, 298,  518, 300,  518, 300,  518, 300,  516, 300,  518, 1066,  516, 1066,  516};  // TCL112AC
uint8_t state[14] = {0x23, 0xCB, 0x26, 0x01, 0x00, 0x24, 0x01, 0x06, 0x00, 0x00, 0x00, 0x00, 0x80, 0xC0};

crankyoldgit, 2025-01-18:
Probably easy enough to code up. But can you see if there is some ways to tell the models apart from the message code/array? If we can find that, we can have the library detect the difference.

Please let us know if you find something.

Geravandi, 2025-01-18:
> Probably easy enough to code up. But can you see if there is some ways to tell the models apart from the message code/array? If we can find that, we can have the library detect the difference.
> 
> Please let us know if you find something.

I have no idea how to do this or where to begin. I can send you some more dumps from my remote if that helps. Picture below of my remote. 

[image]

crankyoldgit, 2025-01-18:
Compare the hex string generated by the library and your remote. If there are other differences other then the but you're talking about, they may help

Geravandi, 2025-01-20:
It all looks the same, except that some functions are stored in different places (like Turbo) and for the (extra) functions that my remote has, like iFeel.