crankyoldgit/IRremoteESP8266 #2194: Receiving Delonghi N* devices

sb37, 2025-04-12:
Hoi,

I can send the commands for my Delonghi Pinguino N82, everything works fine.
As I am looking for addition for this protocol to your fine library I have made all changes that are described.

When I am sending a command with the remote, I get the following results from IRrevDumpV2:
Library   : v2.8.6

Protocol  : NEC_LIKE
Code      : 0x12188100 (32 Bits)
uint16_t rawData[67] = {9098, 4438,  582, 524,  556, 550,  556, 524,  582, 1624,  558, 524,  582, 548,  558, 1626,  560, 546,  556, 526,  580, 498,  580, 550,  556, 1630,  606, 1602,  558, 522,  582, 522,  580, 524,  556, 1654,  556, 524,  580, 524,  556, 548,  560, 522,  582, 522,  556, 548,  558, 1626,  582, 522,  556, 548,  558, 548,  558, 520,  582, 522,  556, 522,  582, 548,  556, 522,  556};  // NEC_LIKE 12188100
uint32_t address = 0x1848;
uint32_t command = 0x0;
uint64_t data = 0x12188100;

(This is Fan High, Cool mode, on at 16 degrees celsius)
I've used this data and much more to decode the various buttons and resend them. No problems. Working like a charm.
But the protocol bothers me. The library decodes it as NEC_LIKE and sets the decode_type like wise! All unit tests and thus decoding fail because it jums to the wrong decoder.

What can I do to have to library find the right decode type?

Groeten,
Frank.

crankyoldgit, 2025-04-13:
Try moving your decoding routine higher up in the `IRrecv::decode()` function.
e.g. Put it _before_ the NEC ones. That will preference it to decode it as your new protocol, instead of `NEC_LIKE`.
I.e.
https://github.com/crankyoldgit/IRremoteESP8266/blob/3390e72877ac15c74f602151de4a2fb9a154bdb0/src/IRrecv.cpp#L805-L815

sb37, 2025-04-13:
Worked like a charm! PR on its way.

closed 2025-04-13 by sb37