crankyoldgit/IRremoteESP8266 #2198: memory overflow in irrecv.decode

msilenius, 2025-04-26:
### Version/revision of the library used
2.8.6

### Describe the bug
When calling irrecv.decode(&results, NULL, 10, 0) then a variable that a has no relation with it is changed

#### To Reproduce

  Serial.printf("--> Debug ip: %d  %d  %d  %d  \n",  ServerIP[0],ServerIP[1],ServerIP[2],ServerIP[3]);
if (!irrecv.decode(&results, NULL, 10, 0)) return;
  Serial.printf("--> Debug ip: %d  %d  %d  %d  \n",  ServerIP[0],ServerIP[1],ServerIP[2],ServerIP[3]);

_What steps did you do, and what did or didn't actually happen? How can we reproduce the issue?_

global variables:

IPAddress NullIP;
byte ServerIP[] = {0, 0, 0, 0};
IPAddress BroadCastIP(255,255,255,255);
IPAddress LocalIp = NullIP;

byte MAC[6];
unsigned int localUdpPort = PORTNUMBER;
unsigned int ServerPort = PORTNUMBER;
bool ACKPending;

bool IsVest = false;

decode_results results;
IRrecv irrecv(PIN_IR_IN);
IRsend irsend(PIN_IR_OUT);

in setup
irrecv.enableIRIn();  // Start the ir receiver
irsend.begin();

in loop
ManageIRSensor();

then definition ofManageIRSensor();

//manage Ir reciever
void ManageIRSensor()
{
  //decode_results results;
  Serial.printf("--> Debug ip before: %d  %d  %d  %d  \n",  ServerIP[0],ServerIP[1],ServerIP[2],ServerIP[3]);
	if (!irrecv.decode(&results, NULL, 10, 0)) return;
  Serial.printf("--> Debug ip after: %d  %d  %d  %d  \n",  ServerIP[0],ServerIP[1],ServerIP[2],ServerIP[3]);

.......
}

result 
23:05:18.075 -> --> Debug ip before: 192  168  1  155  
23:05:18.075 -> --> Debug ip after: 24  0  1  155  

### Has this library/code previously worked as expected for you?
i didn't used this variable before so no idea.

msilenius, 2025-04-26:
more precision I also use this:

#define _IR_ENABLE_DEFAULT_ false
#define DECODE_MILESTAG2  true
#define DECODE_SONY  true

and I am using esp8266

msilenius, 2025-04-26:
meanwhile I moved decode_results results as a local variable. That fix it for now. but i guess it might overflow something else.

void ManageIRSensor()
{
  decode_results results;
  Serial.printf("--> Debug ip before: %d  %d  %d  %d  \n",  ServerIP[0],ServerIP[1],ServerIP[2],ServerIP[3]);
	if (!irrecv.decode(&results, NULL, 10, 0)) return;
  Serial.printf("--> Debug ip after: %d  %d  %d  %d  \n",  ServerIP[0],ServerIP[1],ServerIP[2],ServerIP[3]);

result 
23:14:24.125 -> --> Debug ip before: 192  168  1  155  
23:14:24.125 -> --> Debug ip after: 192  168  1  155

crankyoldgit, 2025-04-26:
Um, wow. That's bad. It shouldn't do that at all. 
Reviewing the code (on my phone) I can't see any mistakes. 
Are you able to radically simplify your code for the purposes minimal code to reproduce the issue and make it available?

This is puzzling.

msilenius, 2025-04-27:
i disabled everything in my loop except the 

ManageIRSensor() and the yield. 

same problem.

i don't really want to change the rest of the program, because with memory overflow, changing anything in variable creation order could move the bug or hide it.

I am using esp 8266 board 2.7.4

the bug doesn't seems to happen in IrRemote8266 2.5.0, 2.5.4, 2.5.6
it does happen in 2.6.0 (I disabled DECODE_MILESTAG2 as it doesn t exist before 2.6.0)

of course with memory overflow quited hard to tell if it's just luck.

msilenius, 2025-04-27:
that took a while but:

2.8.6 bug
2.7.20 bug
2.7.9 bug
2.7.5 bug
2.7.4 bug
2.7.3 bug
2.7.2 seems working
2.7.1 seems working
2.7.0 seems working

i had to to use this for the older version: if (!irrecv.decode(&results, NULL/*, 10, 0*/)) return;

crankyoldgit, 2025-04-27:
Okay. so looking at the changes from 2.7.2 to 2.7.3. i.e. https://github.com/crankyoldgit/IRremoteESP8266/compare/v2.7.2...v2.7.3
It seems 2.7.3 we added the ability to disable decoding of individual protocols. e.g.  `DECODE_SONY`

So, to debug it further, can you try setting `DECODE_SONY` to `false` and test that, and then ditto for MILESTAG etc.
That might help narrow it down to which protocol decoder is broken.

In the meantime, I'll look at the code for both to see if I can spot something.

crankyoldgit, 2025-04-27:
Looking over the code for both of those decoders, I can't see anything obvious. They only write to the _passed_ results structure/variable via a pointer (i.e. `&results`).
So, it has to be something in how it is dealing with the `results` structure/variable somewhere.

As I said before, it would be really helpful for you to remove everything you can from your code, to make the smallest amount of code that still triggers the issue. If/when I have that code, that may help me pinpoint exactly what is causing it.

crankyoldgit, 2025-04-27:
> if (!irrecv.decode(&results, NULL, 10, 0)) return;

Can you please humour me, and remove the `max_skip` parameter from this call?
e.g. Change it to just:
```c++
  if (!irrecv.decode(&results)) return;
  # OR
  if (!irrecv.decode(&results, NULL)) return;
```

msilenius, 2025-04-27:
2.7.3 with
#define _IR_ENABLE_DEFAULT_ false
#define DECODE_MILESTAG2 false
#define DECODE_SONY  false
still bugged 
(and decoding still happen with a results.decode_type = 4)

 if (!irrecv.decode(&results)) return;
still bugged 

I purged 98% of the code this is my remaining program

[image]

[image]

[image]

msilenius, 2025-04-27:
#define _IR_ENABLE_DEFAULT_ false
#define DECODE_MILESTAG2 true
#define DECODE_SONY false
bug
and still decode type 4

if i remove the parsebyte it doesn't bug

if i change the byte MAC[6]= {0,0,0,0,0,0}; to byte MAC[6]= {255,255,255,255,255,255};  it doesnt bug

if i start to change remove any other variable it doesn't bug (i guess memory overflow just happen somewhere else)

parse byte seems to work as expected.

crankyoldgit, 2025-04-27:
Looking at your code, I'm very concerned about your `parsebyte` procedure.
I think you're accidentally modifying the stack. That maybe overwriting parts of the variables that IRrecv is using causing it to misbehave.

I.e. you're modifying the cstr that is passed to it. The reason it is a `const` is because you're not supposed to modify it. The compiler is not expecting it to be modified because you told it you wouldn't. That's not great programming form as it can lead to unexpected results. 

Try copying the incoming C string to a local static array in that function. And only modify that string.

I'll continue to look at my code, to see if can spot anything. 
Can you supply the code on those images as text so I can use it to make some tests?

msilenius, 2025-04-28:
thanks for the tips,I changed the parsebyte  but still bugged:

to test  using a tsop 4856  I send a mile tag ir packet with an android phone which should be an arduino equivalent to 

void sendMiles2Test()
{
  byte IRPacket[] = { 0x83, 0x05, 0xE8 };
  sendMilesIR(IRPacket, 3);
}

// sending a miles tag 2 packet over IR
// Args:
//  data: a pointer a an aray of byte the send over ir.
void sendMilesIR(byte* data, byte size)
{
  if (SERIAL_ON) Serial.printf("IR sending: %d %d %d \n", data[0], data[1], data[2]);
  irsend.sendGeneric(2400, 600, 1200, 600, 600, 600, 0, 3200, data, size, 56000, true, 0U, 50U);
}

the full code on the receiver (this still bug )

#define SERIAL_ON 1

#include <ESP8266WiFi.h>

#if SERIAL_ON == 1 
  #define DEBUG_LAZERGUN(fmt, ...) Serial.printf_P( (PGM_P)PSTR(fmt), ## __VA_ARGS__ )
#else
  #define DEBUG_LAZERGUN(fmt, ...)
#endif

#define PIN_DFPLAYER D0
#define PIN_LED_RED D1
#define PIN_LED_GREEN D2
#define PIN_LED_BLUE D3
#define PIN_IR_IN SD3
#define PIN_IR_OUT D5
#define PIN_LASER D7
#define PIN_TRIGGER D6
#define PIN_RELOAD D4
#define PIN_VIBRATION D8 
#define PIN_SPOTLIGHT D8
#define SD3 10

#include <WiFiUdp.h>

#define _IR_ENABLE_DEFAULT_ false
#define DECODE_MILESTAG2 true
#define DECODE_SONY false

#include <IRremoteESP8266.h>
#include <IRrecv.h>
#include <IRutils.h>
#include <IRsend.h>

#include <SoftwareSerial.h>
#include <ESP8266httpUpdate.h>

#define CURRENT_VERSION "0.0.1.9"
#define CURRENT_VERSION_BYTE  19
#define FULL_NET_DEBUG 1

WiFiUDP Udp;
IPAddress NullIP;
byte ServerIP[] = {0, 0, 0, 0};
IPAddress BroadCastIP(255,255,255,255);
IPAddress LocalIp = NullIP;

byte MAC[6]= {0,0,0,0,0,0};

decode_results results;
IRrecv irrecv(PIN_IR_IN);
IRsend irsend(PIN_IR_OUT);

void setup()
{
	if (SERIAL_ON) Serial.begin(115200);
	if (SERIAL_ON) Serial.println("Go");
	
	irrecv.enableIRIn();  // Start the ir receiver
	irsend.begin();

	parseBytes(WiFi.macAddress().c_str(), ':', MAC, 6, 16);
	pinMode(PIN_IR_IN, INPUT);
}

void parseBytes(const char* str, char sep, byte* bytes, int maxBytes, int base) 
{
  char * toto = strdup(str);
    
  for (int i = 0; i < maxBytes; i++) 
  {
    bytes[i] = strtoul(toto, NULL, base);  // Convert byte
    toto = strchr(toto, sep);               // Find next separator
    if (toto == NULL || *toto == '\0') {
      break;                            // No more separators, exit
    }
    //DEBUG_LAZERGUN("%s \n", str);
    toto++;                                // Point to next character after separator
  }
}

void loop()
{
   byte before[]  = {0,0,0,0};
  before[0] = ServerIP[0];
  before[1] = ServerIP[1];
  before[2] = ServerIP[2];
  before[3] = ServerIP[3];
  
  //decode_results results;

  if (!irrecv.decode(&results/*, NULL, 10, 0*/)) return;
  Serial.printf("--> Debug ip before: %d  %d  %d  %d  \n",  before[0],before[1],before[2],before[3]);
  Serial.printf("--> Debug ip after: %d  %d  %d  %d  \n",  ServerIP[0],ServerIP[1],ServerIP[2],ServerIP[3]);

  DEBUG_LAZERGUN("IR recieved: %d ", results.decode_type);
  
  irrecv.resume();  // Receive the next value
  
	yield();
}

board setup 

[image]

result:

9:23:34.236 -> --> Debug ip before: 0  0  0  0  
09:23:34.236 -> --> Debug ip after: 24  0  0  0  
09:23:34.236 -> IR recieved: 4

crankyoldgit, 2025-04-28:
> char * toto = strdup(str);

FYI .. you need to `free()` this string before exiting the procedure, otherwise you will have a memory leak

crankyoldgit, 2025-04-28:
For the record, decode_type of 4 means it's detecting it as a SONY protocol message.

Can you please:
Use the IRrecvDumpV2 program (in the examples) and send me the text output of the IR message it captures, of the IR message that is triggering this. It will give me some data to simulate with.

Also, please turn on DEBUG, located here:
https://github.com/crankyoldgit/IRremoteESP8266/blob/3390e72877ac15c74f602151de4a2fb9a154bdb0/src/IRremoteESP8266.h#L1505-L1506
and recompile your code with the latest version of the library ... run the code so it produces the bug, and send me *ALL* the output please.

msilenius, 2025-04-28:
the irecvdump

11:19:13.216 -> IRrecvDump is now running and waiting for IR input on Pin 10
11:19:18.464 -> Attempting Aiwa RC T501 decode
11:19:18.464 -> Matching MARK 2580 vs 8960 + 50. Matching: 6757 <= 2580 <= 11263
11:19:18.464 -> Attempting Sanyo LC7461 decode
11:19:18.464 -> Matching MARK 2580 vs 8960 + 50. Matching: 6757 <= 2580 <= 11263
11:19:18.464 -> Attempting Carrier AC decode
11:19:18.464 -> Attempting Pioneer decode
11:19:18.464 -> Attempting Epson decode
11:19:18.464 -> Attempting NEC decode
11:19:18.464 -> Matching MARK 2580 vs 8960 + 50. Matching: 6757 <= 2580 <= 11263
11:19:18.464 -> Attempting MilesTag2 decode
11:19:18.464 -> Matching MARK 2580 vs 2400 + 50. Matching: 1837 <= 2580 <= 3063
11:19:18.511 -> Matching SPACE 584 vs 600 - 50. Matching: 412 <= 584 <= 688
11:19:18.511 -> Matching MARK 1294 vs 1200 + 50. Matching: 937 <= 1294 <= 1563
11:19:18.511 -> Matching SPACE 582 vs 600 - 50. Matching: 412 <= 582 <= 688
11:19:18.511 -> Matching MARK 652 vs 1200 + 50. Matching: 937 <= 652 <= 1563
11:19:18.511 -> Matching MARK 652 vs 600 + 50. Matching: 487 <= 652 <= 813
11:19:18.511 -> Matching SPACE 600 vs 600 - 50. Matching: 412 <= 600 <= 688
11:19:18.511 -> Matching MARK 650 vs 1200 + 50. Matching: 937 <= 650 <= 1563
11:19:18.511 -> Matching MARK 650 vs 600 + 50. Matching: 487 <= 650 <= 813
11:19:18.511 -> Matching SPACE 582 vs 600 - 50. Matching: 412 <= 582 <= 688
11:19:18.558 -> Matching MARK 668 vs 1200 + 50. Matching: 937 <= 668 <= 1563
11:19:18.558 -> Matching MARK 668 vs 600 + 50. Matching: 487 <= 668 <= 813
11:19:18.558 -> Matching SPACE 598 vs 600 - 50. Matching: 412 <= 598 <= 688
11:19:18.558 -> Matching MARK 648 vs 1200 + 50. Matching: 937 <= 648 <= 1563
11:19:18.558 -> Matching MARK 648 vs 600 + 50. Matching: 487 <= 648 <= 813
11:19:18.558 -> Matching SPACE 600 vs 600 - 50. Matching: 412 <= 600 <= 688
11:19:18.558 -> Matching MARK 652 vs 1200 + 50. Matching: 937 <= 652 <= 1563
11:19:18.558 -> Matching MARK 652 vs 600 + 50. Matching: 487 <= 652 <= 813
11:19:18.558 -> Matching SPACE 600 vs 600 - 50. Matching: 412 <= 600 <= 688
11:19:18.605 -> Matching MARK 1274 vs 1200 + 50. Matching: 937 <= 1274 <= 1563
11:19:18.605 -> Matching SPACE 602 vs 600 - 50. Matching: 412 <= 602 <= 688
11:19:18.605 -> Matching MARK 1276 vs 1200 + 50. Matching: 937 <= 1276 <= 1563
11:19:18.605 -> Matching SPACE 584 vs 600 - 50. Matching: 412 <= 584 <= 688
11:19:18.605 -> Matching MARK 666 vs 1200 + 50. Matching: 937 <= 666 <= 1563
11:19:18.605 -> Matching MARK 666 vs 600 + 50. Matching: 487 <= 666 <= 813
11:19:18.605 -> Matching SPACE 584 vs 600 - 50. Matching: 412 <= 584 <= 688
11:19:18.605 -> Matching MARK 666 vs 1200 + 50. Matching: 937 <= 666 <= 1563
11:19:18.650 -> Matching MARK 666 vs 600 + 50. Matching: 487 <= 666 <= 813
11:19:18.650 -> Matching SPACE 584 vs 600 - 50. Matching: 412 <= 584 <= 688
11:19:18.650 -> Matching MARK 666 vs 1200 + 50. Matching: 937 <= 666 <= 1563
11:19:18.650 -> Matching MARK 666 vs 600 + 50. Matching: 487 <= 666 <= 813
11:19:18.650 -> Matching SPACE 602 vs 600 - 50. Matching: 412 <= 602 <= 688
11:19:18.650 -> Matching MARK 700 vs 1200 + 50. Matching: 937 <= 700 <= 1563
11:19:18.650 -> Matching MARK 700 vs 600 + 50. Matching: 487 <= 700 <= 813
11:19:18.650 -> Matching SPACE 548 vs 600 - 50. Matching: 412 <= 548 <= 688
11:19:18.650 -> Matching MARK 654 vs 1200 + 50. Matching: 937 <= 654 <= 1563
11:19:18.697 -> Matching MARK 654 vs 600 + 50. Matching: 487 <= 654 <= 813
11:19:18.697 -> Matching SPACE 582 vs 600 - 50. Matching: 412 <= 582 <= 688
11:19:18.697 -> Matching MARK 1290 vs 1200 + 50. Matching: 937 <= 1290 <= 1563
11:19:18.697 -> Matching SPACE 584 vs 600 - 50. Matching: 412 <= 584 <= 688
11:19:18.697 -> Matching MARK 650 vs 1200 + 50. Matching: 937 <= 650 <= 1563
11:19:18.697 -> Matching MARK 650 vs 600 + 50. Matching: 487 <= 650 <= 813
11:19:18.697 -> Matching SPACE 616 vs 600 - 50. Matching: 412 <= 616 <= 688
11:19:18.697 -> Matching MARK 1276 vs 1200 + 50. Matching: 937 <= 1276 <= 1563
11:19:18.697 -> Matching SPACE 584 vs 600 - 50. Matching: 412 <= 584 <= 688
11:19:18.744 -> Matching MARK 1292 vs 1200 + 50. Matching: 937 <= 1292 <= 1563
11:19:18.744 -> Matching SPACE 582 vs 600 - 50. Matching: 412 <= 582 <= 688
11:19:18.744 -> Matching MARK 1360 vs 1200 + 50. Matching: 937 <= 1360 <= 1563
11:19:18.744 -> Matching SPACE 532 vs 600 - 50. Matching: 412 <= 532 <= 688
11:19:18.744 -> Matching MARK 1276 vs 1200 + 50. Matching: 937 <= 1276 <= 1563
11:19:18.744 -> Matching SPACE 584 vs 600 - 50. Matching: 412 <= 584 <= 688
11:19:18.744 -> Matching MARK 682 vs 1200 + 50. Matching: 937 <= 682 <= 1563
11:19:18.744 -> Matching MARK 682 vs 600 + 50. Matching: 487 <= 682 <= 813
11:19:18.791 -> Matching SPACE 566 vs 600 - 50. Matching: 412 <= 566 <= 688
11:19:18.791 -> Matching MARK 1278 vs 1200 + 50. Matching: 937 <= 1278 <= 1563
11:19:18.791 -> Matching SPACE 598 vs 600 - 50. Matching: 412 <= 598 <= 688
11:19:18.791 -> Matching MARK 668 vs 1200 + 50. Matching: 937 <= 668 <= 1563
11:19:18.791 -> Matching MARK 668 vs 600 + 50. Matching: 487 <= 668 <= 813
11:19:18.791 -> Matching SPACE 584 vs 600 - 50. Matching: 412 <= 584 <= 688
11:19:18.791 -> Matching MARK 752 vs 1200 + 50. Matching: 937 <= 752 <= 1563
11:19:18.791 -> Matching MARK 752 vs 600 + 50. Matching: 487 <= 752 <= 813
11:19:18.791 -> Matching SPACE 496 vs 600 - 50. Matching: 412 <= 496 <= 688
11:19:18.837 -> Matching MARK 700 vs 1200 + 50. Matching: 937 <= 700 <= 1563
[... cut ...]
11:19:18.837 -> Code      : 0x8305E8 (24 Bits)
11:19:18.837 -> uint16_t rawData[49] = {2580, 584,  1294, 582,  652, 600,  650, 582,  668, 598,  648, 600,  652, 600,  1274, 602,  1276, 584,  666, 584,  666, 584,  666, 602,  700, 548,  654, 582,  1290, 584,  650, 616,  1276, 584,  1292, 582,  1360, 532,  1276, 584,  682, 566,  1278, 598,  668, 584,  752, 496,  700};  // MILESTAG2 8305E8
11:19:18.884 -> uint32_t address = 0x3;
11:19:18.884 -> uint32_t command = 0x5;
11:19:18.884 -> uint64_t data = 0x8305E8;
11:19:18.884 -> 
11:19:18.884 ->

msilenius, 2025-04-28:
while i was at it I rewrite the parsebyte function to something more specific for my case and to reduce complexity and memory risk:

parseMac(WiFi.macAddress().c_str(), MAC);
  DEBUG_LAZERGUN("MAC: %02X:%02X:%02X:%02X:%02X:%02X\n ", MAC[0],MAC[1],MAC[2],MAC[3],MAC[4],MAC[5]);
}

void parseMac(const char* str, byte (&bytes)[6]) 
{
  for (int i = 0; i < 6; i++) 
  {
    bytes[i] = strtoul(&str[i*3], NULL, 16);
  }
}

still bug

then i removed the function all together:
void setup()
{
  pinMode(PIN_IR_IN, INPUT);
	if (SERIAL_ON) Serial.begin(115200);
	if (SERIAL_ON) Serial.println("\nGo");
	
	irrecv.enableIRIn();  // Start the ir receiver
	irsend.begin();

  MAC[0]=0x60;
  MAC[1]=0x01;
  MAC[2]=0x94;
  MAC[3]=0x51;
  MAC[4]=0xBC;
  MAC[5]=0xDA;
 
	//parseMac(WiFi.macAddress().c_str(), MAC);
  DEBUG_LAZERGUN("MAC: %02X:%02X:%02X:%02X:%02X:%02X\n ", MAC[0],MAC[1],MAC[2],MAC[3],MAC[4],MAC[5]);
}

still bugged

crankyoldgit, 2025-04-28:
Thanks for that. The `24` in your IP output looks like it's the number of bits in the protocol. So, that at least it gives me something to hunt for in my code.

msilenius, 2025-04-28:
irrecv.enableIRIn();  // Start the ir receiver
	irsend.begin();

        Serial.printf("MAC: %02X:%02X:%02X:%02X:%02X:%02X\n", MAC[0],MAC[1],MAC[2],MAC[3],MAC[4],MAC[5]);
}

simplified still bugged

crankyoldgit, 2025-04-28:
I've looked & looked at my code. I can't see the mistake, but it seems there is. I'll take another look tomorrow with fresher eyes

msilenius, 2025-04-28:
I have a few hour of train in front of me. I brought enough hardware to test. I gonna put a bunch of early return in your lib to see if I can find the line which Trigger the bug.

msilenius, 2025-04-28:
here are my test 

I changed the decode function to just result->bits=24; return true; 

still bugged

remove all the irsend reference 
still bug

I removed the enableirin(), the resume, the inside of whole irrecc::irrecc in the library. still bug

i can't find where you setup and start the interrupt to disable too.

msilenius, 2025-04-28:
i digged much deepper and i end up on this:

```
/// Results returned from the decoder
class decode_results {
 public:
  decode_type_t decode_type;  // NEC, SONY, RC5, UNKNOWN
  // value, address, & command are all mutually exclusive with state.
  // i.e. They MUST NOT be used at the same time as state, so we can use a union
  // structure to save us a handful of valuable bytes of memory.
  union {
    struct {
      uint64_t value;    // Decoded value
      uint32_t address;  // Decoded device address.
      uint32_t command;  // Decoded command.
    };
    uint8_t state[kStateSizeMax];  // Multi-byte results.

  };
  uint16_t bits;              // Number of bits in decoded value
  volatile uint16_t *rawbuf;  // Raw intervals in .5 us ticks
  uint16_t rawlen;            // Number of records in rawbuf.
  bool overflow;
  bool repeat;  // Is the result a repeat code?
};
```

kStateSizeMax = 8; 

 so changing. 

uint8_t state[kStateSizeMax];  // Multi-byte results.
by 
uint8_t state[8];  // Multi-byte results.

should change nothing

yet the bug disapear

also changing to 
uint8_t state[16];  // Multi-byte results. (size of the other union size)
uint8_t state[28];  // Multi-byte results.  (size of the kHitachiAc2StateLength)

doesn't generate bug.

so i used the bug to figure the  kStateSizeMax  at run time with your original code 
```

bool IRrecv::decode(decode_results *results, irparams_t *save,
                    uint8_t max_skip, uint16_t noise_floor) {

                      results->bits=kStateSizeMax;
    return true;
}
```

with 
```

void setup()
{  
  pinMode(PIN_IR_IN, INPUT);
	if (SERIAL_ON) Serial.begin(115200);
	if (SERIAL_ON) Serial.println("\nGo");
	
//	irrecv.enableIRIn();  // Start the ir receiver

  Serial.printf("size of results : %d \n", sizeof(results));
  Serial.printf("size of results.state: %d \n", sizeof(results.state));
 
  Serial.printf("MAC: %02X:%02X:%02X:%02X:%02X:%02X\n", MAC[0],MAC[1],MAC[2],MAC[3],MAC[4],MAC[5]);

   byte before[]  = {0,0,0,0};
  before[0] = ServerIP[0];
  before[1] = ServerIP[1];
  before[2] = ServerIP[2];
  before[3] = ServerIP[3];

  if (!irrecv.decode(&results, NULL, 10, 0)) return;
  Serial.printf("--> Debug ip before: %d  %d  %d  %d  \n",  before[0],before[1],before[2],before[3]);
  Serial.printf("--> Debug ip after: %d  %d  %d  %d  \n",  ServerIP[0],ServerIP[1],ServerIP[2],ServerIP[3]);
  //DEBUG_LAZERGUN("IR recieved: %d \n", results.decode_type);

  Serial.printf("--> value in results.bits: %d  \n",  results.bits);

  decode_results results1;
  Serial.printf("size of results.state: %d \n", sizeof(results1.state));
}

```
and this is my result

23:42:24.270 -> Go
23:42:24.270 -> size of results : 40 
23:42:24.270 -> size of results.state: 8 <-- size at creation
23:42:24.270 -> MAC: 00:00:00:00:00:00
23:42:24.270 -> --> Debug ip before: 0  0  0  0  
23:42:24.270 -> --> Debug ip after: 53  0  0  0   <- weird
23:42:24.270 -> --> value in results.bits: 0   <- weirder
23:42:24.270 -> size of results.state: 8 

seems at runtime kStateSizeMax became 53

then i tried the same with 

uint8_t state[8];  // Multi-byte results.

23:46:30.691 -> Go
23:46:30.691 -> size of results : 40 
23:46:30.691 -> size of results.state: 8  <-- size at creation
23:46:30.691 -> MAC: 00:00:00:00:00:00
23:46:30.691 -> --> Debug ip before: 0  0  0  0  
23:46:30.691 -> --> Debug ip after: 0  0  0  0  <-- but no bug
23:46:30.691 -> --> value in results.bits: 53  <-- weird 
23:46:30.691 -> size of results.state: 8

msilenius, 2025-04-28:
my opinion class + union +struct + dynamic array (not allowed in c++) = compiler lost

crankyoldgit, 2025-04-28:
Oh wait. State. That's a static sized array.

msilenius, 2025-04-29:
you are not going to like this:

```
bool IRrecv::decode(decode_results *results, irparams_t *save,
                    uint8_t max_skip, uint16_t noise_floor) {

                      Serial.printf("size  results.state of global ino once passed in IRrecv::decode: %d \n", sizeof(results->state));
                      Serial.printf("value of kStateSizeMax in IRrecv::decode: %d \n", kStateSizeMax);
                      decode_results test;               
                      Serial.printf("size of a decode_results.state created in IRrecv::decode: %d \n", sizeof(test.state));
     results->bits=kStateSizeMax;
    return true;
```
```

void setup()
{  
  pinMode(PIN_IR_IN, INPUT);
	if (SERIAL_ON) Serial.begin(115200);
	if (SERIAL_ON) Serial.println("\nGo");
	
//	irrecv.enableIRIn();  // Start the ir receiver

  Serial.printf("size of results : %d \n", sizeof(results));
  Serial.printf("size of results.state declare in global ino: %d \n", sizeof(results.state));
 
  Serial.printf("MAC: %02X:%02X:%02X:%02X:%02X:%02X\n", MAC[0],MAC[1],MAC[2],MAC[3],MAC[4],MAC[5]);

   byte before[]  = {0,0,0,0};
  before[0] = ServerIP[0];
  before[1] = ServerIP[1];
  before[2] = ServerIP[2];
  before[3] = ServerIP[3];

  if (!irrecv.decode(&results, NULL, 10, 0)) return;
  Serial.printf("--> Debug ip before: %d  %d  %d  %d  \n",  before[0],before[1],before[2],before[3]);
  Serial.printf("--> Debug ip after: %d  %d  %d  %d  \n",  ServerIP[0],ServerIP[1],ServerIP[2],ServerIP[3]);
  //DEBUG_LAZERGUN("IR recieved: %d \n", results.decode_type);

  Serial.printf("--> value in results.bits: %d  \n",  results.bits);

  decode_results results1;
  Serial.printf("size of results1.state declare in setup ino: %d \n", sizeof(results1.state));
}

```
07:06:15.640 -> Go
07:06:15.640 -> size of results : 40 
07:06:15.640 -> size of results.state declare in global ino: 8 
07:06:15.640 -> MAC: 00:00:00:00:00:00
07:06:15.640 -> size  results.state of global ino once passed in IRrecv::decode: 53 
07:06:15.640 -> value of kStateSizeMax in IRrecv::decode: 53 
07:06:15.640 -> size of a decode_results.state created in IRrecv::decode: 53 
07:06:15.687 -> --> Debug ip before: 0  0  0  0  
07:06:15.687 -> --> Debug ip after: 53  0  0  0  
07:06:15.687 -> --> value in results.bits: 0  
07:06:15.687 -> size of results1.state declare in setup ino: 8 

kStateSizeMax change value during the course of the program (which is not supposed to be as it is a const) then  when you write in nbits in irrecv it write it 53 - 8  = 45 Byte to far in memory creating the memory overflow. using a literal define  array uint8_t state[static value set with a #define]; would fix the symptoms. but it doesn't explain why  kStateSizeMax change value. 

btw i tried with 2 different esp board so it's not hardware related.

msilenius, 2025-04-29:
here is my last test program. it can reproduce the bug without even sending a ir packet

```
bool IRrecv::decode(decode_results *results, irparams_t *save,
                    uint8_t max_skip, uint16_t noise_floor) {
                      Serial.printf("size  results.state of global ino once passed in IRrecv::decode: %d \n", sizeof(results->state));
                      Serial.printf("value of kStateSizeMax in IRrecv::decode: %d \n", kStateSizeMax);
                      decode_results test;               
                      Serial.printf("size of a decode_results.state created in IRrecv::decode: %d \n", sizeof(test.state));
     results->bits=kStateSizeMax;
    return true;
}
```

```
#define SERIAL_ON 1

#include <ESP8266WiFi.h>

#define PIN_IR_IN SD3
#define SD3 10

#define _IR_ENABLE_DEFAULT_ false
#define DECODE_MILESTAG2 true
#define DECODE_SONY true

#include <IRremoteESP8266.h>
#include <IRrecv.h>
#include <IRutils.h>

IPAddress NullIP;
byte ServerIP[] = {0, 0, 0, 0};
IPAddress BroadCastIP(255,255,255,255);
IPAddress LocalIp ;//= NullIP;

byte MAC[6]= {0,0,0,0,0,0};

decode_results results;
IRrecv irrecv(PIN_IR_IN);

void setup()
{  
  pinMode(PIN_IR_IN, INPUT);
  delay(1000); //to wait for monitor to be ready after code upload 
	if (SERIAL_ON) Serial.begin(115200);
	if (SERIAL_ON) Serial.println("\nGo");
	
  Serial.printf("size of results : %d \n", sizeof(results));
  Serial.printf("size of results.state declare in global ino: %d \n", sizeof(results.state));

  Serial.printf("MAC: %02X:%02X:%02X:%02X:%02X:%02X\n", MAC[0],MAC[1],MAC[2],MAC[3],MAC[4],MAC[5]); //removing this line fix a part of the bug 

  byte before[]  = {0,0,0,0};
  before[0] = ServerIP[0];
  before[1] = ServerIP[1];
  before[2] = ServerIP[2];
  before[3] = ServerIP[3];

  if (!irrecv.decode(&results, NULL, 10, 0)) return;
  Serial.printf("--> Debug ip before: %d  %d  %d  %d  \n",  before[0],before[1],before[2],before[3]);
  Serial.printf("--> Debug ip after: %d  %d  %d  %d  \n",  ServerIP[0],ServerIP[1],ServerIP[2],ServerIP[3]);

  Serial.printf("--> value in results.bits: %d  \n",  results.bits);

  decode_results results1;
  Serial.printf("size of results1.state declare in setup ino: %d \n", sizeof(results1.state));
}

void loop()
{
	yield();
}
```
WITH Serial.printf("MAC: %02X:%02X:%02X:%02X:%02X:%02X\n", MAC[0],MAC[1],MAC[2],MAC[3],MAC[4],MAC[5]); 

07:42:53.587 -> size of results : 40 
07:42:53.587 -> size of results.state declare in global ino: 8 
07:42:53.587 -> MAC: 00:00:00:00:00:00
07:42:53.587 -> size  results.state of global ino once passed in IRrecv::decode: 53  <-- bad should be 8
07:42:53.635 -> value of kStateSizeMax in IRrecv::decode: 53  <-- bad should be 8
07:42:53.635 -> size of a decode_results.state created in IRrecv::decode: 53 <-- bad should be 8
07:42:53.635 -> --> Debug ip before: 0  0  0  0  
07:42:53.635 -> --> Debug ip after: 53  0  0  0  <-- bad should be 0
07:42:53.635 -> --> value in results.bits: 0  <-- bad should be 8
07:42:53.635 -> size of results1.state declare in setup ino: 8 

WITHOUT Serial.printf("MAC: %02X:%02X:%02X:%02X:%02X:%02X\n", MAC[0],MAC[1],MAC[2],MAC[3],MAC[4],MAC[5]); 

07:45:50.835 -> Go
07:45:50.835 -> size of results : 40 
07:45:50.835 -> size of results.state declare in global ino: 8 
07:45:50.835 -> size  results.state of global ino once passed in IRrecv::decode: 53    <-- bad should be 8
07:45:50.835 -> value of kStateSizeMax in IRrecv::decode: 53   <-- bad should be 8
07:45:50.835 -> size of a decode_results.state created in IRrecv::decode: 53   <-- bad should be 8
07:45:50.835 -> --> Debug ip before: 0  0  0  0  
07:45:50.835 -> --> Debug ip after: 0  0  0  0  <--- ok
07:45:50.835 -> --> value in results.bits: 0    <-- bad should be 8
07:45:50.835 -> size of results1.state declare in setup ino: 8 

that being said with memory overflow any small change in code has impact.

i don't think i can help you more than that. good luck

crankyoldgit, 2025-04-29:
The compiler should be throwing errors left, right, and centre if something tries to write to `kStateSizeMax`.
Yes, it's a `const`. It should NEVER change. The fact it is, is terrifying.
It's getting the value of 53 from `kHitachiAc2StateLength`.
I.e.
https://github.com/crankyoldgit/IRremoteESP8266/blob/3390e72877ac15c74f602151de4a2fb9a154bdb0/src/IRrecv.h#L65-L71

You've disabled 99% of the protocols, so no AC protocols should be being used/on, so `DECODE_AC` should be `false`.
I.e.
https://github.com/crankyoldgit/IRremoteESP8266/blob/3390e72877ac15c74f602151de4a2fb9a154bdb0/src/IRremoteESP8266.h#L962-L987

so, `kStateSizeMax` *should* be defined as `8`. That's the `sizeof(uint64_t)`

The fact your getting *BOTH* those answers tells me something is broken in the compiler or it's environment.

This is a long shot, but simple thing to test. Remove all \*.o and \*.a files. i.e. build the library from a super clean/new environment.
What tools are you using to build the code?
It should detect that something fundamental has changed (that is, a `#define` or a `const`) and rebuild what is needed.
My guess is there is a mix of object files (\*.o files) that have a different concept as to the structure of `decode_results` and because of that, it is mashing the code together during the linker phase, and thus parts of the code are writing to the wrong pieces of memory.

That's the only explanation I can think of why the same constant can be different in different places in the code.
That, or something is writing over where the constants/literals are stored in the code. That's even worse than, say, a buffer overrun, or bad pointer.

Btw .. top work on the debugging so far. Many Thanks.

msilenius, 2025-04-29:
I am using Arduino 1.8.13.
I gona try to clean everything and see if it fix it.
but in gonna have much less time to dig deeper from tomorow. 

it could be a linker problem ( multiple include without the #ifdefine set up directly to prevent double build?

msilenius, 2025-04-29:
it would be great you can try on your side to see if it reproduce on your side to see if it's environment or code problem. 

I screenshot my build parameter earlier with Arduino

crankyoldgit, 2025-04-29:
The Arduino IDE is a poor tool in this respect. Almost all build tools use file timestamps as a guide for if a file needs to be recompiled. If someone makes a change in the main header file that everything includes, then that is supposed trigger everything that depends on that file to be rebuilt. For whatever reason in your environment, the Arduino IDE failed to do that or got confused. This has happened before, not just with this library or project. 
That's why a lot of serious projects & developers use better tools, like `platformio` or `make` etc. which do a much better job. 

So, I'm guessing from your response that, in a clean build, things work correctly now. Please confirm. I sure hope so, because I still couldn't see anything wrong in the code. I'm not saying there isn't, but it was very odd something this bad had escaped detection for this long.

msilenius, 2025-04-29:
even with a clean build, the library will not work on Arduino ide because library are build independently of the main program (I tried  after deleting the Whole cache). 
the define in the  main programe aren't passed down.

msilenius, 2025-04-29:
what you say is correct. so If I #define decode true or false in my code it's not going to work. that's what All my bugs are about.

crankyoldgit, 2025-04-30:
Yep. Phew. So I'll consider this bug closed.
So, as you're using the Arduino IDE, you need to make those changes to the library in the file `IRremoteESP8266.h` directly.

Other IDEs (VisualCodeStudio, platformio, etc) allow you to set flags(defines) for the *entire* build, so that you don't need to modify the library files directly.
E.g. the `platformio.ini` files.

crankyoldgit, 2025-04-30:
E.g.
https://github.com/crankyoldgit/IRremoteESP8266/blob/3390e72877ac15c74f602151de4a2fb9a154bdb0/examples/SmartIRRepeater/platformio.ini#L38