crankyoldgit/IRremoteESP8266 #2199: 5 step cooling  is not implimented for samsung ac

AyuuSaxena, 2025-05-02:
5 Step Cooling is an energy-saving feature found in some Samsung air conditioners. It allows the compressor to operate at five different levels (100%, 80%, 60%, 40%, and 20%), adjusting power based on room conditions and cooling needs.

AyuuuSaxena, 2025-05-03:
> You will need to provide details on which changes is needed. At a minimum dumps from the original remote when this feature is used, as described by the issue template that was removed when creating this issue.

 decodeSamsungAC: offset=1, nbits=112, strict=1, rawlen=350, computed=234
14:37:48.759 -> , rawbuf=[1,410,8846,1613,4388,372,127,373,628,371,116,381,131,370,127,371,131,370,129,369,133,368,142,358,643,355,145,359,140,361,139,363,135,370,130,366,634,371,629,367,632,366,635,367,634,365,642,356,146,355,148,352,149,351,148,353,146,355,144,357,147,337,165,331,169,326,175,320,181,317,183,315,185,315,185,314,186,312,189,312,189,310,190,309,201,298,202,297,203,295,205,300,200,307,193,307,193,306,194,302,198,301,198,300,200,297,203,298,201,299,702,299,702,296,714,286,716,294,1448,1531,4485,278,722,286,214,280,220,276,226,273,230,265,230,273,231,269,232,268,232,263,733,275,232,262,239,261,739,262,240,258,740,262,741,267,730,271,730,272,730,268,732,268,234,263,239,260,240,259,241,260,241,258,246,254,253,244,256,244,255,248,253,247,249,260,243,262,238,260,239,264,237,261,238,261,238,263,236,263,237,263,233,266,237,262,239,260,241,261,240,261,240,260,247,253,244,254,249,256,238,265,236,270,231,270,227,269,235,268,233,265,232,269,234,265,1482,1505,4495,284,725,274,228,261,239,258,248,253,251,252,249,249,248,255,241,260,245,261,737,269,233,268,232,263,237,267,228,266,735,273,726,273,232,262,739,264,737,262,744,259,736,264,736,271,731,271,729,266,730,272,237,258,242,260,241,257,744,261,739,258,744,260,248,247,254,244,256,247,253,247,253,253,248,252,248,256,243,257,739,266,736,261,735,269,236,256,745,258,742,262,248,247,253,245,253,247,256,253,247,249,251,254,246,256,742,262,735,268,731,267,735,267]
14:37:48.888 -> Matching MARK 820 vs 586 + 50. Matching: 477 <= 820 <= 796
14:37:48.888 -> Header mark failed!

i am using decodeSamsungAC this function but it is failing in matchMark function.

rawBuf for 5 step cooling: [https://github.com/AyuuuSaxena/Ac/blob/main/samsungRawBuf5StepCooling.txt](url)

AyuuSaxena, 2025-05-03:
> You will need to provide details on which changes is needed. At a minimum dumps from the original remote when this feature is used, as described by the issue template that was removed when creating this issue.

decodeSamsungAC: offset=1, nbits=112, strict=1, rawlen=350, computed=234
14:37:48.759 -> , rawbuf=[1,410,8846,1613,4388,372,127,373,628,371,116,381,131,370,127,371,131,370,129,369,133,368,142,358,643,355,145,359,140,361,139,363,135,370,130,366,634,371,629,367,632,366,635,367,634,365,642,356,146,355,148,352,149,351,148,353,146,355,144,357,147,337,165,331,169,326,175,320,181,317,183,315,185,315,185,314,186,312,189,312,189,310,190,309,201,298,202,297,203,295,205,300,200,307,193,307,193,306,194,302,198,301,198,300,200,297,203,298,201,299,702,299,702,296,714,286,716,294,1448,1531,4485,278,722,286,214,280,220,276,226,273,230,265,230,273,231,269,232,268,232,263,733,275,232,262,239,261,739,262,240,258,740,262,741,267,730,271,730,272,730,268,732,268,234,263,239,260,240,259,241,260,241,258,246,254,253,244,256,244,255,248,253,247,249,260,243,262,238,260,239,264,237,261,238,261,238,263,236,263,237,263,233,266,237,262,239,260,241,261,240,261,240,260,247,253,244,254,249,256,238,265,236,270,231,270,227,269,235,268,233,265,232,269,234,265,1482,1505,4495,284,725,274,228,261,239,258,248,253,251,252,249,249,248,255,241,260,245,261,737,269,233,268,232,263,237,267,228,266,735,273,726,273,232,262,739,264,737,262,744,259,736,264,736,271,731,271,729,266,730,272,237,258,242,260,241,257,744,261,739,258,744,260,248,247,254,244,256,247,253,247,253,253,248,252,248,256,243,257,739,266,736,261,735,269,236,256,745,258,742,262,248,247,253,245,253,247,256,253,247,249,251,254,246,256,742,262,735,268,731,267,735,267]
14:37:48.888 -> Matching MARK 820 vs 586 + 50. Matching: 477 <= 820 <= 796
14:37:48.888 -> Header mark failed!

i am using decodeSamsungAC function but it is failing in matchMark function.

rawBuf for 5 step cooling: [https://github.com/AyuuuSaxena/Ac/blob/main/samsungRawBuf5StepCooling.txt](https://github.com/crankyoldgit/IRremoteESP8266/issues/url)

crankyoldgit, 2025-05-05:
That's because the first `MARK` is out of the tolerance range. It is expecting a `MARK` signal of about (586+50)useconds +/- 25%. i.e. Something between 477us & 796us. 820us is above & outside that range.

You can use the [`setTolerance()`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/classIRrecv.html#aa091c449db70c65fd0221669df7438ea) method to adjust that range. The default value is [`kTolerance`](https://crankyoldgit.github.io/IRremoteESP8266/doxygen/html/IRrecv_8h.html#a7884008b3a738dfc7bd8658655e10272) which is **25**%. Try setting it to 30% or 35% perhaps.

Looking at the data provided (which by the way is NOT the `rawData[]` we use, you are look at the values UNDERNETHER that. It's not the same) your subsequent `MARK` signals are around (2 * 372 = 744us) which is inside the normal/expected range (477us & 796us)

We like for people to use the IRrecvDumpV2/3 example code to report their finding as it gives us the data in a trusted/recognised format. (see https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino)

AyuuSaxena, 2025-05-05:
> That's because the first `MARK` is out of the tolerance range. It is expecting a `MARK` signal of about (586+50)useconds +/- 25%. i.e. Something between 477us & 796us. 820us is above & outside that [...]

hi crankyoldgit,
i have extracted rawBuf from dumpV2 file. you can see file that i provided above

crankyoldgit, 2025-05-06:
Have you adjusted the Tolerance value like I suggested?

sid8840, 2025-05-18:
`remote_transmitter:
    - pin: GPIO26
      carrier_duty_percent: 33%
      id: fan
    
button:
  - platform: template
    name: "AC WindFree 120"
    on_press:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38000Hz
          code: [616, -16984, 3018, -8990, 575, -422, 576, -1421, 576, -421, 577, -422, 528, -470, 530, -468, 531, -465, 533, -467, 600, -394, 605, -1394, 602, -393, 605, -394, 604, -1394, 603, -1397, 599, -1397, 600, -399, 599, -1397, 599, -1420, 503, -1493, 531, -1466, 601, -1394, 604, -393, 605, -394, 603, -393, 605, -392, 606, -1392, 604, -394, 604, -395, 603, -396, 602, -398, 600, -398, 600, -420, 578, -420, 528, -470, 530, -469, 598, -397, 603, -395, 604, -393, 605, -394, 604, -394, 604, -393, 605, -393, 605, -392, 606, -395, 603, -394, 604, -397, 601, -421, 577, -421, 577, -400, 598, -419, 579, -421, 528, -469, 531, -1466, 602, -1393, 605, -1391, 606, -1393, 603, -2902, 2993, -8989, 603, -1391, 606, -395, 603, -395, 603, -421, 577, -421, 577, -400, 598, -421, 577, -419, 530, -469, 599, -1396, 604, -394, 605, -395, 603, -1393, 603, -1390, 606, -394, 604, -1391, 605, -395, 603, -1420, 576, -1420, 576, -1418, 579, -419, 503, -1493, 602, -396, 603, -1394, 604, -1391, 605, -1393, 603, -1391, 605, -1391, 605, -1395, 601, -1420, 576, -1420, 576, -419, 579, -419, 504, -495, 531, -468, 599, -397, 603, -1394, 604, -394, 604, -392, 606, -1392, 604, -1392, 604, -393, 605, -1393, 603, -396, 602, -1397, 599, -421, 577, -421, 577, -420, 530, -468, 531, -466, 601, -396, 604, -394, 605, -1392, 605, -1392, 604, -1392, 604, -1392, 604]          
          repeat: 0
          transmitter_id: fan

  - platform: template
    name: "AC WindFree 100"
    on_press:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38000Hz
          code: [618, -17675, 3039, -8964, 532, -464, 534, -1464, 532, -465, 533, -465, 533, -466, 532, -466, 532, -470, 528, -470, 528, -493, 505, -1490, 481, -518, 529, -468, 531, -467, 532, -466, 532, -467, 532, -1465, 531, -1466, 530, -1465, 531, -1464, 532, -1491, 505, -1470, 527, -493, 505, -470, 528, -491, 507, -493, 528, -468, 532, -468, 531, -467, 532, -465, 533, -466, 532, -466, 532, -466, 532, -465, 533, -465, 533, -465, 533, -467, 531, -467, 531, -493, 505, -493, 505, -493, 505, -493, 505, -494, 504, -493, 528, -469, 531, -467, 532, -465, 533, -467, 532, -465, 533, -464, 534, -466, 532, -466, 532, -464, 534, -1463, 533, -1466, 530, -1469, 527, -1492, 530, -2948, 3018, -8990, 528, -1468, 531, -468, 531, -465, 534, -465, 533, -465, 533, -465, 533, -465, 533, -465, 533, -466, 532, -1465, 531, -465, 533, -492, 506, -469, 529, -1492, 504, -1491, 502, -1492, 532, -468, 531, -1465, 533, -1462, 535, -1464, 532, -466, 532, -1464, 532, -465, 533, -1468, 528, -1468, 528, -469, 529, -493, 505, -492, 506, -1491, 530, -1466, 531, -1467, 530, -465, 533, -466, 532, -467, 531, -466, 532, -466, 532, -1464, 532, -467, 531, -468, 530, -1491, 505, -1490, 506, -492, 506, -1490, 530, -468, 531, -1465, 532, -466, 532, -465, 533, -464, 534, -465, 533, -466, 532, -465, 533, -465, 533, -1465, 531, -1467, 529, -1491, 505, -1491, 532]
          repeat: 0
          transmitter_id: fan
          

  - platform: template
    name: "AC WindFree 80"
    on_press:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38000Hz
          code: [619, -17689, 2991, -9016, 505, -464, 534, -1463, 533, -465, 533, -463, 535, -466, 532, -467, 531, -465, 533, -470, 528, -493, 505, -1489, 508, -493, 601, -396, 532, -1465, 532, -1464, 532, -1463, 533, -464, 534, -1463, 533, -1463, 533, -1465, 531, -1491, 505, -1470, 526, -491, 507, -492, 600, -1393, 535, -464, 534, -464, 534, -464, 534, -465, 532, -465, 533, -465, 533, -463, 535, -464, 534, -463, 535, -464, 534, -466, 532, -467, 531, -467, 531, -492, 506, -491, 507, -493, 505, -491, 602, -397, 532, -465, 533, -465, 533, -465, 533, -466, 532, -465, 533, -465, 533, -464, 534, -465, 533, -464, 534, -464, 534, -1465, 531, -1470, 526, -1492, 504, -1488, 535, -2949, 3018, -8988, 533, -1463, 534, -465, 533, -464, 534, -464, 534, -464, 534, -464, 534, -464, 534, -464, 534, -465, 533, -1467, 529, -468, 530, -493, 505, -1489, 508, -1491, 601, -395, 604, -1392, 534, -464, 534, -1462, 534, -1463, 533, -1463, 533, -463, 535, -1464, 532, -468, 530, -1491, 505, -1465, 531, -1490, 531, -1466, 603, -1393, 534, -1462, 534, -1463, 534, -1463, 533, -463, 535, -464, 534, -465, 533, -466, 532, -466, 532, -1466, 530, -493, 505, -491, 507, -1490, 601, -1396, 532, -466, 532, -1463, 508, -489, 535, -1463, 533, -464, 534, -464, 534, -464, 534, -465, 533, -468, 530, -466, 532, -469, 529, -1469, 527, -1490, 531, -1464, 605, -1392, 534]
          repeat: 0
          transmitter_id: fan

  - platform: template
    name: "AC WindFree 60"
    on_press:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38000Hz
          code: [615, -17666, 3014, -8989, 531, -466, 532, -1466, 530, -467, 531, -469, 529, -466, 532, -468, 530, -468, 530, -471, 527, -495, 503, -1493, 503, -494, 504, -495, 502, -1493, 530, -1466, 531, -1466, 531, -466, 532, -1466, 530, -1465, 531, -1465, 531, -1468, 528, -1494, 502, -495, 503, -495, 503, -494, 504, -1493, 528, -470, 529, -468, 530, -468, 530, -468, 530, -467, 531, -467, 531, -467, 531, -467, 531, -468, 530, -467, 531, -468, 530, -472, 526, -470, 528, -472, 526, -495, 503, -494, 504, -494, 503, -494, 530, -469, 530, -469, 529, -467, 531, -466, 532, -467, 531, -467, 531, -467, 531, -468, 530, -467,  [... cut ...] 4, 534, -463, 535, -1461, 510, -1488, 509, -1486, 510, -1487, 509, -1486, 510, -1489, 507, -1492, 504, -1491, 505, -514, 484, -515, 532, -467, 533, -464, 534, -463, 535, -1460, 511, -488, 510, -488, 510, -1487, 509, -1486, 510, -489, 509, -1488, 508, -492, 506, -1492, 504, -493, 505, -516, 482, -515, 534, -464, 535, -463, 535, -463, 534, -464, 509, -1487, 510, -1488, 508, -1487, 509, -1487, 509]
          repeat: 0
          transmitter_id: fan`

ckeck this code and tell me if its working or not

AyuuSaxena, 2025-05-19:
> > 5 Step Cooling is an energy-saving feature found in some Samsung air conditioners. It allows the compressor to operate at five different levels (100%, 80%, 60%, 40%, and 20%), adjusting power based on room conditions and cooling needs.
> 
> ```
> remote_transmitter:
>     - pin: GPIO26
>       carrier_duty_percent: 33%
>       id: fan
>     
> button:
>   - platform: template
>     name: "AC WindFree 120"
>     on_press:
>       - remote_transmitter.transmit_raw:
>           carrier_frequency: 38000Hz
>           code: [616, -16984, 3018, -8990, 575, -422, 576, -1421, 576, -421, 577, -422, 528, -470, 530, -468, 531, -465, 533, -467, 600, -394, 605, -1394, 602, -393, 605, -394, 604, -1394, 603, -1397, 599, -1397, 600, -399, 599, -1397, 599, -1420, 503, -1493, 531, -1466, 601, -1394, 604, -393, 605, -394, 603, -393, 605, -392, 606, -1392, 604, -394, 604, -395, 603, -396, 602, -398, 600, -398, 600, -420, 578, -420, 528, -470, 530, -469, 598, -397, 603, -395, 604, -393, 605, -394, 604, -394, 604, -393, 605, -393, 605, -392, 606, -395, 603, -394, 604, -397, 601, -421, 577, -421, 577, -400, 598, -419, 579, -421, 528, -469, 531, -1466, 602, -1393, 605, -1391, 606, -1393, 603, -2902, 2993, -8989, 603, -1391, 606, -395, 603, -395, 603, -421, 577, -421, 577, -400, 598, -421, 577, -419, 530, -469, 599, -1396, 604, -394, 605, -395, 603, -1393, 603, -1390, 606, -394, 604, -1391, 605, -395, 603, -1420, 576, -1420, 576, -1418, 579, -419, 503, -1493, 602, -396, 603, -1394, 604, -1391, 605, -1393, 603, -1391, 605, -1391, 605, -1395, 601, -1420, 576, -1420, 576, -419, 579, -419, 504, -495, 531, -468, 599, -397, 603, -1394, 604, -394, 604, -392, 606, -1392, 604, -1392, 604, -393, 605, -1393, 603, -396, 602, -1397, 599, -421, 577, -421, 577, -420, 530, -468, 531, -466, 601, -396, 604, -394, 605, -1392, 605, -1392, 604, -1392, 604, -1392, 604]          
>           repeat: 0
>           transmitter_id: fan
> 
>   - platform: template
>     name: "AC WindFree 100"
>     on_press:
>       - remote_transmitter.transmit_raw:
>           carrier_frequency: 38000Hz
>           code: [618, -17675, 3039, -8964, 532, -464, 534, -1464, 532, -465, 533, -465, 533, -466, 532, -466, 532, -470, 528, -470, 528, -493, 505, -1490, 481, -518, 529, -468, 531, -467, 532, -466, 532, -467, 532, -1465, 531, -1466, 530, -1465, 531, -1464, 532, -1491, 505, -1470, 527, -493, 505, -470, 528, -491, 507, -493, 528, -468, 532, -468, 531, -467, 532, -465, 533, -466, 532, -466, 532, -466, 532, -465, 533, -465, 533, -465, 533, -467, 531, -467, 531, -493, 505, -493, 505, -493, 505, -493, 505, -494, 504, -493, 528, -469, 531, -467, 532, -465, 533, -467, 532, -465, 533, -464, 534, -466, 532, -466, 532, -464, 534, -1463, 533, -1466, 530, -1469, 527, -1492, 530, -2948, 3018, -8990, 528, -1468, 531, -468, 531, -465, 534, -465, 533, -465, 533, -465, 533, -465, 533, -465, 533, -466, 532, -1465, 531, -465, 533, -492, 506, -469, 529, -1492, 504, -1491, 502, -1492, 532, -468, 531, -1465, 533, -1462, 535, -1464, 532, -466, 532, -1464, 532, -465, 533, -1468, 528, -1468, 528, -469, 529, -493, 505, -492, 506, -1491, 530, -1466, 531, -1467, 530, -465, 533, -466, 532, -467, 531, -466, 532, -466, 532, -1464, 532, -467, 531, -468, 530, -1491, 505, -1490, 506, -492, 506, -1490, 530, -468, 531, -1465, 532, -466, 532, -465, 533, -464, 534, -465, 533, -466, 532, -465, 533, -465, 533, -1465, 531, -1467, 529, -1491, 505, -1491, 532]
>           repeat: 0
>           transmitter_id: fan
>           
> 
>   - platform: template
>     name: "AC WindFree 80"
>     on_press:
>       - remote_transmitter.transmit_raw:
>           carrier_frequency: 38000Hz
>           code: [619, -17689, 2991, -9016, 505, -464, 534, -1463, 533, -465, 533, -463, 535, -466, 532, -467, 531, -465, 533, -470, 528, -493, 505, -1489, 508, -493, 601, -396, 532, -1465, 532, -1464, 532, -1463, 533, -464, 534, -1463, 533, -1463, 533, -1465, 531, -1491, 505, -1470, 526, -491, 507, -492, 600, -1393, 535, -464, 534, -464, 534, -464, 534, -465, 532, -465, 533, -465, 533, -463, 535, -464, 534, -463, 535, -464, 534, -466, 532, -467, 531, -467, 531, -492, 506, -491, 507, -493, 505, -491, 602, -397, 532, -465, 533, -465, 533, -465, 533, -466, 532, -465, 533, -465, 533, -464, 534, -465, 533, -464, 534, -464, 534, -1465, 531, -1470, 526, -1492, 504, -1488, 535, -2949, 3018, -8988, 533, -1463, 534, -465, 533, -464, 534, -464, 534, -464, 534, -464, 534, -464, 534, -464, 534, -465, 533, -1467, 529, -468, 530, -493, 505, -1489, 508, -1491, 601, -395, 604, -1392, 534, -464, 534, -1462, 534, -1463, 533, -1463, 533, -463, 535, -1464, 532, -468, 530, -1491, 505, -1465, 531, -1490, 531, -1466, 603, -1393, 534, -1462, 534, -1463, 534, -1463, 533, -463, 535, -464, 534, -465, 533, -466, 532, -466, 532, -1466, 530, -493, 505, -491, 507, -1490, 601, -1396, 532, -466, 532, -1463, 508, -489, 535, -1463, 533, -464, 534, -464, 534, -464, 534, -465, 533, -468, 530, -466, 532, -469, 529, -1469, 527, -1490, 531, -1464, 605, -1392, 534]
>           repeat: 0
>           transmitter_id: fan
> 
>   - platform: template
>     name: "AC WindFree 60"
>     on_press:
>       - remote_transmitter.transmit_raw:
>           carrier_frequency: 38000Hz
>           code: [615, -17666, 3014, -8989, 531, -466, 532, -1466, 530, -467, 531, -469, 529, -466, 532, -468, 530, -468, 530, -471, 527, -495, 503, -1493, 503, -494, 504, -495, 502, -1493, 530, -1466, 531, -1466, 531, -466, 532, -1466, 530, -1465, 531, -1465, 531, -1468, 528, -1494, 502, -495, 503 [... cut ...] 09, -1486, 510, -1489, 507, -1492, 504, -1491, 505, -514, 484, -515, 532, -467, 533, -464, 534, -463, 535, -1460, 511, -488, 510, -488, 510, -1487, 509, -1486, 510, -489, 509, -1488, 508, -492, 506, -1492, 504, -493, 505, -516, 482, -515, 534, -464, 535, -463, 535, -463, 534, -464, 509, -1487, 510, -1488, 508, -1487, 509, -1487, 509]
>           repeat: 0
>           transmitter_id: fan
> ```
> 
> ckeck this code and tell me if its working or not

Thanks, man. It worked when I removed repeat: 0.