crankyoldgit/IRremoteESP8266 #2201: Misidentified Gree air conditioner protocol

TTXL3099, 2025-05-03:
I am from China, the origin of Gree air conditioners. I own two Gree air conditioners, one from 2013 (remote control model YAD0F) and one from 2024 (YAP0F20). Their remote controls can be interchanged and controlled normally. But when I use this library to try to decode and control my air conditioner, the remote control is decoded into Kelvinator protocol (128 Bits)However, the power, mode, temperature and other data obtained through decoding are all correct.
````
Timestamp : 000009.561
Library   : v2.8.6

Protocol  : KELVINATOR
Code      : 0x010920500040008001092070000080C0 (128 Bits)
Mesg Desc.: Power: Off, Mode: 1 (Cool), Temp: 25C, Fan: 0 (Auto), Turbo: Off, Quiet: Off, XFan: Off, Ion: Off, Light: On, Swing(H): Off, Swing(V): 0 (Off)
uint16_t rawData[279] = {9050, 4484,  664, 1622,  666, 538,  644, 562,  646, 538,  644, 538,  644, 562,  646, 536,  648, 516,  662, 1670,  642, 538,  646, 536,  644, 1622,  662, 566,  642, 538,  644, 518,  666, 564,  646, 536,  644, 538,  644, 562,  646, 538,  646, 536,  644, 1666,  648, 534,  646, 536,  646, 564,  644, 538,  644, 538,  644, 562,  646, 1638,  648, 536,  648, 1664,  644, 536,  644, 520,  664, 1640,  646, 562,  646, 19960,  668, 562,  646, 534,  644, 538,  648, 562,  646, 534,  646, 536,  644, 564,  644, 538,  644, 538,  644, 564,  646, 536,  644, 516,  668, 562,  646, 536,  644, 1620,  666, 562,  646, 536,  646, 536,  646, 562,  644, 518,  662, 536,  644, 566,  644, 516,  666, 536,  646, 562,  644, 516,  666, 538,  644, 564,  644, 536,  644, 536,  648, 560,  646, 1618,  666, 39962,  9028, 4504,  646, 1638,  648, 562,  646, 538,  644, 536,  646, 562,  646, 536,  646, 534,  648, 560,  652, 1634,  644, 538,  644, 564,  618, 1666,  646, 536,  644, 540,  646, 560,  646, 536,  648, 536,  646, 562,  644, 538,  644, 538,  644, 562,  646, 1618,  666, 536,  646, 564,  644, 536,  644, 540,  646, 562,  646, 532,  644, 1642,  646, 1640,  646, 1664,  644, 538,  644, 536,  646, 1666,  644, 538,  648, 19960,  668, 562,  646, 538,  644, 516,  666, 564,  644, 536,  646, 536,  646, 562,  646, 536,  644, 536,  646, 562,  644, 536,  646, 538,  644, 542,  666, 536,  644, 536,  646, 562,  646, 536,  644, 540,  644, 562,  646, 536,  644, 538,  642, 544,  664, 538,  644, 1640,  646, 562,  646, 536,  646, 538,  646, 564,  644, 536,  644, 518,  664, 1668,  646, 1620,  668};  // KELVINATOR
uint8_t state[16] = {0x01, 0x09, 0x20, 0x50, 0x00, 0x40, 0x00, 0x80, 0x01, 0x09, 0x20, 0x70, 0x00, 0x00, 0x80, 0xC0};

Timestamp : 000038.170
Library   : v2.8.6

Protocol  : KELVINATOR
Code      : 0x7C0D2050112100607C0D207000005080 (128 Bits)
Mesg Desc.: Power: On, Mode: 4 (Heat), Temp: 29C, Fan: 5 (High), Turbo: Off, Quiet: Off, XFan: Off, Ion: Off, Light: On, Swing(H): On, Swing(V): 1 (Auto)
uint16_t rawData[279] = {9088, 4420,  720, 506,  700, 506,  698, 1610,  698, 1610,  694, 1610,  700, 1610,  702, 1602,  700, 506,  700, 1608,  698, 508,  696, 1610,  700, 1606,  700, 506,  700, 508,  698, 506,  700, 506,  698, 508,  702, 504,  700, 506,  700, 506,  702, 504,  698, 1610,  700, 504,  702, 502,  700, 506,  698, 506,  700, 506,  698, 508,  700, 1606,  700, 504,  700, 1608,  698, 508,  698, 508,  696, 1610,  700, 506,  700, 19916,  722, 1608,  700, 504,  700, 506,  698, 506,  700, 1608,  698, 508,  700, 506,  700, 504,  702, 1604,  700, 506,  698, 508,  698, 506,  698, 506,  700, 1588,  718, 508,  698, 506,  700, 508,  702, 504,  698, 508,  698, 506,  698, 508,  700, 508,  696, 510,  698, 506,  698, 508,  700, 506,  700, 506,  700, 506,  702, 502,  704, 1606,  698, 1608,  700, 508,  696, 39930,  9062, 4420,  722, 506,  700, 506,  700, 1604,  702, 1606,  700, 1606,  668, 1638,  700, 1610,  696, 508,  698, 1608,  726, 480,  702, 1606,  702, 1606,  700, 506,  670, 538,  668, 536,  668, 536,  670, 516,  690, 538,  670, 534,  670, 534,  672, 514,  690, 1636,  700, 506,  698, 506,  698, 506,  700, 506,  702, 504,  698, 488,  720, 1586,  722, 1606,  700, 1604,  700, 508,  700, 506,  698, 1608,  670, 534,  670, 19948,  726, 504,  704, 502,  726, 506,  700, 506,  702, 502,  702, 504,  700, 504,  702, 504,  700, 506,  672, 534,  670, 538,  668, 536,  666, 538,  668, 536,  668, 538,  670, 538,  666, 536,  670, 536,  670, 534,  646, 540,  666, 1662,  698, 506,  700, 1608,  702, 504,  700, 506,  702, 504,  668, 534,  670, 534,  698, 508,  668, 536,  668, 536,  698, 1610,  702};  // KELVINATOR
uint8_t state[16] = {0x7C, 0x0D, 0x20, 0x50, 0x11, 0x21, 0x00, 0x60, 0x7C, 0x0D, 0x20, 0x70, 0x00, 0x00, 0x50, 0x80};

Timestamp : 000039.419
Library   : v2.8.6

Protocol  : KELVINATOR
Code      : 0x740D2050112100E0740D207000005000 (128 Bits)
Mesg Desc.: Power: Off, Mode: 4 (Heat), Temp: 29C, Fan: 5 (High), Turbo: Off, Quiet: Off, XFan: Off, Ion: Off, Light: On, Swing(H): On, Swing(V): 1 (Auto)
uint16_t rawData[279] = {9030, 4472,  670, 536,  670, 538,  668, 1616,  690, 562,  670, 1638,  670, 1614,  692, 1638,  670, 534,  670, 1638,  668, 534,  670, 1638,  670, 1638,  668, 516,  690, 536,  668, 536,  668, 516,  690, 516,  692, 538,  668, 514,  690, 538,  668, 514,  692, 1636,  670, 538,  668, 536,  668, 540,  666, 536,  668, 538,  668, 538,  668, 1636,  668, 538,  666, 1620,  690, 516,  690, 534,  672, 1638,  670, 534,  668, 19950,  692, 1636,  670, 534,  670, 534,  670, 538,  668, 1640,  670, 534,  670, 534,  670, 538,  702, 1606,  668, 536,  670, 534,  670, 538,  698, 506,  724, 1562,  720, 486,  720, 506,  698, 506,  670, 536,  700, 506,  670, 536,  698, 508,  672, 532,  700, 508,  668, 538,  670, 536,  666, 538,  698, 506,  702, 504,  698, 508,  668, 1636,  666, 1640,  670, 1616,  692, 39954,  9030, 4452,  690, 516,  668, 558,  698, 1608,  700, 508,  700, 1608,  700, 1606,  700, 1606,  700, 504,  702, 1606,  670, 538,  668, 1638,  700, 1608,  700, 504,  698, 508,  696, 508,  698, 508,  698, 508,  700, 484,  724, 502,  702, 504,  702, 502,  726, 1588,  742, 482,  700, 506,  700, 504,  700, 508,  698, 508,  700, 506,  698, 1608,  702, 1608,  668, 1640,  698, 504,  700, 508,  698, 1608,  702, 482,  720, 19916,  722, 506,  698, 506,  700, 506,  702, 506,  700, 504,  700, 506,  700, 506,  700, 506,  698, 508,  700, 506,  698, 506,  700, 506,  700, 506,  700, 506,  700, 484,  722, 504,  702, 502,  702, 506,  698, 506,  670, 516,  718, 1608,  702, 502,  702, 1608,  700, 506,  700, 506,  698, 506,  698, 506,  698, 508,  698, 508,  698, 510,  700, 506,  700, 506,  700};  // KELVINATOR
uint8_t state[16] = {0x74, 0x0D, 0x20, 0x50, 0x11, 0x21, 0x00, 0xE0, 0x74, 0x0D, 0x20, 0x70, 0x00, 0x00, 0x50, 0x00};
`
```

TTXL3099, 2025-05-03:
[image]

[image]

[image]

[image]

NiKiZe, 2025-05-03:
Several ACs use the same protocol, it is not uncommon for one manufacturer to buy the base from someone else.
Since there almost never is any public documentation on these things, we can only give any new protocol a best guess name.

I read you post as everything working, and in that case there won't be anything to change.

closed 2025-05-03 by NiKiZe

crankyoldgit, 2025-05-03:
@TTXL3099 All the protocols are reverse engineered from collected captures. Adding timer support is very low on the list as there are other ways to provide that functionality rather than via native support. i.e. Home Automation or user programming etc.

I have an A/C that uses the Kelvinator protocol. It doesn't respond to the "single" message  version. It needs both. Yes, the protocols are very similar. They could probably be merged. You're welcome to do so.